PX4/PX4-Autopilot #6844: Changing gazebo SITL source files doesn't rebuild

matanhavi, 2017-03-18:
There is no dependency rule to invoke rebuild in case of internal files on the SITL has been changed

dagar, 2017-03-19:
This is a limitation of cmake ExternalProject_Add. https://cmake.org/cmake/help/v3.5/module/ExternalProject.html

@jgoppert what was the benefit of pulling the gazebo_sitl build into PX4's build system? I'm wondering if we could structure this differently and avoid this issue and make it easier to build sitl_gazebo with or without PX4.

jgoppert, 2017-03-20:
I'm in favor of moving the sitl_gazebo stuff for sim and CI outside of px4/Firmware. We are getting ready to sync with rotors and the current approach is more of this, so see my comments here: https://github.com/PX4/Firmware/pull/6380

matanhavi, 2017-03-21:
I have found way to rebuild external. you can see at PR #6845 
From my tests it works, can you check?

I seen today that also changes on nuttx doesn't re-compile the lib, but in different way. if you rebuild second time, it rebuild the whole project, not just the lib (or the changed file). I think its better to leave it to other issue, mybe check all the libs/files for good dependency build.

closed 2017-03-23 by mhkabir