PX4/PX4-Autopilot #6851: LPE_GPS_ON parameter not found

tomatac, 2017-03-19:
After I set SYS_MC_EST_GROUP = 1, I cannot set the LPE_GPS_ON parameter.
I get the error LPE_GPS_ON not found

I am trying to set the use PX4FLOW without the GPS and I understood that the recommended position estimator is LPE.

FW version: v1.5.5 (1.5.5 255), 17106431

mhkabir, 2017-03-19:
The parameter no longer exists. 

Use the LPE_FUSION bitmask for fusion control.

closed 2017-03-19 by mhkabir

tomatac, 2017-03-20:
I could not find LPE_FUSION as a parameter. In fact I could not find any parameters that includes LPE but a lot of parameters that includes EKF2. I used both QGroundControl and nsh. That is strange because the SYS_MC_EST_GROUP is set to 1, so the LPE should be the current state estimator. The only documentation I could find for PX4FLOW recommends to use LPE. Is LPE still supported for PX4FLOW?

iozfiliz, 2017-03-23:
You prolly use a firmware version built without LPE in it. So changing SYS_MC_EST_GROUP won't help. Checkout your cmake file if LPE is commented in or out.

nancy03610, 2017-04-08:
Do you have any progress with this? I met with the same problem. I checked the nuttx_px4fmu-v2_default.cmake, and the modules/local_position_estimator is not commented. But still I cannot find LPE_FUSION. How do you fix this?