PX4/PX4-Autopilot #6852: PX4Flow and LPE

ecmnet, 2017-03-19:
@jgoppert @mhkabir I am not sure whether this is an issue or just on my side, but currently I try to re-integrate PX4Flow to my position setup using LPS with vision to cover situations, where vision cannot provide any information. Unfortunately with 1.6 Rc1, Flow initializes only very rarely in my use cases and looses confidence very quickly. I guess this could be due to the noise model added lately where I see a lot of constraints (like agl, etc). With EKF2 however, flow seem to work. Regarding Flow, I tried both, the standard firmware and in addition the exposure setup provided by James. Did I miss something or is there a known issue?

jgoppert, 2017-03-20:
I've flown LPE + flow about 200 flights within the last month. I was using latest master at the time, about a week ago was my last flight. Can you provide some logs? I'd like to see your flow quality. I recently made a patch to my exposure branch to lower the exposure, so this might have thrown you off.

tomatac, 2017-03-20:
I cannot use the PX4Flow with LPE. The SYS_MC_EST_GROUP is set to 1, so the LPE should be the current state estimator but I cannot find  LPE_FUSION or any LPE* as a parameter. I used both QGroundControl and nsh to check. 
Is there a specific firmware we should use with PX4Flow and LPE?
Is EKF2 better now for PX4Flow?

FW version: v1.5.5 (1.5.5 255), 17106431

jgoppert, 2017-03-20:
For px4fmu-v2 we couldn't fit both ekf2 and lpe so there is another image  (px4fmu-v2_lpe.px4) please upload that. You can find it in the 1.6 release zip file.

I haven't flown ekf2 with flow so I can't say.

ecmnet, 2017-03-20:
@jgoppert Currently I have no log. But flow quality was reported ok (I also changed min.quality). I am using px4fmu-v4. I also checked raw flow measurements with both QGC Analysis and with MAVGCL. I also used your recent change to PX4Flow-Firmware. Later I'll change to latest master and provide some logs.

jgoppert, 2017-03-20:
Thanks. If the flow quality is good and the picture looks fine it is probably worth looking at your lpe fusion flags. I haven't tested on px4fmu-v4, but can't think of any issues there.

ecmnet, 2017-03-20:
Of course I checked the fusion flags. I used to rely on vision position only the last few weeks, but my settings for flow tests were like this:

[image]

I disabled land detector fusion on purpose, btw.

ecmnet, 2017-03-20:
@jgoppert As I have no complete setup available (to fly with and to record a full log), I just share the MAVGCL graphs for integrated flow x,y and quality. AGL was about 1 m above ground. Note, that flow quality is normalized here.

[image]

[image]

Although flow seems at least to be available in this test, flow init was not triggered. Even more curious is the fact, that I get huge movements of the position (about +/- 3m when moving the vehicle by hand for 10 cm) if flow is initialized (which happens from time to time). So the reason could be some scaling issues, although the scale is set to 1.3m (default). I noticed also a change in LPE to prefer LIDAR instead of fusing LIDAR and flow-sonar, but I assume, that for scaling flow agl is used (based on LIDAR if available).

jgoppert, 2017-03-20:
Lpe has a terrain altitude estimate that is used with altitude to compute
agl, so not lidar/sonar dependent, will use either.  If you have land
disabled and pick it up 1 m you should get a solid green light. If not let
me know firmware version and I will test. Current flow setup is working
very well for me so likely something config related.

ecmnet, 2017-03-20:
@jgoppert Picking it up to 1m and holding it still, the light is flashing green with disabled land fusion, although flow quality > 150.

ecmnet, 2017-03-21:
@jgoppert I created a small log file holding the vehicle in the air:

http://logs.px4.io/plot_app?log=ece824ff-20f7-47b7-be78-def875b36ade

ecmnet, 2017-03-21:
@jgoppert Got it! I have set baro noise to 100 to rely purely on LIDAR distance. Although this worked for vision position, AGL was never determined, which caused flow not scaled at all. Reverting baro noise to 3m, flow initializes. Sorry for any circumstances.

closed 2017-03-21 by ecmnet

mhkabir, 2017-03-21:
> A bit confusing is, that if baro fusion is disabled, flow is never initialized. Wouldn't it be better to fall back to raw distance measurements in the case when agl is not available?

I fully agree with this :)

Ahrovan, 2017-03-21:
@ecmnet plz report your config for Optical Flow for future test 
thx I want you to repeat the test

ecmnet, 2017-03-21:
@Ahrovan Nothing special about the config. Hardware is setup as described in the documentation, settings are mostly defaults.

tomatac, 2017-03-23:
@ecmnet Could you share your LPE_FUSION settings?

ecmnet, 2017-03-23:
In my case, I set them to 162. (Refer to screenshot in this thread)

vinhk, 2017-04-06:
@ecmnet How is your hovering when set LPE_FUSION to 162?