PX4/PX4-Autopilot #6855: attitude tracking

Stifael, 2017-03-19:
Tested in SITL:
it seems that the transformation from quaternion to roll,pitch and yaw are not done correctly and do not represent what actually happens. giving roll command, the tracking of the roll setpoint seems to be correct at the beginning, but over time the logged roll setpoint is opposite the estimated roll. It is to note that this does not mean that the vehicle does not fly correctly, but it tells us that the transformation is not correct. 

[image]

bkueng, 2017-03-20:
Same issue as https://github.com/PX4/pyulog/issues/13

Stifael, 2017-03-22:
I think we should really try to solve this issue, since I reached a point where I am not sure anymore what is tracking and what is transformation.
[image]
This tracking looks terrible, but during actual flight I did not have the impression that the tracking was that bad. It also seems that the logged tracking is worse in non-manual control

dagar, 2017-04-12:
Did you find the problem?

Stifael, 2017-04-18:
not found yet, but the transformation seems not consistent.

TSC21, 2017-08-07:
@Stifael I also saw the same behavior with a customized VTOL model on SITL, where I get the RPY setpoints reversed WRT with the RPY reported by the attitude estimation. Did you find anything that may be leading to the problem?

Stifael, 2018-10-12:
@bresch can you add your findings as well?

bresch, 2018-10-16:
During missions, for example in a survey, the yaw setpoint has large jumps at each waypoint and a pitch is commanded at the same time. Everything seems correct during the flight, but when looking at the log, the estimated roll is far from its setpoint. When the yaw setpoint is reached, the signals are perfectly aligned again. 
Example: https://logs.px4.io/plot_app?log=bd578b0d-995d-4cfd-a4d6-abb7cc215466

closed 2019-02-04 by stale[bot]