PX4/PX4-Autopilot #6856: VTOL Standard transition improvements

dagar, 2017-03-19:
**Transition to FW**
 - FW actuators 100% immediately (integrators reset at start)
 - MC actuators 100% and will ramp down automatically as vehicle produces lift. Require MC thrust setpoint to drop below a parameterized threshold before considering forward transition successful.
 - use roll and yaw setpoints from FW position controller?

**Transition to MC**
 - FW actuators ramp down proportionally to transition progress (airspeed or time)
 - MC actuators ramp up to 100% sooner (over 1-2 seconds?)
 - monitor ground speed and consider transition complete if ground speed <= max MC speed
 - check MPC initialization. Shouldn't be trying to abruptly stop immediately.

**Ideas for future improvement**
 - We need position control at all times. With any airspeed L1 navigation is more appropriate for a standard VTOL.
 - Multicopter with pusher should be allowed to achieve speeds much closer to FW minimum airspeed. This bridges the gap during transitions.
 - Transition to FW start using FW position controller sooner, but prevent TECS from pitching down
    - use MC position controller (with appropriate VT_DWN_PITCH_MAX) to first accelerate vehicle to an appropriate level that FW position (and TECS) can takeover. TECS can have control of the throttle, but not allowed to pitch down.
 - Transition to MC
   - Two phase transition
   - Phase 1: FW position control with throttle cut to brake (forced to hold altitude) once speed decreases to acceptable level for MC position controller move on to phase 2. MC actuators are blended in to be at 100% for phase 2. Flaps could also be deployed for braking.
   - Phase 2: MC position controller initialized with current velocity and given full control for reasonable braking

tubeme, 2017-03-22:
Very good ideas @dagar 

> use roll and yaw setpoints from FW position controller?

Why not use the pitchSP from FW pos controller as well? In my opinion the moment the transition begins, we should think of the aircraft as airplane, but in underspeed condition so the MC keeps the frame from stalling. But otherwise we should accept the machine as airplane so the FW SetPoints but controlled by the MC until the airspeed is enough.

tubeme, 2017-03-22:
> MC actuators ramp up to 100% sooner (over 1-2 seconds?)

There is no point in ramping up to 100% because we have lift at that moment and speed. So it must be gradual ramp without disturbing the alt SP. If we ramp up, we will climb at that moment. Tha ramp should be more gradient, and once bellow the min airspeed should be considerable to hold the altitude.

tubeme, 2017-03-22:
> FW actuators ramp down proportionally to transition progress (airspeed or time)

The actuators can ramp down after we've reached the min FW airpseed because up to that moment we have lift in the wing and can control the frame with the FW control surfaces. In strong weather it is even more important to have this control.

But there is another aspect of the back transition which is very important.

Our best back transitions are when the plane is without throttle and gliding a little. Not much but we stop the pusher about 1-2-3 seconds before we switch the back transition. This way the transition is more smoother and the MC is getting grip faster. We stop faster as well without making flare. We loose a little altitude but with proper tuning not much. Also when gliding the plane is with a little more angle of attack and when the MC kicks in it gives braking as well, it turns like a natural flare, becuase once the MC is on it pushes back the nose of the vtol and levels.

tubeme, 2017-03-22:
> monitor ground speed and consider transition complete if ground speed <= max MC speed

Does this mean that if we are bellow max MC speed the vtol will be considered as MC, the MC Pos controller will kick in and the vtol will do stop at its position, given we fly in Position Mode and we don't touch the controls?

sanderux, 2017-03-22:
@tubeme 

> Why not use the pitchSP from FW pos controller

Because in underspeed condition the FW would pitch down

> There is no point in ramping up to 100% because we have lift at that moment and speed

100% weight is not the same as 100% throttle, giving it full weight will enable it to hold altitude with whatever throttle required.

> The actuators can ramp down after we've reached the min FW airpseed

You are disregarding VTOL's without airspeed sensor

> Does this mean that if we are bellow max MC speed the vtol will be considered as MC, the MC Pos controller will kick in and the vtol will do stop at its position, given we fly in Position Mode and we don't touch the controls?

yes

dagar, 2017-03-22:
Below min airspeed TECS will want to pitch down, which we can't allow during forward transition.

The idea behind the having MC actuators at 100% sooner (or for longer) is that the multicopter position controller is better able to handle it. If the vehicle has airspeed and is holding altitude then multicopter thrust will be minimal and only increase (automatically ramp up) on its own as the vehicle is no longer able to hold altitude.

tubeme, 2017-03-22:
> Below min airspeed TECS will want to pitch down, which we can't allow during forward transition.

We've had a lot of this cases in the past... I understand now.

Thanks @sanderux and @dagar

tubeme, 2017-03-22:
@dagar @sanderux 

> Because in underspeed condition the FW would pitch down

We should consider also the PSP offset here during the transitions if we want to be consistent with the difference between the two vehicles if such. This concerns a lot the pitchSP. Now we balance it with other means but in reality more than 3 degrees of difference is a serious difference in the angle of attack if we don't account for it.

sanderux, 2017-03-22:
@tubeme already implemented

https://github.com/PX4/Firmware/pull/6728

tubeme, 2017-03-23:
@sanderux 

I suppose this fix is in the current master and not in the 1.6.0 RC1?

Sorry I did not follow on this... Because we mainly use the stable versions for the current phase of our tests, do you have any idea when is the RC2 or next release going to be available?

dagar, 2017-04-11:
TODO - enable weathervane during transitions. MC yaw is either useless or potentially harmful.

sanderux, 2017-04-11:
Not sure if thats a good idea, MC needs to try and hold position even in a cross wind untill the FW has enough authority.

tubeme, 2017-04-11:
@dagar That is definitely NO NO for the transition! We've had cases where the yaw was not holding properly in strong wind and in a mission flight the VTOL did not hold its YAW, it pointed toward the pilot who wanted to make a very short take off transition mission at 2 meters height and it got almost at his head, and then crashed in a tree behind him. 

Because the strongest point of the VTOL is take off in tight spaces we have to make sure we ALWAYS get position control especially the YAW when transitioning. Imagine it like a regular airplane take off but in the air and the same is true for the back transition.

dagar, 2017-04-12:
Thanks for the feedback @tubeme. My main thought here is that once the vehicle has any airspeed it should be using the FW control surfaces to maintain heading and MC should be dedicated to keeping the vehicle at altitude.

closed 2018-02-21 by PX4BuildBot

TSC21, 2018-02-21:
@dagar how much of these have support on the firmware already?