PX4/PX4-Autopilot #6864: PX4Flow to Pixhawk configurations to get stable LPE

vinhk, 2017-03-20:
Dear all,

I am working on indoor small drone where GPS is not allowed. I have followed the tutorials on Px4 documentations in connecting the PX4Flow to Pixhawk. Here is what I have done:

1. From list of options for Firmware in QGC, I flashed Pixhawk using Master in QGC.
2. I downloaded github directory containing Px4Flow files: https://github.com/PX4/Flow
3. I flashed the downloaded files and flashed it using this instructions: https://pixhawk.org/dev/px4flow
4. I connected Px4Flow to Pixhawk as follows:
 
[image]

5. I noticed that the SENS_FLOW_ROT is set to 270 degrees in QGC but the documentation says make it 0 degrees based on this: 
[image]

6. So I set the SENS_FLOW_ROT to 0 degrees using the documentation

7. I set the SYS_MC_EST_GROUP to EKF2 and the POS_CTRL mode was bad. It did not stay hovering.

I would like to tried LPE with the community suggestion. I did not have stable hover before, but since I have just flashed these codes for Pixhawk and Px4Flow using masters I feel that I am doing that part correct. @andre-nguyen @ChristophTobler suggested me to check for orientation. Could someone assist me with getting this setup to work using LPE in POS_CTRL mode. I would really appreciate if I could get a stable hover like so: https://www.youtube.com/watch?v=CccoyyX-xtE
 
Thank you

ChristophTobler, 2017-03-21:
https://pixhawk.org/dev/px4flow is outdated and I guess we need to remove that page or link it correctly. We are sorry for that.
Please make sure you have the following things working correctly
* As we have already told you on Gitter, `SENS_FLOW_ROT` needs to be `270 degrees` if you use master.
* Flow output: fwd +y, rev -y, right -x, left +x
* Is your quad tuned correctly? 
* Does the distance sensor work? 
* Does altitude hold work?
* What is your flow quality?
* Is optical flow enabled in `LPE_FUSION`?
* If you use EKF2, have you set `EKF2_AID_MASK` to use optical flow?
* If you use EKF2, have you set `EKF2_HGT_MODE` to use the range finder?

And please [provide a log](http://logs.uaventure.com/) of a flight.

ChristophTobler, 2017-03-21:
Have a look at [this guide](https://dev.px4.io/optical-flow-outdoors.html)

mhkabir, 2017-03-21:
@ChristophTobler Lets move everything into Devguide and update the Optical Flow page to reflect latest LPE params.

https://dev.px4.io/optical-flow-outdoors.html is outdated in terms of LPE params.

ChristophTobler, 2017-03-21:
@mhkabir I agree. I guess it is also a bit confusing that we have two pages in the dev guide: https://dev.px4.io/flow_lidar_setup.html, https://dev.px4.io/optical-flow-outdoors.html. I think we should just have one page for optical flow with all estimators on there.
I'll create one with EKF2 (and maybe INAV with the warning that it is outdated). Can you then add the LPE part?

vinhk, 2017-03-21:
Based on the suggestions, I have reset all params and recalibrate the Pixhawk in QGC:

- [ ]  `SENS_FLOW_ROT` needs to be 270
**Complete**

- [ ] Flow output: fwd +y, rev -y, right -x, left +x, Does the distance sensor work? 
 `M1:OPTICAL_FLOW_RAD.integrated_y`, `M1:OPTICAL_FLOW_RAD.integrated_x`, and  `M1:OPTICAL_FLOW_RAD.distance`
**Complete**
https://www.youtube.com/watch?v=11pUv1R0G-M&feature=youtu.be
Is your quad tuned correctly?

Does altitude hold work?
What is your flow quality?
Is optical flow enabled in LPE_FUSION?
If you use EKF2, have you set EKF2_AID_MASK to use optical flow?
If you use EKF2, have you set EKF2_HGT_MODE to use the range finder?

tomatac, 2017-03-22:
Is LPE still the recommended position estimator to be used with PX4FLOW?
Is there a list of the LPE parameters to be used with PX4FLOW?

vinhk, 2017-03-22:
@jgoppert @mhkabir @ChristophTobler , what are your suggestions?

ChristophTobler, 2017-03-22:
You have to tune your quad obviously. Otherwise it won't fly (well) at all. Have you ever flown it manually?

vinhk, 2017-03-22:
@ChristophTobler, I have flown it many times. We had drift issues in manual flights.

ChristophTobler, 2017-03-22:
Then please provide a log here: http://logs.uaventure.com/

ChristophTobler, 2017-03-22:
@tomatac Yes, LPE is still the recommended one. @mhkabir is updating the parameters here https://github.com/PX4/Devguide/pull/98

vinhk, 2017-03-22:
@ChristophTobler Here is what I'll do:

1st flight: Manual flight and control the quad using only the RC Transmitter. Log it
2nd flight: Altitude hold mode flight and log it as well.

mhkabir, 2017-03-22:
Also send a log for position control please; I want to see what goes wrong.

vinhk, 2017-03-22:
I'll do that. Thanks

ChristophTobler, 2017-03-23:
@vinhk I just realized that your pixhawk is running ekf2 (default). That's also why you can't see `LPE_FUSION` but all `EKF2_*` parameters, right?
@mhkabir I guess we need to mention that in the dev guide (flow related, but also generally).

vinhk, 2017-03-23:
@ChristophTobler, I do see all of EK2_* parameters.

vinhk, 2017-03-23:
@mhkabir @ChristophTobler Even after I change it to LPE, there is no LPE_FUSION in QGC.

ChristophTobler, 2017-03-23:
@vinhk yes. LPE is not supported on a Pixhawk

mhkabir, 2017-03-23:
It is usable just fine. You need to use the px4fmu-v2_lpe target, because we do not have enough flash for both estimators on the Pixhawk with old STM32.

vinhk, 2017-03-23:
@mhkbir does that mean that I need to flash the px4 firmware masters from GitHub or from QGC masters?

ChristophTobler, 2017-03-23:
@vinhk you would need to flash it yourself using the source code from Github. Or you could just use EKF2

vinhk, 2017-03-23:
@ChristophTobler I'll flashed it using Github this time. Please give me all the parameters to enable in QGC so I can test it properly for EKF2 to work with Pixhawk using Px4Flow.

ChristophTobler, 2017-03-23:
@vinhk You flashed it using `make px4fmu-v2_lpe` (LPE) or `make px4fmu-v2_default` (EKF2)? For EKF2 you need to set the two parameters I mentioned https://github.com/PX4/Firmware/issues/6864#issuecomment-288001037

vinhk, 2017-03-23:
@ChristophTobler Previously, I used QGC. Instead of using the default firmware, I used Masters from the list of Px4 firmware.

vinhk, 2017-03-23:
@mhkabir Please give me all parameters I need to set in QGC if I flashed the firmware using `make px4fmu-v2_lpe` as well.

jgoppert, 2017-03-23:
px4fmu-v2_lpe is in the zip file on the github release page.

vinhk, 2017-03-23:
@jgoppert I just flashed Px4 firmware masters from Github and use `make px4fmu-v2_lpe upload`. I can see `LPE_FUSION 247` on QGC.  I also set `SYS_MC_EST_GROUP` to `local_position_estimator, attitude_estimator_q` on QGC. What is the next step?

vinhk, 2017-03-23:
is `LPE_FUSION 247` correct to use PX4Flow without Lidar only sonar that came with Px4Flow indoors?

mhkabir, 2017-03-23:
LPE_FUSION should be 162

vinhk, 2017-03-23:
@mhkabir I'll do that.

tomatac, 2017-03-27:
@mhkabir
162: 1010 0010
OFF bit 0 fuse GPS, requires GPS for alt. init
ON  bit 1 fuse optical flow
OFF bit 2 fuse vision position
OFF bit 3 fuse vision yaw
OFF bit 4 fuse land detector
ON  bit 5 pub agl as lpos down
OFF bit 6 flow gyro compensation
ON  bit 7 fuse baro

Why is the flow gyro compensation OFF?
Do you know if there is any document or post that explains the best practices on setting these parameters?

vinhk, 2017-03-27:
@tomatac You have that opposite. 1010 0010. It start from: GPS, Optical Flow, Vision Position, ... etc.

In this case flow gyro is on based on the 1010 00**1**0

dagar, 2017-03-27:
Did you try using QGC?

[image]

tomatac, 2017-03-27:
Thank you @dagar! I will try using the QGC now that I know that we need a different firmware file (px4fmu-v2_lpe in the zip file on the github). 
The question remains: what are best practices on setting these parameters for position estimation using only PX4FLOW and no GPS? 

If the Default is: 247: 1111 0111
ON  bit 0 fuse GPS, requires GPS for alt. init
ON  bit 1 fuse optical flow
ON  bit 2 fuse vision position
OFF bit 3 fuse vision yaw
ON  bit 4 fuse land detector
ON  bit 5 pub agl as lpos down
ON  bit 6 flow gyro compensation
ON  bit 7 fuse baro

Are these the best settings: 242: 1111 0010 ?
OFF bit 0 fuse GPS, requires GPS for alt. init
ON  bit 1 fuse optical flow
OFF bit 2 fuse vision position (?)
OFF bit 3 fuse vision yaw (?)
ON  bit 4 fuse land detector (?)
ON  bit 5 pub agl as lpos down
ON  bit 6 flow gyro compensation
ON  bit 7 fuse baro (?)

I am not sure on: fuse vision position, fuse vision yaw, fuse land detector, fuse baro.

Does anybody have a recommendation based on flight experience using only PX4FLOW and no GPS?

HockyYoung, 2017-04-08:
I use EKF2 and I Set SENS_FLOW_ROT to 270 degrees ，EKF2_AID_MASK to use optical flow ，EKF2_HGT_MODE to use the range finder，and the Alttitude and Position mode is working well, but I use make px4fmu-v2_lpe upload,i can't make the alttitude mode hold ,what's the problem with LPE?

closed 2018-02-21 by PX4BuildBot