PX4/PX4-Autopilot #6867: yaw estimation lost

Stifael, 2017-03-21:
tested in sitl: 
rotating in yaw causes the estimation to break.
http://logs.px4.io/plot_app?log=132fcf5f-ab37-4d31-bbd6-d30cd9674b2b

priseborough, 2017-03-21:
There is evidence of a significant scale factor error in the integrated gyro data because the yaw innovation rapidly builds with spin in the opposite direction to rotation. This is consistent with dropped gyro samples or errors in time stamping. The build up in yaw error is over 90 degrees and causes a flyaway.

This is a plot of the sensor_combined time stamp (blue) and the accumulated gyro delta time. It clearly shows that the integrated time from the IMU sensor is not keeping up with autopilot time.

[image]

Which SITL model was being used (jmavsim or gazebo)? This indicates either a lack of lockstep scheduling and a simulator that is not running fast enough or an error in the simulated sensor code and/or sensor driver

I noticed that the mag fusion had been forced to use heading fusion through the whole flight rather than switching to 3-axis fusion after takeoff by setting EKF2_MAG_TYPE = 1 (0 is the default). I assume this was done to help with diagnosis.

priseborough, 2017-03-22:
@Stifael , I have been unable to replicate the same loss of yaw with my SITL testing using either gazebo or jmavsim. I was able to generate up to 11 deg of heading error with sustained rapid yaw with reversals, but not the 90 deg in the log attached to this issue.

Are you able to repeat the test on a faster computer and also try using jmavsim to see if that solves the issue.

Stifael, 2017-03-22:
@MaEtUgR can you please try as well on your machine?

MaEtUgR, 2017-03-22:
@Stifael I think both of our machines (T460s and T460p) are not adequate enough to run the performance hungry and wasteful programed gazebo simulation without any side effects.

That's really pitty because decent low performance gaming is possible at least on mine...

Stifael, 2017-03-23:
@ChristophTobler can you please test this on your machine, since you have a more powerful computer. I position control, just rotate around yaw for about 1min+, if nothing happens then change direction in yaw and repeat for about 1min+.

ChristophTobler, 2017-03-23:
I couldn't observe anything special. Here's the log http://logs.px4.io/plot_app?log=f4300632-7e4d-4de3-b469-0571735b6b82

mhkabir, 2017-03-23:
@Stifael I did the test, couldn't repro either; I'm on a pretty fast computer.

Stifael, 2017-03-23:
@ChristophTobler @mhkabir thanks for testing. Because it is not a firmware issue, I will close it.

mhkabir, 2017-03-23:
@Stifael @ChristophTobler I found something interesting. After yawing significantly, I experienced a flyaway in SITL. Managed to repro this twice.

Logs : http://logs.px4.io/plot_app?log=e222d6c8-d245-428e-8d6d-ea80b3a97f34

I flew it around a bit before starting the yaw tests, and it was fine then. But after yawing a bit, and commanding position changes, it just flies off.

Stifael, 2017-03-23:
Ok, thats what I experienced as well. The question still remains why some of us can do it and others not. I can reproduce it quite easily

priseborough, 2017-03-23:
Not much time scale / gyro scale error is required to break the estimator in this scenario. Lets say for example we rotate at 200 deg/sec continuously in yaw and there is a 3% scale factor error. That is equivalent to a step change of 6 deg/sec in bias which will break the estimator.

The lack of lockstep scheduling with the PX4 SITL implementation makes it absolutely critical that both the vehicle and flight computer simulation are both running at 100% real time and sensor data is not being dropped.

@mhkabir if you were using gazebo which is a cpu hog, try repeating your test with jmavsim. Your simulation logs shows that there is a large scale factor error in the integrated delta angles becasue yaw innovation starts to climb linearly as soon as the yaw rotation  starts.

[image]

I believe this to be caused by an underlying problem with the simulation architecture which is the lack of lockstep scheduling. We can probably work around this by changing the way that the angular rates are converted to delta angles when using SITL. I am not familiar with where/how that is currently done.

priseborough, 2017-03-23:
I am unable to fix this because it is not an estimator error.

Stifael, 2017-06-19:
I will close this issue for now. 
@hamishwillee I think it makes sense to add simulation limitation in the devguide

closed 2017-06-19 by Stifael

hamishwillee, 2017-06-19:
So I was planning to add a "you can't run this successfully in virtual environment note" anyway, but it sounds like this problem could also affect simulation on relatively "ordinary" hardware. 

1. Is this only going to be a problem due to estimator - or to put it otherwise, is it worth specifically mentioning this limitation or just commenting that real time performance is required, and in most cases you won't get this running the simulator in a VM or low spec hardware?
1. Can I have a guess at some sort of minimum spec that people could rely on for this to work?
1. Getting lock-step simulation sounds fairly essential going forward - is there an enhancement request for that?

priseborough, 2017-06-19:
See  https://github.com/PX4/Firmware/issues/3279 for further discussion.

This is looking more like a bug in the simulation SW.