PX4/PX4-Autopilot #6876: Takeoff overshoot in real world but not in the simulator

lovettchris, 2017-03-22:
When I ask the drone to take off to 5 meters it shoots aggressively up to 10 meters then comes back down:
[image]

[14-25-29-input.zip](https://github.com/PX4/Firmware/files/860207/14-25-29-input.zip)

Looking at the code in mc_pos_control, around line 1561 under the comment "special thrust setpoint generation for takeoff from ground" I see that it ramps up throttle all the way to MPC_THR_MAX if necessary in order to get the drone off the ground, and when it detects drone has reached MPC_TKO_SPEED speed then it switches to normal setpoint altitude hold.  The problem I see with this is that the GPS sensor is very slow, so it always cranks to MPC_THR_MAX for very agressive takeoff.  

I want to fix this, but I could use some pointers.  I don't want any new parameters, we already have too many, instead I would love the code to be self tuning, for example, if I strap something heavy on my drone, it should "just work".  I assume that the current behavior came about because you also do not want the drone to "hop" during takeoff because then it thinks it has landed again and disarms.  

But clearly from the picture above, PWM shot all the way to up max throttle (90% of 1950 is 1755) before it even detected drone leaving the ground, then because the acceleration was so great, even with PWM falling quickly to 1283, the drone was essentially flung up to 10 meters.    What is very interesting is the red line I drew on the PWM signal.  Drone is still ascending quickly, and PWM is going up??? from 1283 to 1443.  I realize you don't want the drone to freefall, but there must be something wrong there.

So I propose to add a takeoff that is more gentle, like landing.  Where it slowly spins up the props just to get some idea of what it takes to get any vertical acceleration.  For example, when I look at the charts, z-mag responded 1 full second before GPS noticed any change.  In fact, z-mag was seeing something when PWM was way down at 1100.  This slow ramp up ensures safety, if anyone is near they will back off before it shoots up into the air.  So from there it should ramp up quickly to HOVER_THROTTLE + some delta, where that delta is self tuning.  This sudden ramp will cause a hop, but to keep it in the air we need more than hover throttle, so it can start more aggressively to ensure no hops, but then it will write a private setting to calibrate this value so that the next take off will be smoother and smoother each time you try it.  This is a little bit like auto-tuning a PID algorithm I suppose.

Any objections or better ideas ?

Stifael, 2017-04-13:
That sounds reasonable. But how do you know hover thrust. As default it is set to 0.5, but the actual hover thrust can easily be below it. That is where the integrator comes on, which corrects for the wrong hover thrust. But you dont know that before actually being in air.

MartinSollie, 2017-04-16:
I think the problem with takeoff overshoot comes from the hover throttle feedforward added in #5529023 that is not taken into account in the takeoff code:
```C++
// copter has reached our takeoff speed. split the thrust setpoint up
// into an integral part and into a P part
_thrust_int(2) = _takeoff_thrust_sp - _params.vel_p(2) * fabsf(_vel(2));
```

Try something like this instead:
```C++
_thrust_int(2) = _takeoff_thrust_sp - _params.vel_p(2) * fabsf(_vel(2)) - _params.thr_hover;
```

lovettchris, 2017-04-16:
Thanks, yes, when I switch the simulator to use the same PID values as the drone then the simulator also overshoots.  I tried your suggested fix and the before and after look promising (this is recorded on Pixhawk v2 running in HIL simulation mode before and after the above change, clean reboot between each recording.

[image]

[takeoff.xlsx](https://github.com/PX4/Firmware/files/924622/takeoff.xlsx)

Do I need to put this into a pull request ?

LorenzMeier, 2017-04-16:
Would be good if you could put it into a PR, yes. That's how we get changes in. The core dev team doesn't have enough bandwidth to scavenge changes off discussions and there is a lot of room for ambiguity with respect to what you exactly changed and tested.

Thanks!

lovettchris, 2017-04-17:
[Done](https://github.com/PX4/Firmware/pull/7061).

LorenzMeier, 2017-04-30:
Thanks! We've further improved this on master.

closed 2017-04-30 by LorenzMeier