PX4/PX4-Autopilot #6877: Set distance_sensor from PX4FLOW OR Lidar

Ahrovan, 2017-03-22:
I can not set distance_sensor ref from PX4FLOW  or lidar on 1.6.0 Release RC1 (px4fmu-v2_default)
I khow distance_sensor is a topic and every publisher can publish on it. but how to select distance_sensor ref from PX4FLOW or lidar or so...

ecmnet, 2017-03-23:
LPE uses LIDAR automatically instead of sonar, if available. But you need to enable LIDAR in 'Sensor enable'

Ahrovan, 2017-03-23:
@ecmnet I khow we have several reference for alt source like: 1)baro, 2)sonar/lidar, 3)px4flow sonar, 4)Mavlink distance-sensor
in this case i want to use (4). but how to select ? is only way edit and create customized code ?
I check source code and see Topic distance_sensor use with px4flow and Mavlink distance-sensor, so if we use both of them they can not publish same topic simultaneous

Ahrovan, 2017-03-28:
@ecmnet have a way for configuration with Qgroundcontrol?

TSC21, 2017-07-25:
@Ahrovan is this solved to you?

Ahrovan, 2017-07-25:
No.

TSC21, 2017-07-25:
Check if this helps you: https://github.com/PX4/Firmware/blob/28017105f456e27034771fa3a7f3befb1b4a1a34/src/modules/local_position_estimator/BlockLocalPositionEstimator.cpp#L228-L252

Ahrovan, 2017-07-26:
OK maybe. I will check Again

closed 2018-02-21 by PX4BuildBot