PX4/PX4-Autopilot #6882: `make px4fmu_v2_lpe upload` causes issues during radio calibration

vinhk, 2017-03-24:
Hi Everyone,

I am working on testing LPE using Optical flow but after I flashed the masters using `make px4fmu_v2_lpe upload`, I can not get pass radio calibration step in QGC. It gives me `Detected 0 radio channels. To operate PX4, you need at least 5 channels`.

davids5, 2017-03-24:
Please see https://github.com/PX4/Firmware/issues/6762#issuecomment-287955542

It may be the same issue. Please try and revert those commits and let me know if it resolves the issue for you.

vinhk, 2017-03-24:
@davids5 The thing is I wanted to test position hold using LPE. By reverting it, do you mean just flash using the default master firmware from QGC?

davids5, 2017-03-24:
@vinhk 

no. 

The title "make px4fmu_v2_lpe" led me to believe you are building the code. Correct?

If so the suggestion I made is to to do:

```
git revert --no-commit 63857c6
git revert --no-commit a716b10
git revert --no-commit b8e7278
```

Then run ```make px4fmu_v2_lpe upload" and see if the problem goes away.

Let me know

vinhk, 2017-03-24:
Thanks. I am hoping to get rid of radio calibration. As now I am having that issue which is preventing me from using the pixhawk.

vinhk, 2017-03-24:
I successfully build the code and upload but then the radio calibration in QGC prevents me from going further.

davids5, 2017-03-24:
"The code"

master ?
Or 
master with the revert commands ?

vinhk, 2017-03-24:
The master. Based on your suggestion I did the following: 
`
mkdir -p ~/src
cd ~/src
git clone https://github.com/PX4/Firmware.git
cd Firmware
git revert --no-commit 63857c6
git revert --no-commit a716b10
git revert --no-commit b8e7278

make px4fmu-v2_lpe upload`

davids5, 2017-03-24:
Ok that test separates this from the from the V4  #6762 (comment) issue.

Please do a ```git reset --hard origin/master```

Then ```make px4fmu-v2_default upload``` 

Does that work?

If NOT this may be a HW issue.

vinhk, 2017-03-24:
Is there a way to reset hw?

davids5, 2017-03-24:
I assume from your response the ```px4fmu-v2_default`` build on master did not fix the radio issue 

Is that correct?

The FMU is reset on power cycle.

If you list the model number for your Transmitter and Receiver @kd0aij  may be able to help it.

Also  check your cabling with ohm meter or a scope.

vinhk, 2017-03-24:
that's correct. I still have that issue with the QGC.

davids5, 2017-03-24:
If it is only the RC then it is a HW problem.

vinhk, 2017-03-24:
@kd0aij I am using Spektrum DX6 and Spektrum AR7700 receiver.

vinhk, 2017-03-24:
I was able to flash it a day before. Now Radio calibration never work.

yidengfeiwu, 2017-03-27:
i got the same message "Detected 0 radio channels" after i upload the firmware.

LorenzMeier, 2017-03-27:
@PX4TestFlights @davids5 Can you please look into this today?

yidengfeiwu, 2017-03-27:
this video help me,now i can calibrate my radio,although i still don't know why

https://www.youtube.com/watch?v=4lRFwVN1vFY

santiago3dr, 2017-03-27:
today's master (v3) does not recognize rc input
tried forcing i/o boot by holding down the safety switch to no avail

using pixhawk mini

davids5, 2017-03-27:
@vinhk @yidengfeiwu 

@santiago3dr - has done some great detective work and has found a correlation to the version of the compiler.

The build servers and dev team should all be using:
```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
Copyright (C) 2015 Free Software Foundation, Inc.
```

Please do a ```arm-none-eabi-gcc --version``` and post you finding here.

If you are not using 5.4.1 please see https://dev.px4.io/starting-installing-linux-boutique.html#toolchain-installation and install it and retest. 

I have built and tested:
px4fmu-v3_default and px4fmu-v2_lpe and calibration works fine.

```
ver all
HW arch: PX4FMU_V2
FW git-hash: 79d15ad2f04eb83856fe4bb1b95dcdd49520f7b0
FW version: 1.6.0 0 (17170432)
OS: NuttX
OS version: Release 1.8.0 (17301759)
Build datetime: Mar 27 2017 13:02:04
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MFGUID: 323833363335511000400038
MCU: STM32F42x, rev. 3
UID: 400038:33355110:32383336
```

vinhk, 2017-03-28:
@davids5 
I did 

arm-none-eabi-gcc --version

and I got 

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I troubleshoot the wiring for RC using a meter for connection issues and there seems to be no short circuit or lose connection. However, the FMU PWR and Io PWR LEDs are always orange. Previously they were green. After calibrating the sensors, the main LED smothly flashes blue. When I try to calibrate the RC it flashes gren and doesn't allow me to calibrate it. I tried to use a mucro SD card of a working Pixhawk and there is no change. After re uploading the firmware using make px4fmu-v2_default, I tried to reboot it using nsh and I got the following message. RC still not recognized by qgroundcontrol.

NuttShell (NSH)
nsh> INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 21, OBS buf: 16)
WARN  [commander] Not ready to fly: Sensors not set up correctly

�[boot] Fault Log info File No 3 Length 3257 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
WARN  [px4io] check CRC failed: -22
INFO  [px4io] default PWM output device
[PX4IO] using firmware from /etc/extras/px4io-v2.bin
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bootloader not responding
WARN  [px4io] check CRC failed: -22
MS5611_SPI on SPI bus 4 at 2 (20000 KHz)
WARN  [ms5611] no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V2
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 2 (type: 1)
WARN  [mpu6000] no device on bus 5
WARN  [mpu9250] no device on bus 4
MPU6000 on SPI bus 1 at 4 (1000 KHz)
WARN  [mpu6000] no device on bus 5
WARN  [mpu9250] no device on bus 3
WARN  [mpu9250] no device on bus 4
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
INFO  [load_mon] stack check enabled
ERROR [param] Parameter UAVCAN_ENABLE not found
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
ERROR [param] Parameter UAVCAN_ENABLE not found
INFO  [ver] match: PX4FMU_V2
px4flow [207:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
ERROR [mixer] can't open /dev/pwm_output0

ERROR [mixer] failed to load mixer
ERROR  [init] Failed loading mixer: /etc/mixers/quad_x.main.mix 

NuttShell (NSH)
nsh> INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 21, OBS buf: 16)
WARN  [commander] Not ready to fly: Sensors not set up correctly

davids5, 2017-03-28:
@vinhk 

You may have bad FW on the IO.

Let's force bootload mode on it*.

1) If using one Unplug the power module. 
2) disconnect the USB
3) Hold down the safety switch and plug in USB.
4) Release safety.

(*This can also be done with using the reset switch)

Did the IO LED recover?

vinhk, 2017-03-28:
@davids5 Thanks a lot. It is working now

vinhk, 2017-03-28:
@davids5 the IO LED is still orange instead of green. But at least the RC calibration is working.

closed 2017-03-28 by vinhk

HockyYoung, 2017-04-07:
I also encountered the same problem, RC can not identify, but I forced brush PX4IO BootLoader, with the previous firmware can work with the latest github code is not recognized, what measures can be resolved?

davids5, 2017-04-07:
@HockyYoung Please test with today's [master FW](https://github.com/PX4/Firmware/issues/6999#issuecomment-292573103)

See https://github.com/PX4/Firmware/pull/6993#issuecomment-292593202 to know if the PX4IO is running OK

HockyYoung, 2017-04-08:
I have used the previous firmware can be radio calibration, but the use of the latest firmware, will lead to IO BootLoader no response, this back with the previous firmware, remote calibration also problems, so I restart io BootLoader, radio calibration is ok.And I can judge the current lastest firmware will damage IO, I currently use the Pixhawk, arm-none-eabi-gcc version is 5.4.1.can you tell me what the next step?