PX4/PX4-Autopilot #6887: Snapdragon Spektrum RC not working

ChristophTobler, 2017-03-24:
Once I turn on the Spektrum RC I get the following:
```
pxh> ERROR [muorb] ERROR: Getting Bulk data from fastRPC link
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [telemetry_status] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [telemetry_status] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [telemetry_status] topic data over comm_channel
```

and the mini-dm output is:
```
[08500/02]  42:20.945  mmpm_register: MMPM client for USM ADSP core 12  0117  UltrasoundStreamMgr_Mmpm.cpp
[08500/02]  42:20.945  ADSP License DB: License validation function with id 164678 stored.  0280  adsp_license_db.cpp
[08500/02]  42:20.945  AvsCoreSvc: StartSvcHandler Enter  0518  AdspCoreSvc.cpp
[08500/02]  42:20.945  AdspCoreSvc: Started successfully  0534  AdspCoreSvc.cpp
[08500/02]  42:20.950  loading BLSP configuration  0217  blsp_config.c
[08500/02]  42:21.192  HAP:159:unable to open the specified file path  0167  file.c
[08500/04]  42:21.192  failed to open /usr/share/data/adsp/blsp.config  0233  blsp_config.c
[08500/00]  42:21.192  loading the blsp config file failed, loading the default config  0263  main.c
[08500/02]  42:21.192  Loaded default UART-BAM mapping  0036  blsp_config.c
[08500/02]  42:21.192  UART tty-1: BAM-9  0044  blsp_config.c
[08500/02]  42:21.192  UART tty-2: BAM-6  0044  blsp_config.c
[08500/02]  42:21.192  UART tty-3: BAM-8  0044  blsp_config.c
[08500/02]  42:21.192  UART tty-4: BAM-2  0044  blsp_config.c
[08500/02]  42:21.192  UART tty-5: BAM N/A  0049  blsp_config.c
[08500/02]  42:21.192  UART tty-6: BAM N/A  0049  blsp_config.c
[08500/02]  42:21.192  HAP:110:cannot find /oemconfig.so  0141  load.c
[08500/03]  42:21.192  HAP:111::error: -1: 0 == dynconfig_init(&conf, "security")   0696  sigverify.c
[08500/02]  42:21.193  HAP:79:cannot find /voiceproc_tx.so  0141  load.c
[08500/02]  42:21.193  HAP:79:cannot find /voiceproc_rx.so  0141  load.c
```

@julianoes @jywilson any ideas why that is?

julianoes, 2017-03-24:
Looks like the DSP crashes.

I would comment out everything in the Spektrum driver, try again. If that works, you add in about half of the code and you do a binary search until you have the line that causes the crash.

LorenzMeier, 2017-03-26:
@ChristophTobler @julianoes Can you look into that on Monday?

ChristophTobler, 2017-03-27:
@julianoes The error occurs as soon as the topic gets advertised. https://github.com/PX4/Firmware/blob/d1372ddb6728c2f61fbd3f26be8dae477424a643/src/drivers/spektrum_rc/spektrum_rc.cpp#L130

julianoes, 2017-03-27:
@ChristophTobler Interesting. Do you think the advertisement fails or it crashes because of the fact that rc_input is published and is used somewhere?

ChristophTobler, 2017-03-27:
@julianoes I think it's the advertisement, because it happens even if I don't publish (`orb_publish()`). And it just happens if `rc_lost` is false https://github.com/PX4/Firmware/blob/d1372ddb6728c2f61fbd3f26be8dae477424a643/src/drivers/spektrum_rc/spektrum_rc.cpp#L188

LorenzMeier, 2017-03-27:
@ChristophTobler Advertising does also publish. If rc_lost is false then it sounds like some other code part crashes which reads the rc_lost flag.

Can you publish rc_lost = true and see if that works?

ChristophTobler, 2017-03-27:
@LorenzMeier Ah, ok. yes, there is no error when I set it to `true`

ChristophTobler, 2017-03-27:
So without this https://github.com/PX4/Firmware/blob/11484b0d51876d3d4404520440cd88995a23d57d/src/modules/sensors/rc_update.cpp#L445-L446, I don't get the error and warnings. If I replace `orb_subscribe_multi(ORB_ID(telemetry_status)...` and `orb_subscribe(ORB_ID(manual_control_setpoint))` with `0`, I don't get the warnings but still get the error (see first comment)

julianoes, 2017-03-27:
> If I replace orb_subscribe_multi(ORB_ID(telemetry_status)... and orb_subscribe(ORB_ID(manual_control_setpoint)) with 0, I don't get the warnings but still get the error (see first comment). @julianoes So I guess it must be the publishing?

I don't understand what you mean. It seems like something is triggered and crashes as soon as you have RC, right?

ChristophTobler, 2017-03-27:
@julianoes by replacing them with 0, I make sure it's because of publishing the the data and not some other code using it.

jywilson, 2017-03-27:
+@julianoes 

@ChristophTobler,

@wangxdflight has been working on some fixes to the Spektrum RC driver, but has not had a chance to upstream them yet.  I don't know if this will help, but I have asked him to push his changes into a "WIP" PR. I would recommend that you look at these changes to see if it relates to your issues. 

I will post the PR link to this thread once he creates it.

Jim W.

jywilson, 2017-03-28:
@ChristophTobler 

This PR contains a fix removing the hard coded channel assignment in the Spektrum driver.  This may not be related to your problem, on second thought, but this could still be helpful.

https://github.com/PX4/Firmware/pull/6906

Jim W.

ChristophTobler, 2017-03-28:
@jywilson Thanks! I'll have a look at it.

ChristophTobler, 2017-03-28:
@jywilson https://github.com/PX4/Firmware/pull/6906 does not solve the issue, unfortunately.

ChristophTobler, 2017-03-28:
This issue has magically disappeared by changing https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L716 to `10000`, build, and then change back...

closed 2017-03-28 by ChristophTobler

jywilson, 2017-03-28:
@ChristophTobler Not sure I understand.  You changed the value for the build, but did not retain the change?

That parameter controls the stack size, so its possible there is an intermittent issue.

Can you provide a branch that I can use to test with my vehicle?

ChristophTobler, 2017-03-28:
@jywilson now it's the master branch. So I had the error when connected to the Spektrum RC. Then we changed https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L716 to `10000`, uploaded it to the Snapdragon, ran PX4 and there was no error. Then I tried lowering the value again and checked if it still works. In the end, it was even working with a value of `1000`. So now it works normally without any change

julianoes, 2017-03-29:
@jywilson I can confirm this.

We debugged that line where the `orb_subscribe_auto` (https://github.com/PX4/Firmware/blob/master/src/modules/sensors/rc_update.cpp#L445-L446) lead to a crash and what we found is that it mattered if `orb_subscribe_auto` was compiled but it didn't matter if it was actually called. E.g. we put a long sleep before `orb_subscribe_auto` and it would crash immediately before the sleep. Or we would set some condition that would prevent it from being called, and it still crashed.

This lead us to believe that the issue could be connected to the amount of stack that is used/reserved.

The odd behaviour about changing it and changing it back is however very peculiar.