PX4/PX4-Autopilot #6895: sdlog bug?

lovettchris, 2017-03-25:
I'm reviewing this code because I want to add logging of actual barometric pressure, but can someone explain how this works?

[sdlog2.c, line 1529-1538](https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L1529):
```

			switch (log_type) {
				case LOG_TYPE_ALL:
					if (fds[0].revents & POLLIN) {
						orb_copy(ORB_ID(sensor_combined), subs.sensor_sub, &buf.sensor);
					}

					if (fds[1].revents & POLLIN) {
						orb_copy(ORB_ID(ekf2_replay), subs.replay_sub, &buf.replay);
					}
					break;
```

The "buf" variable is a "union" of structs. This code is copying sensor_compbined, then also copying efk2_reply into the sab buf, which is a union, so surely that leaves the sensor_combined data corrupted?  So should this be "if () { } else if... " or does it need to store the efk2 buffer elsewhere?

dagar, 2017-03-25:
This doesn't answer your question, but I'd recommend you give the new logger a try. It directly logs selected uorb messages, so it's quite easy to extend. https://dev.px4.io/advanced-logging.html

lovettchris, 2017-03-25:
Actually, that solves all my  problems.  Thanks!

closed 2017-03-25 by lovettchris