PX4/PX4-Autopilot #6898: pixhawk 2.1 mag calibration failure with external

dagar, 2017-03-26:
http://discuss.px4.io/t/pixhawk-2-mag-calibration-failure-with-here-gps-module/2869

Start the individual calibration steps by clicking one of the buttons to the left.

    [cal] calibration started: 2 mag
    [cal] Skipped scale calibration, sensor 1
    [cal] Skipped scale calibration, sensor 2
    [cal] Mag 3 no device id, abort
    [cal] calibration failed: mag

If I remove the module, then I can calibrate the compass without issues.

dagar, 2017-03-26:
Either need to update PX4 to handle 4 mags, or realistically we probably don't need 3 internal mags.

tubeme, 2017-03-26:
Is this the X2.1??

dagar, 2017-03-26:
Yes

LorenzMeier, 2017-04-24:
Fixed in https://github.com/PX4/Firmware/pull/7080

closed 2017-04-24 by LorenzMeier