PX4/PX4-Autopilot #6899: Why do not use the mavlink module but add a simulator module in SITL loop?

yajingwang94, 2017-03-27:
Hi,

I am doing some` SITL` and `HIL` experiments in `gazebo`. When I learn the inner operating mechanisms of the `SITL` loop, I find the `mavlink` module dose nothing related with the connection between gazebo and the px4 package, but the `simulator` module dose , which receives the vehicle status from gazebo and send back the control commands using socket with TCP/IP protocols.

I tried to figure out the difference among `SITL` , `HIL`, and the onboard loop(which refers to when pixhak FMU is mounted on a` fw` or `mc` plane), finding that it is the `mavlink` module who bears the task of communication with ROS node and QGC in HIL and onboard situations. 

I wonder if there is something I misunderstand, and why do not use the `mavlink` module in the `SITL` loop as well? 

Thanks.

bkueng, 2017-03-27:
Hi @yajingwang94 

Your analysis is correct. The reason for the separation is that the `simulator` does some additional things like wait for gazebo on startup, time accounting, sensor simulation and in future maybe lock-stepping. But I agree it would be nicer if the 2 were combined. If you want to work on that you're welcome to do so.

yajingwang94, 2017-03-28:
I see. Thanks for the reply.

yajingwang94, 2017-03-28:
@bkueng I see. Thanks so much for your reply.

bkueng, 2017-03-28:
You're welcome - I'm closing this then

closed 2017-03-28 by bkueng