PX4/PX4-Autopilot #6900: Bad LPSP.Z data leads to crash

JohnSnowball, 2017-03-27:
I suffered a crash with my 450 X8,  I use EKF2 and a range finder, with EKF2_HGT_SOURCE = rng_finder.
My copter works fine in MANUAL mode, but when I switch to ALTITUDE mode, it goes crash.
I checked the log and found that it has extremly bad LPSP.Z data during ATL mode, as shown in picture, the units is meter. The RC input is OK, everything else seems to be OK. 

[image]

(Just one day before the crash, everything is just fine! I tested MANUAL/ATL/POS HOLD modes, and tested for about half an hour. And after the crash, I can't make that bug occur again!!!)
The log link is here http://logs.uaventure.com/view/V5NVorsJhhikyBoHPABSd#Local_Z_PLOT

tumbili, 2017-03-27:
@MaEtUgR @Stifael @priseborough I think the problem is related to this function here
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L837

When the ekf is not able to initialize it's origin (in this case due to gps not being available) then it will always reset the reference timestamp here
https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_main.cpp#L833

Depending on the altitude reported by the gps the altitude setpoint can be shifted by insane values.
I think this function needs reconsideration and it needs to play in harmony with the ekf reset logic.
I already mentioned this problem in the slack control channel.

Stifael, 2017-03-27:
This update ref function is indeed something I like to change/move. Please see here for further discussion
https://github.com/PX4/Firmware/pull/6459

JohnSnowball, 2017-03-28:
[image]
true story! thanks! I will comment the update_ref function for the moment until you are sure you fixed this bug. Thanks!

mzahana, 2017-03-28:
I am facing the same EXACT issue! I wull try to uoload some logs later!

Stifael, 2017-04-28:
is this still present? the origin initialization changed now in the ekf

closed 2018-02-21 by PX4BuildBot