PX4/PX4-Autopilot #6902: Pixracer: MANUAL CONTROL LOST on param save

bkueng, 2017-03-27:
When landing, the flight duration param is set, leading to a param save. On pixracer, this causes a manual control loss and failsafe mode switch. The following shows the order of events (timestamps in ms), on #6901 branch:
```
nsh> WARN  [land_detector] param land set 24929
WARN  [land_detector] param land set 24929 done
INFO  [commander] Landing detected
WARN  [modules__systemlib] entering param_save_default 25229 
WARN  [commander] MANUAL CONTROL LOST (at t=25739ms)
INFO  [commander_tests] no RC
WARN  [commander] failsafe mode on
INFO  [commander] Takeoff detected
WARN  [modules__systemlib] param_save_default done 26424
WARN  [commander] failsafe mode off
```

And this shows the sampling is thrown off during param save (timestamps do not match the above output):
[image]

Happens both on master (8810f70fdcf036b with nuttx semaphore patch) and with https://github.com/PX4/Firmware/pull/6901

This is not a prio inversion/inheritance problem, all task priorities stay the same. The problem is that we disable interrupts during save, apparently for too long, throwing off RC and sensor sampling, done here: https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param.c#L922. If I comment out this lock, the problem disappears.

@davids5 can you look into that?

davids5, 2017-04-13:
@bkueng https://github.com/PX4/Firmware/pull/7036 - would you please test as suggested.

LorenzMeier, 2017-06-04:
Closing as the fix was merged.

closed 2017-06-04 by LorenzMeier