PX4/PX4-Autopilot #6905: Level horizon from QGC doesn't survive reboot

sanderux, 2017-03-27:
When leveling horizon the new offsets get applied but return to previous state after a reboot. 
(aka no param write)

@LorenzMeier do you know who to assign?

bkueng, 2017-03-28:
@sanderux I just tested level calibration with QGC several times with pixracer, and it stays persistent every time. Is it reproducible for you? Which HW? And do you have the same problem on the #6901 branch?

sanderux, 2017-03-28:
Its on a dropix (fmuv2)
it's consistent over multiple FC's
I'm now working on a more critical issue https://github.com/PX4/Firmware/issues/6911
after that i'll try your suggestion

sanderux, 2017-03-28:
@bkueng the issue seems already resolved on master, i could only reproduce it in the branch we were flying (somewhere jan)

bkueng, 2017-03-28:
Ok, I'm closing it then.

closed 2017-03-28 by bkueng