PX4/PX4-Autopilot #6909: Confusions about topics in gazebo_mavlink_interface.cpp.

yajingwang94, 2017-03-28:
Hi, 

I find in the `gazebo_mavlink_interface.cpp` that the `control information` has two sources:
-1.  from `udp_port 14560 `(which is bind with the `socket` of `simulator` module) ;
-2.  from topic `/HilControl` ;

When I searched for the information of `/HilControl` topic, I failed to find its publisher:
[image]

What's more, I find the `gazebo_mavlink_interface` also subscribe to several topics: `/imu`, `/lidar`, `/opticalFlow.`  Do they have someone who already publishes them? Did I miss anything important? 

And, there are "hil" in names of these topics, do they have something to do with hardware in the loop simulation?

Thanks for the time.

dagar, 2017-04-12:
@ziyangli did you figure this out?

TSC21, 2017-07-25:
Though SITL uses HIL msgs, those are not related with using HIL.

closed 2017-07-25 by TSC21