PX4/PX4-Autopilot #6911: VTOL pusher motor doesnt start on current master

sanderux, 2017-03-28:
The ESC beeps indicate no signal to the pusher motor.
I am currently bisecting

sanderux, 2017-03-28:
@kd0aij i bisected this issue to your oneshot commits
https://github.com/PX4/Firmware/commit/aa9fbbedd54a7a0e346ea03f439dcdf4f90c6113
https://github.com/PX4/Firmware/commit/2d5588ae02560aae029844410212aab38785d671

vtol pusher motor is on aux ports, please fix this

LorenzMeier, 2017-03-28:
@davids5 Can you chime in and have a look?

sanderux, 2017-03-28:
@LorenzMeier any way to revert these? it is impeding our ability to go out and test fly

sanderux, 2017-03-28:
@davids5 
regular pixhawk, autostart 13006 (standard delta vtol)
pusher on aux3

davids5, 2017-03-28:
@sanderux 

Also what HW is this running on and what make command (px4fmu-v4_default etc ) is used?

I just found a bug where if the fmu is stopped and started the publications were duplicated.

please list the results of ```ls /obj``` on the nsh console here if you would.

sanderux, 2017-03-28:
hw is dropix
make is px4fmu-v2_default

```
nsh> ls /obj
/obj:
 _obj_
 actuator_armed0
 actuator_controls_00
 actuator_controls_10
 actuator_controls_20
 actuator_controls_30
 actuator_controls_virtual_fw0
 actuator_controls_virtual_mc0
 actuator_outputs0
 adc_report0
 airspeed0
 battery_status0
 commander_state0
 control_state0
 cpuload0
 differential_pressure0
 distance_sensor0
 ekf2_innovations0
 ekf2_timestamps0
 estimator_status0
 fw_pos_ctrl_status0
 fw_virtual_attitude_setpoint0
 fw_virtual_rates_setpoint0
 geofence_result0
 gps_inject_data0
 home_position0
 input_rc0
 led_control0
 log_message0
 manual_control_setpoint0
 mavlink_log0
 mc_att_ctrl_status0
 mc_virtual_attitude_setpoint0
 mc_virtual_rates_setpoint0
 mission_result0
 multirotor_motor_limits0
 offboard_control_mode0
 offboard_mission0
 onboard_mission0
 optical_flow0
 parameter_update0
 position_setpoint_triplet0
 rc_parameter_map0
 safety0
 sensor_accel0
 sensor_accel1
 sensor_baro0
 sensor_combined0
 sensor_correction0
 sensor_gyro0
 sensor_gyro1
 sensor_mag0
 sensor_mag1
 sensor_preflight0
 servorail_status0
 subsystem_info0
 system_power0
 task_stack_info0
 tecs_status0
 telemetry_status0
 uavcan_parameter_value0
 vehicle_attitude0
 vehicle_attitude_setpoint0
 vehicle_command0
 vehicle_command_ack0
 vehicle_control_mode0
 vehicle_global_position0
 vehicle_global_velocity_setpoint0
 vehicle_gps_position0
 vehicle_land_detected0
 vehicle_local_position0
 vehicle_local_position_setpoint0
 vehicle_rates_setpoint0
 vehicle_status0
 vehicle_status_flags0
 vehicle_vision_attitude0
 vehicle_vision_position0
 vtol_vehicle_status0
 wind_estimate0
```

davids5, 2017-03-28:
@sanderux - thank you 

Let me see what I can find.

kd0aij, 2017-03-28:
@sanderux Sorry this got broken

@davids5 Were the subsequent changes to my oneshot PR tested before merging? My last flight test was on fmu-v4 here: https://github.com/PX4/Firmware/pull/6616#issue-208740733 and both FMU and IO outputs were working.

davids5, 2017-03-28:
@kd0aij - not to that logic per say - I did question that change a few times for the IO but not for the FMU. I  wonder if a better fix would be to ensure there is a "once" then use the n_updates like you had. Would you please have a close look at it?

sanderux, 2017-03-28:
Has been fixed, closing this issue.
@kd0aij if you find anything worthy of change can you open a new issue?

closed 2017-03-28 by sanderux