PX4/PX4-Autopilot #6913: Parameter reference page not updating

dagar, 2017-03-28:
https://pixhawk.org/firmware/parameters

dagar, 2017-04-02:
Should this be moved to the dev guide? Auto generated and pushed to the gitbook repo, or is sync still problematic?

bkueng, 2017-04-04:
That would be preferable so everything is in one place. Either by automatic sync or by including (`iframe`) external content. I think @hamishwillee intended to look into that.

hamishwillee, 2017-04-04:
In my opinion yes it should be moved to the devguide. How is this built now? What are the existing sync issues?

bkueng, 2017-04-04:
There's a script (`python ./Tools/px_process_params.py -s src --wiki`) that generates docuwiki output in the form:
```
====== Parameter Reference ======
<note>**This list is auto-generated from the source code** and contains the most recent parameter documentation.</note>

==== Attitude EKF estimator ====

|< 100% 25% 45% 10% 10% 10% >|
^ Name ^ Description ^  Min ^  Max ^  Default ^
^ :::  ^ Comment ^^^^
| **EKF_ATT_V3_Q0** | Body angular rate process noise |   |   |  1e-4 |
| ::: | <div></div> ||||
| **EKF_ATT_V3_Q1** | Body angular acceleration process noise |   |   |  0.08 |
| ::: | <div></div> ||||
| **EKF_ATT_V3_Q2** | Acceleration process noise |   |   |  0.009 |
| ::: | <div></div> ||||
| **EKF_ATT_V3_Q3** | Magnet field vector process noise |   |   |  0.005 |
| ::: | <div></div> ||||
| **EKF_ATT_V4_R0** | Gyro measurement noise |   |   |  0.0008 |
| ::: | <div></div> ||||
| **EKF_ATT_V4_R1** | Accel measurement noise |   |   |  10000.0 |
| ::: | <div></div> ||||
| **EKF_ATT_V4_R2** | Mag measurement noise |   |   |  100.0 |
| ::: | <div></div> ||||
| **ATT_J11** | Moment of inertia matrix diagonal entry (1, 1) |   |   |  0.0018 |
| ::: | <div></div> ||||
| **ATT_J22** | Moment of inertia matrix diagonal entry (2, 2) |   |   |  0.0018 |
| ::: | <div></div> ||||
| **ATT_J33** | Moment of inertia matrix diagonal entry (3, 3) |   |   |  0.0037 |
| ::: | <div></div> ||||
| **ATT_J_EN** | Moment of inertia enabled in estimator |   |   |  0 |
| ::: | <div>If set to != 0 the moment of inertia will be used in the estimator</div> ||||
```

In the recent past we had issues with syncing between the github dev-guide repo and the gitbook internal repo. We have completely reset the gitbook repo, and it seems to work now, we will have to see how stable it is.

hamishwillee, 2017-04-05:
Thanks! 
@dagar @bkueng I've created an output format for markdown #6975 , which you can preview [here](https://hamishwillee.gitbooks.io/px4-user-guide/content/en/orphans/parameters.html) (for a little while). The output format is actually HTML since markdown is quite limited in its table handling. I also moved comments inside the "name" row, because the [old format](https://pixhawk.org/firmware/parameters) was a bit ugly. 

- Note that my gitbook has a floating right in-page table of contents that allows easy navigation of the list of parameters (and any other headings). Any reason not to add the same plugin to devguide?
- I also made each of the parameters an id="parameter_code", so you will be able to link to any parameter - e.g. https://hamishwillee.gitbooks.io/px4-user-guide/content/en/orphans/parameters.html#ATT_W_GYRO_BIAS

Is this OK? Any thoughts on improving?

To get this included in a build I'd make my build machine automatically run the  parameter extraction tools when triggered, update the devguide git tree and copy in the new file, merge the new file. Let gitbook handle the rebuild their end. That requires some playing with scripts on the build machine.

@dagar Any better way to handle this, and if not, can you sort that part out, since I don't have access ....

hamishwillee, 2017-04-05:
p.s. @bkueng  - where would be the best place in devguide TOC for the parameters to live? Perhaps near https://dev.px4.io/advanced-configurations.html

dagar, 2017-04-05:
Nice! The table of contents on the old page was helpful, but we can live without it if it's not easy.

The param metadata has expanded quite a bit since the old wiki. In particular the values are now quite important, and the reason why many params no longer have a min/max. https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_params.c#L108

On a somewhat related note I believe @LorenzMeier also started doing something similar with airframe metadata. We should look into displaying that as well. https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/13013_deltaquad#L6

hamishwillee, 2017-04-05:
The TOC on the page is easy and desirable (IMO). I've added request for its addition: https://github.com/PX4/Devguide/pull/109

> The param metadata has expanded quite a bit since the old wiki. In particular the values are now quite important, and the reason why many params no longer have a min/max. https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_params.c#L108

Great. Is that just for my information, or is there something I need to "do"?

> On a somewhat related note I believe @LorenzMeier also started doing something similar with airframe metadata. We should look into displaying that as well. 

Lets wait on Lorenz for information on how far he got/intentions/what needs to happen.

Have you any thoughts regarding the last part - automating the build process?

dagar, 2017-04-05:
Values are another nice to have if it looks easy to incorporate, but we can live without. 

Do you have any access to the automated build for the hosted gitbook? One option is we could generate the output in the PX4 Firmware build and upload it to S3. Every time the gitbook updates it could download a copy. Another option is using jenkins (http://sitl01.dronetest.io/) to generate a commit.

hamishwillee, 2017-04-05:
> Values are another nice to have if it looks easy to incorporate, but we can live without.

Not "super" easy. The big problem here is that within a group there are parameters that have values, parameters that have min/max, and parameters that are boolean (others)? We need to work out how to best present that and collect it in a way that we can sensibly render (we could just have columns for values as well, but that would be pretty messy).

This is definitely worth doing though! What are all the files in which parameters are defined?

> Do you have any access to the automated build for the hosted gitbook? One option is we could generate the output in the PX4 Firmware build and upload it to S3. Every time the gitbook updates it [...]

I thought this was just auto-generated by gitbook(.com) using a webhook from submissions to the github repo? Can't really comment without knowing how it is hosted and generated. That said, I'd be tempted to use jenkins to generate a commit in Devguide of the parameters when needed.

bkueng, 2017-04-05:
Thanks! One more thing we can do is display the unit of each parameter (if it has any). And if we find a good way, the parameter values & booleans (I don't think we have more types).

> What are all the files in which parameters are defined?

They are pretty much distributed within the source tree.

> Perhaps near https://dev.px4.io/advanced-configurations.html

Yes, makes sense.

For the airframes, there's already an xml generated and pushed to S3 based on latest master: http://px4-travis.s3.amazonaws.com/Firmware/master/airframes.xml
There's a script in `Tools/px_process_airframes.py`. Should be quite easy to turn this into HTML. The fancy version would be to have schematic pictures of the type and outputs (maybe overlay the text to generic pictures or generate the images with ImageMagick, ...). I talked to @LorenzMeier and it's what he has in mind as well.

> That said, I'd be tempted to use jenkins to generate a commit in Devguide of the parameters when needed.

I agree, it should be completely automated. Using jenkins sounds like a good way to do it.

hamishwillee, 2017-04-05:
Hi Beat,

1.  So is it correct that the Gitbook is just build from our Devguide github repo (triggered via a webhook?)

>>What are all the files in which parameters are defined?
>They are pretty much distributed within the source tree.

I guess what I actually meant is "How can I find all the files that we need to process". Looking a bit deeper I think that we already scan for all the required information  in [srcparser.py)](https://github.com/PX4/Firmware/blob/master/Tools/px4params/srcparser.py) so it is just a matter of reformatting/presenting.
Is that correct/your understanding?

> For the airframes, there's already an xml generated and pushed to S3 based on latest master: http://px4-travis.s3.amazonaws.com/Firmware/master/airframes.xml
> There's a script in [...]

OK, let us know. 

> That said, I'd be tempted to use jenkins to generate a commit in Devguide of the parameters when needed. I agree, it should be completely automated. Using jenkins sounds like a good way to do it.

@dagar Can you do the work needed?

bkueng, 2017-04-05:
> So is it correct that the Gitbook is just build from our Devguide github repo (triggered via a webhook?)

No, gitbook has a separate git repo copy, and the webooks cause a sync of the repos - which then causes a rebuild & publication.

> Is that correct/your understanding?

Ah ok. Yes, exactly. You can look at the generated xml (http://px4-travis.s3.amazonaws.com/Firmware/master/parameters.xml), which contains all the extracted information.

hamishwillee, 2017-04-06:
@dagar @bkueng After reviewing all the parameters I've come up with this [new version](https://hamishwillee.gitbooks.io/px4-user-guide/content/en/orphans/parameters.html). This adds 
* "type" information below the code
* 2 new columns for increment and units
* Enum/values, decimal, bitmasks, reboot required. are displayed in description with "mini heading" IFF they exist.
* Scope is displayed at the TOP of the table if it is the same for all parameters in the table. Otherwise it is displayed in the description in the same way as bitmasks etc.
* Board is NOT displayed - there were no parameters that set this.

Questions:
1. What is decimal? Is it important/should it be displayed.
2. Is this OK/better, suggestions for improvement?

If you are OK with this then I will submit a PR with the changes. This is used by calling
```python ./Tools/px_process_params.py -s src --markdown```

@dagar - Is there anything you need to make the automatic build into gitbook happen, and can you ETA?

LorenzMeier, 2017-04-06:
Hi - on airframe meta data: The goal is to generate a picture for every autopilot which marks the outputs you should be connecting - e.g. MAIN1 - left aileron, MAIN2 - right aileron, etc. We have that meta data already in for a number of airframes and it would allow to generate automatic wiring diagrams. These should also include airspeed and GPS, so that a user gets a complete picture.

hamishwillee, 2017-04-06:
@LorenzMeier That sounds awesome! Is anyone tasked with this tooling (generating the diagrams) and if not, what sort of generation tools were they planning on using?

LorenzMeier, 2017-04-06:
Beat was looking into it but no, we do not have a person on this right now full-time.

hamishwillee, 2017-04-06:
OK, I've submitted #6990  which includes fixes as discussed above and with Beat.

dagar, 2017-04-06:
@hamishwillee it looks like you can include external markdown?
If we do it that way we can avoid creating a new jenkins users with appropriate permissions and the potential gitbook sync issues we've been hitting.

WIP parameters.md generate and upload - https://github.com/PX4/Firmware/pull/6992

hamishwillee, 2017-04-06:
Hi @dagar 

There is nothing to indicate gitbook can [import from arbitrary locations](https://toolchain.gitbook.com/templating/conrefs.html) (only locally or from github), though it might be able to. Another complication is that importing files from github (and presumably elsewhere is VERY slow). Doesn't matter much on the server, but it will slow a local build quite a lot ... which is painful for editors.

The best way from my perspective is that every time a new md file is built it gets submitted into the Devguide on github. 

Thoughts?
H

dagar, 2017-04-06:
Alright, let's give it a try.

@LorenzMeier can we use PX4BuiltBot for this purpose? If so we just need to get his credentials on jenkins and I can do the rest.

dagar, 2017-04-10:
@hamishwillee do you want to manually commit the parameters.md from s3 (https://s3.amazonaws.com/px4-travis/Firmware/param_markdown/parameters.md) initially? I can work on getting jenkins (or something else) push access to the repository as a separate task.

hamishwillee, 2017-04-10:
@dagar How often do they change? ie If I submit manually do I need to put a note stating that these could be out of date? ETA on pushing from repo?

dagar, 2017-04-10:
The version here on the old site (https://pixhawk.org/firmware/parameters) was late updated June 2016, and I still use it multiple times per week. Adding the date generated or based on PX4 version certainly wouldn't hurt.

The auto update needs a few things that are out of my control (github, gitbook, jenkins user with permissions, etc).

hamishwillee, 2017-04-10:
OK, then I will submit an update later today. Thank you!

hamishwillee, 2017-04-10:
Done - you can see it here: https://dev.px4.io/parameter_reference.html

Note the filename. Keep me posted on progress for auto-update please. Once that is done (and possibly before) we'll want to redirect from https://pixhawk.org/firmware/parameters

hamishwillee, 2017-04-18:
@dagar we've now got both DevGuide and User guide "places" for the parameter reference:

DevGuide: https://dev.px4.io/parameter_reference.html
User Guide: https://docs.px4.io/en/advanced_config/parameter_reference.html

So feel free to auto-update these spots whenever you like! The repos are both under PX4 org.

mhkabir, 2017-04-18:
Moving the auto-update to next release since we already have the baseline functionality needed.

closed 2018-02-21 by PX4BuildBot