PX4/PX4-Autopilot #6914: Auto takeoff misbehaving

sanderux, 2017-03-28:
[image]

@LorenzMeier doesnt seem to be related to hover thurst. Do you know who to ping on this?

sanderux, 2017-03-28:
@Stifael it seems to be related to the land detector

[image]

Stifael, 2017-03-28:
Interesting, the thrust is quite large and it still detects landing...
Can u upload the log?

sanderux, 2017-03-28:
Interestingly the second flight i gave a bit of thrust in manual mode before switching to mission, that didnt exhibit the behavior:

Hard takeoff: http://logs.uaventure.com/view/VWPgEbTsU5huWhaKkBSGjf
Soft takeoff: http://logs.uaventure.com/view/5risy3gWaMT8dzQtxa9Jh3

sanderux, 2017-03-29:
@Stifael did you find anything?

Stifael, 2017-03-29:
is there any reason why thrust_min is set to 0.3?

Stifael, 2017-03-29:
https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L325

sanderux, 2017-03-29:
The reason is that during landing a vtol sometimes catches wind under the wing. If the motors go lower then 30% it loses stability and it looks like it will fall out of the sky

sanderux, 2017-03-29:
I still dont understand why it would interfere with the takeoff procedure in mission mode.
It seems that giving a little manual thrust before switching to mission makes it behave correctly

sanderux, 2017-03-29:
Perhaps checking the setpoint type, in type=takeoff it should never consider being landed

Stifael, 2017-03-29:
that could fix that part, but with min thrust of 0.3 I would not be surprised if landing would be detected during flight

Stifael, 2017-03-29:
what you think about introducing a second min thrust specifically for vtol?

Stifael, 2017-03-29:
I think the min thrust should not be a parameter for tuning, but actually a min that the props can produce. otherwise the landdetection does not make much sense

sanderux, 2017-03-29:
A min thrust during landing might be an idea. but i dont see how this is related to the mission takeoff issue,

sanderux, 2017-03-29:
i mean, the setpoint type takeoff should not care if it is landed

Stifael, 2017-03-29:
well, I would say the other way around. the landed should not care if is a takeoff or not

Stifael, 2017-03-29:
it you do takeoff, but doent leave the ground and the min thrust is satisfied, then I think it is valid to detect a landing

sanderux, 2017-03-29:
I think we are talking in circles.

>if you do takeoff, but don't leave the ground and the min thrust is satisfied

I don't want this behavior. I dont want something preventing the takeoff setpoint from being allowed to provide thrust. The current behavior of a takeoff is: stay at min thrust for 2 seconds and then bang it to max thrust. That is a bug and i am trying to figure out why this is happening as it is not the correct procedure. If you want to see what is SHOULD do have a look at the second log i posted. there the only difference is that i forced the land detector not to consider the drone landed before switching to mission mode.

So the question is not if i want land detection during takeoff, the question is: why is the takeoff procedure not functioning correctly? and subsequently: given the data do you think it could ne caused by the new land detector?

Stifael, 2017-03-29:
Well, as I said, the thrust is below the minimum thrust for more than 2 seconds. The landing detection has obviously its limitation, and one criteria for landing detection is the thrust (maybe in the future we can change it to  velocity). Once landing detected, it will throttle down. 
All I am saying is that the landing should stay independent of the rest of the logic. The takeoff is definitely not there yet. But I also think that it is not advisable to increase  the min thrust for flight performance improvement, since, as said, the landdetector uses it.

Stifael, 2017-03-29:
And given the data, of course it does since land detected. I need to look at the log again, but I assume takeoff was detected first and then land detected.

sanderux, 2017-03-29:
@Stifael i am at a loss on how to phrase this so i can get my point across. I will try one more time.

When the mission commands a takeoff the vehicle currently stays on the ground for 2 seconds, this is incorrect behavior. you are saying that as a **result** of this behavior the land detector kicks in. my question is if the land detector could be the **cause** of this behavior because it **should not** stay on the ground at min thrust for 2 seconds.

Stifael, 2017-03-29:
Before landing gets detected there is ground contact, where thrust in the body x and y gets turned off, but not in the z. Unfortunately px4log does not show that message (you have to add it or use ulog). That said, ground contact depends on the min thrust as well, so it was true as well. 
Why can you not decrease the thrust min and try to takeoff again

sanderux, 2017-03-29:
I will if you think that will solve it, but the link to the code you sent i dont see how that would effect things. obviously the land detector will consider the vehicle landed because it IS landed. my question is: why is it preventing a takeoff?

Stifael, 2017-03-29:
If ground contact was detected, then it shouldnt since only body x and y thrust is turned off. But it could be that it is a bug and for some reason limited the thrust in z as well. Do you mind to test it with ulog?

sanderux, 2017-03-29:
ok, i'll do that tomorrow

sanderux, 2017-03-29:
Could it be this?
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1884

Stifael, 2017-03-29:
Yep,  thats where ground contact is handled. I actually cant remember the purpose of limiting the z direction when in ground contact.
But when looking at your plot, the thrust setpoint was non zero, which should be since the desired thrust during takeoff is pointing upward which means this if statement is not true and therefore the thust in z will be set to zero.
I need to have a closer look at it. I fly everyday and I can land and takeoff immediately, thats why i was first surprised to have even landing / groundcontact detected when the throttle is that high.

sanderux, 2017-03-29:
But here it's actually copied to the thrust_sp: https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1887

Stifael, 2017-03-29:
Yes, but thrust_body should be zero and therefore also the thrust_sp. Anyway, I will test it tomorrow in sitl with increased min thrust. I think it should be possible to reproduce it in  sitl

Stifael, 2017-03-30:
actually now I know why this thrust_sp(2) is set to 0. however, once the landdetection criteria is not satisfied, ground contact and landing should immediately be false.

sanderux, 2017-03-30:
ok, but at takeoff the landing criteria should be satisfied correct?

sanderux, 2017-03-30:
the time it stays on the ground looks a lot like the hysteresis time for the land detector

sanderux, 2017-03-30:
my guess is the setpoint goes up and land detector hysteresis blocks the thrust_sp until the it's past it's hysteresis time for thrust

Stifael, 2017-03-30:
The hysteresis just counts but does not block. Once one of the criteria is not satisfied, it just jumps out and the hysteresis gets reset. otherwise it would not make much sense to use a "two-state" land detector. 
The detection goes as follow:
1.) check if ground contact is detected. the criteria has to be true for about 1.5 seconds to actually detect ground. if a criteria fails, it will start from the beginning and no blocking should occur
2.) if ground was detected, then additional criteria are used to decide if we are actually landed. landing detection can only occur if ground contact was detected, otherwise it is not possible. 

However, the mc_pos_control could block it, and I think what is missing is something like this here for the ground contact:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1927 

This also explains why it did not jump out of ground contact, because in the mc_pos_control the thrust_sp is set to 0 when giving throttle upward and all other criteria (such as velocity) is still zero since the vehicle is still on the ground. 
I will add a pr with the takeoff fix. thanks for your patience

Stifael, 2017-03-30:
however, I am still not entirely sure if this is the correct fix. it would still be appreciated if you could do a takeoff with ulog enabled

Stifael, 2017-04-06:
whats the status here?

closed 2017-04-10 by Stifael