PX4/PX4-Autopilot #6915: Mag Timeout

tops4u, 2017-03-28:
Since 1.6 RC I have observed twice the following Error:

[image]

HW: AUAV-X2 / SW: 9235c0fd (1.6 RC)

LorenzMeier, 2017-03-28:
@bkueng I think you were investigating this. Worth talking to David and through the architecture.

tops4u, 2017-03-28:
Just to clarify: I have not observed any Impact on Flight performance.

LorenzMeier, 2017-03-28:
It should not impact it significantly, but its still something we want to resolve.

bkueng, 2017-03-29:
I haven't seen it on mission update so far. It is strange, since both `lpwork` (on which the mag runs) and `sensors` run at a higher prio than `mavlink`, `dataman` and `navigator`.

tops4u, 2017-03-29:
I have also MAVLINK Stream dropouts (#6260) so OSD Data lags siginficantly behind real Fligh Data. I could observe that yesterday where Altitude Data was not realtime and it was hard to hit a Fligh Level of 35m above ground as the update on Alt was only very sporadic. Also the OSD Complained often about Frozen Stream Data...

davids5, 2017-03-30:
@tops4u, @bkueng  - has this been observed since the https://github.com/PX4/Firmware/commit/a6dcbc3a222c22006b4c26e8d2280072b5c399d5 came into master.?

Until that commit priorities might not have been as designed.

tops4u, 2017-03-30:
Ok. I'll retry with current master. Would you consider it more or lass reliable?

davids5, 2017-03-30:
You can see the commits that have been done since 1.6 in in the last 22 days.

I would bench test it with props off first.  

You will have to make your own assessment of stability but I would expect it is more stable.

tops4u, 2017-03-30:
I'll try... Can't promise fast answers as this only happened twice in the last 2 Month (MAG Timeout), I'll be more interested about OSD Updates - but maybe this is fixed too.

tops4u, 2017-04-01:
I'm closing this. Installed Master from today and I'll reopen should the error reoccur.

closed 2017-04-01 by tops4u