PX4/PX4-Autopilot #6916: IC Engine Throttle settings.

tubeme, 2017-03-28:
Hello guys,

I'm preparing a pretty large airplane with IC engine for Monday flights. I played with the FW_THR_MIN and FW_THR_IDLE but i really could not see the difference between the two parameters.

As far as I understand FW_THR_IDLE is the 0 throttle setting for the servo that pushes the throttle. By default it stays at 15%. No effect when I push it to 25%.

Does this concern only the auto modes??? If so then the what it the FW_THR_MIN? While on the ground I stay in Stabilized and this parameter did not have any effect on the throttle in Man-Stabilized.

If it is something else then the Parameter reference page is not correct. 

Any ideas?

LorenzMeier, 2017-03-28:
Idle throttle is intended to be a good "standby" throttle value prior to takeoff. It should be higher than min and lower than trim. It will only be used in the launch detection section right now.

You will want to set pwm disarmed to IC cut and throttle min to a value where the IC engine is safely running.

This is also how most electric systems are set up.

tubeme, 2017-03-28:
OK I understand. 

Besides we are using a throttle cut electronics (the Gas engines cannot be cut with just the trottle) that is how we set it up. The disarm is the lowest possible value almost cutting the engine. The the Min should be such that it guarantees the auto modes not cutting the throttle in flight. 

So I understand that the IDLE does not affect the plane while taking off from airstrip but rather it kicks in when launch is detected correct? How about if we launch with a catapult?

LorenzMeier, 2017-03-28:
It sets the throttle you would have once you activate launch detection on the catapult.

tubeme, 2017-03-28:
Clear. Thanks. I will try to find the param info in the QGC tree and give some more detail on it.

dagar, 2017-04-11:
@tubeme is something else needed for planes with IC engines, or is this mainly a documentation issue?

tubeme, 2017-04-11:
No @dagar it works perfect. It is very good idea to have the Mixer editor implemented in QGC because I spent hour and a half editing the mixer until i got the correct movement for the throttle servo. The margins were so tight on the IC engine I worked on I had to do 20 edits of the mixer. There was cut from the bottom and the upper limits of the servo until I got it right. We did not have a chance to do it mechanically because we had to remove the engine in order to do it this way. The guys that mounted the engine and the servo were just horrible mechanics.

I think somebody started working on mixer editor in QGC. @DonLakeFlyer even made an interface for it.

tubeme, 2017-04-11:
@dagar @LorenzMeier The flights were perfect!!

dagar, 2017-04-11:
Thanks for the update. Hopefully we'll have configurable mixers in the not to distant future.

closed 2017-04-11 by dagar