PX4/PX4-Autopilot #6918: Pixhawk Mini: RC Input is not detected if using GCC 4.9 in Mac OS

d3z0rian, 2017-03-28:
RC Input is not detected if master for px4fmu-v3_default is compiled with Toolchain 4.9.3 in Mac OS.

Not working with 4.9.3:
[image]
[image]

Working with 5.4.1:
[image]
[image]

Autopilot: Pixhawk Mini
Mac OS: Sierra 10.12.3

For Linux there are instructions to install GCC 4.9 or 5.4:
https://dev.px4.io/starting-installing-linux-boutique.html#toolchain-installation

For MacOS, instructions don't specify a version but in my case, it installed 4.9.3
https://dev.px4.io/starting-installing-mac.html#homebrew-installation

I uninstalled GCC running:
`brew uninstall gcc-arm-none-eabi`

And Installed GCC 5.4.1 running:
`brew install gcc-arm-none-eabi-54`

I don't know in which cases I should use 4.9 or 5.4, also there are newer versions (https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads), but for now I will use 5.4

yl5006, 2017-03-29:
I have same problem in linux

dagar, 2017-04-11:
The official builds are using GCC 5.4.

closed 2017-04-11 by dagar