PX4/PX4-Autopilot #6919: How to connect Pixhawk to Lidar Lite V3

vinhk, 2017-03-28:
I am trying to make Pixhawk work with Lidar Lite V3 using PWM. I enable `SENS_EN_LL40LS` in QGC. However, I am unable to see it's data getting integrated into Pixhawk as distance sensor. Here is my wiring based on https://pixhawk.org/peripherals/rangefinder.

[image]

Can anyone verify it's working?

mhkabir, 2017-03-29:
The driver is not compiled for FMUv2 due to limited flash and tiny user base (till v2). You'll need to uncomment the driver from the fmuv2 cmake file, or use the fmuv3 target if your Pixhawk has the Rev 3 chip with no flash limitations.

vinhk, 2017-03-29:
@mhkabir would you be able to direct me to the driver that is needed to be uncomment from fmuv2 cmake file? I am using px4fmu-v2_lpe.

Abelteklu, 2017-03-29:
@vinhk , I think you have to uncomment drivers/ll40ls  from Firmware/cmake/configs/nuttx_px4fmu-v2_default.cmake file. Then do,

make px4fmu-v2_lpe upload

Abelteklu, 2017-03-29:
@vinhk to check if your pixhawk has rev 3 chip, you can,

1- using nsh type, 

_free_

and see the used and free flash memory size. Add the two and if they add up to approximately 2MB, ur Pixhawk has the rev. 3 chip. (Im not sure about this procedure. I will check....but I think the code is on the flash memory)

or 

2- Open pixhawk's cover and see if it has the STM32F427 Cortex chip in the back.
I don't know if there is a software way to do it. If ur pixhawk has the rev 3 board, as @mhkabir noted, you can do

_make px4fmu-v3_default upload_ 

and see what happens.

mhkabir, 2017-03-29:
You can just run "ver hw" from nsh to check the chip revision.

vinhk, 2017-03-30:
@mhkabir I have revision 2. What can we comment out from cmake file as `drivers/ll40ls` is already uncomment for px4fmu-v2_default.cmake ?

mhkabir, 2017-04-06:
Hmm, it seems that it is compiled. 

Can you run `pwm_input test` and send the output here? If you don't see any output, then you have a wiring issue.

vinhk, 2017-04-07:
@mhkabir Here is what I got. I have already enable ll40ls in QGC
`nsh> pwm_input test`
`Showing samples for 5 seconds`
`nsh> `

mhkabir, 2017-04-07:
OK, we can confirm the issue on our side as well. Please continue checking https://github.com/PX4/Firmware/issues/6994 for fixes as and when they come.

closed 2017-04-07 by mhkabir

tomatac, 2017-04-30:
What is this command suppose to show "nsh> pwm_input test"?

kevinjhur, 2017-05-25:
This is due to no power to LiDar.
Actually the aux ports of Pixhawk do not supply 5V Vcc. You can check there is no power from mid of aux 6.
Please use other power line.

sspatank, 2017-08-01:
@kevinjhur What do you mean by the other power line? Won't the Power Enable still need to be connected to AUX 6?

kevinlyu1, 2017-08-21:
I think all he's saying is you need to provide power to the aux PWM's.

IeiuniumLux, 2018-07-17:
@vinhk, were you able to get it working with your [wiring connections](https://cloud.githubusercontent.com/assets/17210431/24431599/d3441eb0-13ea-11e7-85e5-a3eb087f902e.png)?

vinhk, 2018-07-17:
yeah. That should be it. Just make sure you have the power and ground
properly connected.

vinhk, 2018-07-17:
but @IeiuniumLux it's been a while. Again, I think you may need to provide external power to the lidar.

IeiuniumLux, 2018-07-17:
Hamish,

Let me run some tests first to make sure those connections work with the current PX4 driver.  If I get it working, then I'll submit a pull request with the updates for the docs.  Cheers!

hamishwillee, 2018-07-18:
@IeiuniumLux Awesome, and please see my notes in https://github.com/PX4/px4_user_guide/issues/294#issuecomment-405779193 - where I have started this job.