PX4/PX4-Autopilot #6921: Oneshot mode is not entered PWM_RATE=0

davids5, 2017-03-29:
@LorenzMeier 

FYI @kd0aij 

I do not see where PWM_RATE's value is propagated to the io_timer_set_rate(group, rate) ever. On the FMU is it always 50 regardless of the param being PWM_RATE=0. So I think we either need some start up changes or make the fmu/io code read the value and use it.

rolandash, 2017-04-12:
Set PWM_RATE = 0 on MindRacer did not seems to enter OneShot mode properly as expected. @iZhangHui

davids5, 2017-04-25:
@LorenzMeier - I have traced the cause of PWM_RATE = 0 not being propagated to the the low level driver.  I have a [PR](https://github.com/PX4/Firmware/pull/7121) that will fix that part of this. 

But even with that fixed the onshot mode is exited because of the following: 

In testing on a v4pro, I found that PWM_RATE is set from a FRAM parameter  but PWM_AUX_RATE is coming from a environment set.  The driver is called with p:PWM_RATE == 0, but is then is then called again with PWM_AUX_RATE=50 and switches the  AUX channels out of oneshot mode.  

It comes from from /ROMFS/px4fmu_common/init.d/rc.mc_defaults
```
set PWM_AUX_RATE 50
```

This is possibly an artifact of my SYS_AUTOSTART setting: 
```
SYS_AUTOSTART [536,757] : 4011
x + UAVCAN_ENABLE [710,946] : 0
x + PWM_RATE [372,575] : 0
```
Should I be using a different SYS_AUTOSTART or do we need a more active way of setting onshot mode via rcS?

LorenzMeier, 2017-04-30:
Fixed.

closed 2017-04-30 by LorenzMeier

knopancakes, 2018-06-25:
This issue is back

knopancakes, 2018-06-25:
using a custom config.txt and mixers on the SD card. In config.txt, PWM_RATE is set to 0 (zero), and the parameter viewable through QGroundControl is also set to 0 (zero). However, calling 'pwm info' shows everything is set to 50Hz. My PWM_AUX_RATE does configure correctly, confirmed with 'pwm info -d /dev/pwm_output1'

knopancakes, 2018-06-25:
fixed by explicitly setting the following in config.txt

set OUTPUT_MODE fmu

seems the following guard is the problem:

if [ $OUTPUT_MODE == fmu -o $OUTPUT_MODE == io ]
then
	if [ $PWM_OUT != none ]
	then
		#
		# Set PWM output frequency
		#
		if [ $PWM_RATE != none ]
		then
			pwm rate -c ${PWM_OUT} -r ${PWM_RATE}
		fi