PX4/PX4-Autopilot #6924: Tracing malloc on NuttX

bkueng, 2017-03-29:
Ever since #4742 I was wondering how it would look like on NuttX. Now I got down to it and printed the caller addresses and size of malloc, and combined the output into a table.

### Results
In the following, the first column shows the total size of all allocation calls in bytes, the second the number of calls, then the caller method and the allocated size (minimum and maximum if there are several). It does not take into account `free`.
This is on Pixhawk:
```
 TOTAL  CNT SYMBOL                                                                 SIZE(S)
    15    1 setenv + 115                                                           15
    20    1 PX4FMU::pwm_ioctl(file*, int, unsigned long) + 1353                    20
    24    1 MS5611::init() + 39                                                    24
    24    1 LSM303D::init() + 61                                                   24
    24    1 LSM303D::init() + 31                                                   24
    24    1 MPU6000::init() + 79                                                   24
    24    1 L3GD20::init() + 25                                                    24
    24    1 MPU6000::init() + 51                                                   24
    28    1 param_set_internal + 35                                                28
    28    7 device::CDev::store_poll_waiter(pollfd*) + 63                          4
    32    1 uORB::Manager::initialize() + 15                                       32
    40    1 procfs_initialize + 15                                                 40
    48    2 ftl_initialize + 15                                                    24
    50    1 ADC::ADC(unsigned int) + 89                                            50
    56    1 mmcsd_slotinitialize + 15                                              56
    56    1 drv_led_start + 17                                                     56
    72    1 LSM303D::LSM303D(int, char const*, spi_dev_e, Rotation) + 45           72
    72    1 MPU6000::MPU6000(device::Device*, char const*, char const*, Rotation, int) + 35 72
    76    1 LIS3MDL_I2C_interface(int) + 11                                        76
    76    1 MS5611_spi_interface(ms5611::prom_u&, unsigned char) + 19              76
    76    1 MS5611_spi_interface(ms5611::prom_u&, unsigned char) + 33              76
    80    4 SimpleMixer::from_text(int (*)(unsigned int, unsigned char, unsigned char, float&), unsigned int, char const*, unsigned int&) + 143 20
    80    1 uORB::Manager::get_device_master(uORB::Flavor) + 19                    80
    84    1 px4::logger::LogWriter::LogWriter(unsigned char, unsigned int, unsigned int) + 35 84
   112    1 adc_main + 25                                                          112
   120    1 bst_main + 57                                                          120
   144    3 GPS::task_main() + 293                                                 48
   152    2 HMC5883_I2C_interface(int) + 11                                        76
   176    1 ms5611::start_bus(ms5611::ms5611_bus_option&, MS56XX_DEVICE_TYPES) + 69 176
   192    4 SimpleMixer::from_text(int (*)(unsigned int, unsigned char, unsigned char, float&), unsigned int, char const*, unsigned int&) + 69 48
   208    1 meas_airspeed::start(int) + 25                                         208
   208    1 rgbled_main + 193                                                      208
   208    1 rgbled_main + 129                                                      208
   208    1 meas_airspeed::start(int) + 83                                         208
   216    9 stm32_ep_allocreq + 9                                                  24
   216    1 gps::start(char const*, gps_driver_mode_t, GPSHelper::Interface, bool, bool, int) + 45 216
   216    3 MPU9250_SPI_interface(int, bool) + 21                                  72
   228    3 MPU6000_SPI_interface(int, int, bool) + 39                             76
   256    1 EstimatorInterface::initialise_interface(unsigned long long) + 395     256
   256    1 EstimatorInterface::initialise_interface(unsigned long long) + 471     256
   256    1 EstimatorInterface::initialise_interface(unsigned long long) + 319     256
   264    2 PX4IO_serial_interface() + 9                                           132
   264    1 tone_alarm_main + 21                                                   264
   272    2 DataValidatorGroup::add_new_validator() + 11                           136
   280    1 px4::logger::LogWriter::LogWriter(unsigned char, unsigned int, unsigned int) + 73 280
   296    2 ets_airspeed::start(int) + 25                                          148
   392    1 RingBuffer<estimator::magSample>::allocate(int) + 71                   392
   405    1 px4::logger::Logger::run() + 311                                       405
   416    1 create_work_item + 45                                                  416
   480    1 l3gd20::start(bool, Rotation) + 53                                     480
   544    4 DataValidatorGroup::DataValidatorGroup(unsigned int) + 47              136
   576    3 GPS::task_main() + 259                                                 192
   592    1 lsm303d::start(bool, Rotation, unsigned int) + 57                      592
   600    1 px4io_main + 283                                                       600
   600    1 px4io_main + 61                                                        600
   704    9 cdcacm_allocreq + 25                                                   [64...96]
   784    1 fmu_main + 251                                                         784
   824    1 mpu6000::start_bus(mpu6000::mpu6000_bus_option&, Rotation, int, int) + 95 824
   852   71 uORB::DeviceMaster::ioctl(file*, int, unsigned long) + 229             12
   904    1 RingBuffer<estimator::flowSample>::allocate(int) + 81                  904
   904    1 RingBuffer<estimator::extVisionSample>::allocate(int) + 77             904
   904    1 RingBuffer<estimator::gpsSample>::allocate(int) + 77                   904
  1016    1 RingBuffer<estimator::imuSample>::allocate(int) + 75                   1016
[... 47 lines cut ...]
 
+static bool in_printf = false;
+bool g_board_init = false;
+
 FAR void *malloc(size_t size)
 {
+	if (!up_interrupt_context()) {
+		sched_lock();
+		if (!in_printf && g_board_init) {
+			in_printf = true;
+			void *ret_addr = __builtin_return_address(0);
+			printf("malloc 0x%p %i\n", ret_addr, size);
+			in_printf = false;
+		}
+		sched_unlock();
+	}
 #ifdef CONFIG_BUILD_KERNEL
   FAR void *brkaddr;
   FAR void *mem;
```
I had to start tracing only after NuttX booted, otherwise it would crash. It's done by adding `g_board_init = true;` to `board_app_initialize` in `px4fmu2_init.c`.

And the script to create the output: https://gist.github.com/bkueng/c1e82871b26dd105da84e150a4c57c04

Conclusion: there is still quite a bit of memory that we can free up.

davids5, 2017-03-30:
@bkueng 

> cdcacm_initialize: the USB USART buffer, CONFIG_CDCACM_{RX,TX}BUFSIZE. @davids5 does this really need to be that large? For TX, with an active QGC connection, I measured that maximum 50 bytes are needed, with ULog streaming (~40kB/s sending rate), around 250B.

My hazy (Q1 2015)  recollections was that we saw degraded throughput on "Log" downloads when the buffers were too small do to starvation and the effects of the system running with interrupts off from the HRT dispatch. 

A lot of the system has changed with regard to logging, since then and this should be reevaluated.  

I assume we have better tools now to evaluate download speeds. This should be a simple matter of retesting with different size buffers to see if any of this is still applicable. Just keep in mind that the upper bound on the throughput is affected by the interrupts off time and the buffer size. Once the buffer are large enough, the throughput will level off below the calculated maximum due to the delay in servicing the USB while the interrupts are off from the HRT dispatches.

bkueng, 2017-04-07:
Closing this for now, we can come back to it when needed.

closed 2017-04-07 by bkueng