PX4/PX4-Autopilot #6929: Qualcomm (Snapdragon) ESC control problem?

Seanmatthews, 2017-03-29:
**Some upfront info:**
* Using the Snapdragon Flight development kit with the Qualcomm ESC controller
* Using RC controller DX6e
* Installed toolchain according to https://github.com/ATLFlight/cross_toolchain
* Flashed board with: https://support.intrinsyc.com/projects/snapdragon-flight/files
* Hexagon version: https://developer.qualcomm.com/download/hexagon/hexagon-sdk-v3-linux.bin
* Installed Intrinsyc-provided fc_addon: https://support.intrinsyc.com/projects/snapdragon-flight/files 
* Using px4.config for 210qc

What I've found with this configuration is that the PX4 app built with `make eagle_default` does not activate the ESC controller (none of the four blue blinking LEDs). However, when built with `make eagle_legacy_default` is used to build the firmware, the PX4 app does activate the ESC controller. See the debug outputs for both below.

My questions are:
1. Should this repo be built with `make eagle_legacy_default` for the Snapdragon Flight with Qualcomm ESC controller? Should the other build method work at all?
2. If the other build method should work, what should I do differently?
3. What are the essential differences between using the legacy and non-legacy drivers? What am I losing by using the legacy drivers?

Debug output from firmware built with `make eagle_default`:
> DMSS is connected. Running mini-dm...
> [08500/00]  00:45.312  HAP:63:adsp memory mapped   0155  shmem_qurt.c
> [08500/00]  00:45.355  HAP:61:In dspal_entry  0220  main.cpp
> [08500/00]  00:45.355  HAP:61:adsp memory mapped   0155  shmem_qurt.c
> [08500/00]  00:45.356  HAP:4152:TEST px4muorb_topic_advertised of [mavlink_log] on remote side...  0167  px4muorb.cpp
> [08500/00]  00:45.491  HAP:61:attempting to open the ADSP command file: /dev/fs/px4.config  0184  main.cpp
> [08500/00]  00:45.492  HAP:61:reading commands from /dev/fs/px4.config   0191  main.cpp
> [08500/03]  00:45.495  HAP:54:Error subscribing to qshell_req topic  0079  qshell.cpp
> [08500/03]  00:47.092  HAP:61:Parameter UART_ESC_BAUDRTE not found.  0513  param.c
> [08500/00]  00:48.291  HAP:61:2936459 Reset MPU9250   0089  MPU9250.cpp
> [08500/00]  00:48.355  HAP:60:3000684 IMU temperature initialized to: 57.116341   0521  MPU9250.cpp
> [08500/00]  00:48.356  HAP:61:3002141 additional sensor configuration succeeded   0187  BMP280.cpp
> [08500/02]  00:48.380  HAP:49197:LED: getHandle fail   0279  commander_helper.cpp
> [08500/02]  00:48.380  HAP:49197:LED init failed  1372  commander.cpp
> [08500/02]  00:48.380  HAP:49197:Buzzer: px4_open fail   0132  commander_helper.cpp
> [08500/02]  00:48.380  HAP:49197:Buzzer init failed  1376  commander.cpp
> [08500/02]  00:48.382  HAP:49197:Preflight checks always pass on Snapdragon.  0534  PreflightCheck.cpp
> [08500/00]  00:49.450  HAP:61:PX4 Devices:  0560  vdev.cpp
> [08500/00]  00:49.450  HAP:61:DF Devices:  0573  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /dev/imu0  0583  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /dev/baro0  0583  vdev.cpp
> [08500/00]  00:49.450  HAP:61:Files:  0609  vdev.cpp
> [08500/00]  00:49.450  HAP:61:Active Tasks:  0314  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   dspal      2097663  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   hpwork     2097661  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   lpwork     2097660  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   wkr_hrt    2097659  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   sensors    2097656  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   commander  2097655  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   ekf2       2114038  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   mc_pos_control 2589173  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:   mc_att_control 2114036  0318  px4_qurt_tasks.cpp
> [08500/00]  00:49.450  HAP:61:Devices:  0593  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/_obj_  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/sensor_accel0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/sensor_gyro0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/parameter_update0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/sensor_mag0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/input_rc0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/rc_parameter_map0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/sensor_baro0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/differential_pressure0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/vehicle_control_mode0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/actuator_controls_00  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/sensor_correction0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/sensor_combined0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/sensor_preflight0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/led_control0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/vehicle_status0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/actuator_armed0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/safety0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/mission_result0  0599  vdev.cpp
> [08500/00]  00:49.450  HAP:61:   /obj/geofence_result0  0599  vdev.cpp
[... 123 lines cut ...]
> [08500/00]  01:37.875  HAP:61:   /obj/position_setpoint_triplet0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/system_power0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/vtol_vehicle_status0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/cpuload0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/commander_state0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/airspeed0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/optical_flow0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/distance_sensor0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/vision_position_estimate0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/estimator_status0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/wind_estimate0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/ekf2_innovations0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/sensor_baro0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/control_state0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/vehicle_attitude_setpoint0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/vehicle_local_position_setpoint0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/vehicle_global_velocity_setpoint0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/vehicle_rates_setpoint0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/multirotor_motor_limits0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/rc_channels0  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/actuator_controls_30  0574  vdev.cpp
> [08500/00]  01:37.875  HAP:61:   /obj/mavlink_log0  0574  vdev.cpp
> [08500/00]  01:38.498  HAP:4139:EKF using pressure height - commencing alignment  0412  ekf.cpp
> [08500/00]  01:43.015  HAP:4139:EKF alignment complete  0064  control.cpp
> [08500/02]  03:45.382  HAP:8237:REJECT AUTO MISSION  3717  commander.cpp

LorenzMeier, 2017-03-29:
@mcharleb could you chime in?

mcharleb, 2017-03-30:
@Seanmatthews The only drivers for the Qualcomm ESCs are the legacy (binary) drivers. We may have open versions in the future but not currently so eagle_legacy_default is what you want. All the drivers were initially binary thus the "legacy" name. You can compare cmake/configs for the qurt_sdflight_default.cmake and qurt_sdflight_legacy modules to see the differences. You can add any modules you like but will need the platforms/qurt/fc_addon/uart_esc module for the Qualcomm ESC. @jywilson  Let me know if I missed anything.

mhkabir, 2017-04-03:
Closing in favour of https://github.com/PX4/Firmware/issues/6962

closed 2017-04-03 by mhkabir