PX4/PX4-Autopilot #6938: Disable flightmode change on rc input for non-MC vehicles

sanderux, 2017-03-31:
This is a very bad idea for fixed wings / vtols:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2525-L2546

I understand the logic for MC's but any accidental touch of a stick could send a plane flying a mission of into an unknown direction.

LorenzMeier, 2017-04-01:
I think we can discuss if this needs to be configurable, but the basic statement is not true. Fixed wing in position control will hold direction and altitude, so the override analogy works the very same way.

sanderux, 2017-04-01:
Would an airplane not continue to fly straight when such an override occurs? I believe this might be a problem when it goes unnoticed during a mission. The plane could fly quite far possibly beyond reach. Or it could fly in to something like a tree or building if the pilot isnt concious of the fact that control was returned during a mission.

sanderux, 2017-04-04:
@LorenzMeier we discussed this in the dev call, consensus there was to make it a parameter. I'll add a PR for this if you want.

LorenzMeier, 2017-04-06:
Thanks, that would be much appreciated!

sanderux, 2017-04-12:
done https://github.com/PX4/Firmware/pull/7025

closed 2017-04-12 by sanderux