PX4/PX4-Autopilot #6940: UAVCAN UX improvements

pavel-kirienko, 2017-03-31:
@eyeam3 this is a follow-up to our call. The most pressing issues with the UAVCAN UX are the following:

- [ ] GCS parameter transfer is unreliable. Reportedly it sometimes fails, but mostly works reliably. You said that you were able to observe the issues locally, which is good. Also see https://github.com/PX4/Firmware/pull/5724.

- [ ] Status information of UAVCAN nodes should be relayed to QGC and displayed adequately. See the comments starting from https://github.com/PX4/Firmware/pull/5356#issuecomment-241218160.

Concerning the last issue, my proposition is to define a new MAVLink message that would be carrying UAVCAN node status information, and add support for it on the side of QGC in the form of, perhaps, an additional tab in the top panel. Admittedly I have not given much thought to this problem yet, suggestions welcome.

FYI @DonLakeFlyer @LorenzMeier @mhkabir

mhkabir, 2017-03-31:
Weren't the parameter transfers already fixed in https://github.com/PX4/Firmware/pull/6585?

pavel-kirienko, 2017-03-31:
Reportedly they weren't, but I didn't test those changes myself yet.

pavel-kirienko, 2017-04-09:
I propose to add an additional tab in the top panel, as mentioned in the first message of this thread, and place a table in it. The table should contain information about every UAVCAN node that is present on board. The following columns should be defined:

* Node ID.
* Node name, e.g. `io.px4.sapog`. The user will be able to infer the function of the node based on this parameter.
* Health of the node (OK, warning, error, critical).
* Operating mode of the node (operational, initialization, maintenance, software update - these are defined by the UAVCAN spec).
* Uptime.
* Vendor-specific status code. Shall the node fail, the user will be able to understand the nature of the failure and troubleshoot based on this value.

The interface should allow the user to pick any node from the list and modify its configuration parameters using the already existing parameter transfer capability. Essentially this is the same functionality that is provided by the main window of the UAVCAN GUI Tool:

[image]

Observe that this functionality is not related to any specific function that can be performed by a node. Support for that is less critical and can be implemented later.

I am willing to implement the necessary logic on the side of the autopilot, but I need help with the GUI side of this and also with the necessary extensions to MAVLink.

@eyeam3 @mhkabir @DonLakeFlyer Please let me know if this proposition makes sense.

mhkabir, 2017-04-09:
I'm concerned about the Health field. If a node was critical for example, it does not help us to debug the issue, but just know that there's an error 'somewhere', which is worse. The node ID is also opaque, because it doesn't tell us anything about which node it 'really' is.

pavel-kirienko, 2017-04-09:
> I'm concerned about the Health field. If a node was critical for example, it does not help us to debug the issue, but just know that there's an error 'somewhere' and that 'something' is wrong, which is worse.

Be not concerned. The vendor specific status code is used to convey the detailed, vendor specific information about the failure. This can be used to pinpoint and resolve the problem.

> which is worse

Worse than what? Worse than having no information about status of peripherals whatsoever? Surely not. There must be a way to obtain status information from the nodes. Right now there is no such thing. My proposition is intended to fix that.

More specific status handling logic can be added in the future; for example, there can be a separate GUI element dedicated solely to ESC. Actually there is no reason for it to be related to UAVCAN, since other kinds of ESC (e.g. those connected via RS485) can also utilize the same state reporting features.

pavel-kirienko, 2017-05-25:
@LorenzMeier @eyeam3 can we move this discussion forward please?

closed 2018-02-21 by PX4BuildBot