PX4/PX4-Autopilot #6942: Is the mavlink message "rc_channels_override" implemented well?

nitehack, 2017-03-31:
Hi. First of all I would say that I know that developers doesn't recommend using the mavlink rc_channels_override message but I need to use it to make a mix between the RC Radio and the data of some sensors on a mini PC.

I have tried it but I have verified that when I send the message rc_channels_override to pixhawk, the value of the channels is maintained only one cycle of execution and in the next cycle the control passes to RC Radio (I need to block RC Radio control until the mini PC to order). I am watching the [code](https://github.com/PX4/Firmware/blob/ebeb9eba4bc8255f07b8bdba485638d3694cccdf/src/modules/mavlink/mavlink_receiver.cpp#L1058)  and I do not see the mechanism that says mavlink that blocks the control of the RC Radio until it is sent to the channels the value 0.

There are people who say that it works and I don't know if they are using another firmware or I am using it badly.
Greetings.

dagar, 2017-04-11:
I don't think the mechanism to go back and forth exists. Why do you need both?

nitehack, 2017-04-16:
@dagar The official documentation of Mavlink says: RC_CHANNELS_OVERRIDE  "......._A value of UINT16_MAX means no change to that channel. A value of 0 means control of that channel should be released back to the RC radio_......"
Is this mechanism implemented in pixhawk firmware? I think this part does not, or I do not see it.
I need this mechanism for my project.

mhkabir, 2017-04-16:
No it isn't, and why do you even need `RC_CHANNELS_OVERRIDE`? It is not something we'll support for controlling from a companion computer. It is a horrible way to do it. Just use Offboard control : https://dev.px4.io/offboard-control.html

nitehack, 2017-04-17:
@mhkabir I want to mix RC Control with decision of a computer to avoid obstacles. How fast is it to turn on and turn off "offboard mode"?

mhkabir, 2017-04-17:
Yes it is possible. Read the page. You'd want to forward commands from your RC to the flight controller only via offboard.

nitehack, 2017-04-18:
@mhkabir To control the UAV in offboard mode the way I thought it is:
-Read RC Channels
-Use SET_ACTUATOR_CONTROL_TARGET

Is this correct? Is there a faster way to control the UAV in offboard mode?

Thank you very much for your help.

mhkabir, 2017-04-18:
Yes. What do you mean by "faster"?

nitehack, 2017-04-18:
With "faster" I mean another way to map RC channels that requires less time. (The time from which you are ordered to move the UAV until the pixhawk executes the command)
Sorry for my english

closed 2018-02-21 by PX4BuildBot