PX4/PX4-Autopilot #6945: upload error

potato77, 2017-03-31:
I can build the code. But when i try to upload the code with
`make px4fmu-v2_default upload`

The error is:
`QideMacBook-Air:Firmware Qi$ make px4fmu-v2_default upload
PX4 CONFIG: /Users/Qi/src2/Firmware/build_px4fmu-v2_default
ninja: Entering directory `/Users/Qi/src2/Firmware/build_px4fmu-v2_default'
[0/1] uploading /Users/Qi/src2/Firmware/build_px4fmu-v2_default/src/firmware/nuttx/nuttx-px4fmu-v2-default.px4
Traceback (most recent call last):
  File "/Users/Qi/src2/Firmware/Tools/px_uploader.py", line 59, in <module>
    import serial
ImportError: No module named serial
FAILED: src/firmware/nuttx/CMakeFiles/upload 
cd /Users/Qi/src2/Firmware/build_px4fmu-v2_default && /usr/local/bin/python /Users/Qi/src2/Firmware/Tools/px_uploader.py --port "/dev/tty.usbmodemPX*,/dev/tty.usbmodem*" /Users/Qi/src2/Firmware/build_px4fmu-v2_default/src/firmware/nuttx/nuttx-px4fmu-v2-default.px4
ninja: build stopped: subcommand failed.
make: *** [px4fmu-v2_default] Error 1`

dogmaphobic, 2017-03-31:
>ImportError: No module named serial

You don't have the `serial` package installed. Did you follow the *Getting Started* instructions?

https://dev.px4.io/starting-installing-mac.html

The serial package would be installed by:

```
sudo pip install pyserial empy pandas jinja2
```

Which is described in that page.

potato77, 2017-03-31:
@dogmaphobic 

thank u. The problem is solved.
But I already install the package and I succeed upload the firmware several times.
I don't know why this issue happen.

closed 2017-03-31 by julianoes

hamsterasu, 2019-04-09:
Hi @potato77, I have a similar problem to yours. How did you solve it?

julianoes, 2019-04-09:
@hamsterasu please don't necro-bump. Please create a new issue and paste the output of your failure.