PX4/PX4-Autopilot #6946: Mavlink messages getting published slower than requested

helenol, 2017-03-31:
On the stable branch, I'm getting a weird effect on the mavlink output of some high-rate messages such as raw IMU and attitude.
If requesting 20 Hz, get 20 Hz.
If requesting 50 Hz, get 50 Hz.
if requesting 100 Hz, get 83 Hz (was able to get up to 92 Hz by adjusting the loop rate within mavlink_main.cpp)
If requesting 250 Hz, get 160 Hz.
If requesting 500 Hz, get 220 Hz.

And so on... Is anyone else encountering these issues? I've already disabled a bunch of applications to get the CPU load lower and there's definitely enough bandwidth for these messages over the serial link.

LorenzMeier, 2017-03-31:
What is the output of the "mavlink status" command?

Seanmatthews, 2017-03-31:
I was just about to post the same issue...

@helenol @LorenzMeier From what I understand, PX4 is placing an artificial limit of 250Hz on certain (all?) sensors. In my case, I'm using the Snapdragon Flight board, which uses the Invensense MPU9250 IMU. It is 100% capable of higher sample rates, as listed in the data sheet, and demonstrated with Qualcomm's `imu_app`. If you `grep -rn 250HZ` at the repo's top level, you can see these artificial limits. I'm currently working on a quick fix, which I'll post here after I've tested it.

The slowdown to 160Hz and 220Hz that you're seeing is likely from system load.

Seanmatthews, 2017-03-31:
@helenol Pixhawk v2 FMU spec sheet seems to suggest that it uses the MPU9250 (same as Snapdragon Flight). I changed the magic number [here](https://github.com/PX4/Firmware/blob/master/src/platforms/posix/drivers/df_mpu9250_wrapper/df_mpu9250_wrapper.cpp#L618), from 4 to 2, in order to get 500Hz. Though, in reality I'm getting ~440Hz going through mavlink and mavros.

@LorenzMeier Any idea why it's like this? The comment doesn't provide much background: "The driver empties the FIFO buffer at 1kHz, however we only need to publish at 250Hz."

mhkabir, 2017-04-01:
@LorenzMeier @helenol I already reported this here : https://github.com/PX4/Firmware/issues/6235

mhkabir, 2017-04-01:
(Since 2015 : https://github.com/PX4/Firmware/issues/3291)

dagar, 2017-04-12:
Closing as duplicate.

closed 2017-04-12 by dagar