PX4/PX4-Autopilot #6947: Main LED not reflecting battery status 1.6.0 master

rolandash, 2017-03-31:
Main LED not reflecting battery status. Stay blue solid until battery drain and touch down. Should be red instead.

Vehicle get into wrong status of 'Lost of control' with LED blink purple after battery drained and touch down, remote controller became not responsible and not able to disarm the spinning propellers.

rolandash, 2017-05-03:
@dagar I would think this issue is more sever than an enhancement, as the LED is actually in wrong status, as it just never updating, may misleading pilots.

The RC-lost issue after touch down may better be in a seperate issue. But i will have a double check first to see if it still exists in master now.

dagar, 2017-05-03:
Do you have a log? Were the battery thresholds configured (BAT_LOW_THR, BAT_CRIT_THR, BAT_EMERGEN_THR)?

rolandash, 2017-05-03:
Yes i think they are configured. I will try to upload log tomorrow.

rolandash, 2017-05-14:
Sry for the late, here is the log. Purposely use a NanoMind frame to focus on battery status (3.7V 1S/650mah):
http://review.px4.io/plot_app?log=bbdc6bdd-c847-4f0e-88c5-5a2d2814547f

Seems that the original behavior of the issue has disappear, when battery reached critical level and LED turned to red(return to land advised, but not taken, continued to fly), but after that for a while then the LED back to blue.

Is that normal behavior? Still not seems correct.

davids5, 2017-06-30:
@dagare FYI
Bench testing this with power supply.

Because of the narrow differences in Emergency 5%, and Critical 7% It makes it looks, there is a bug that the battery Voltage has to rise after dropping before the annunciation of sound and light occurs.  But there is also a **bug** or **2**.

I adjusted my setting to test with

Emergency 10%, and Critical 20%, Low 40%
On Warning 0->1 I get slow beep/orange
On Warning 1->2 I get fast beep/red
**On  Warning 2->3 I get Silent/Blue**

On raising the voltage, 
**At 3 Silent/Blue**
On Warning 3->2 I get fast beep/**Blue**
On Warning 2->1 I get Slow beep/**Blue**
**Then we are stuck at 1 Slow beep/Blue**

The downward tracking may make sense in real life, as opposed to a bench test, but the at Warning 3 it should still indicate constant beep and red - think flat lined. If we also want to do upward tracking we should fix the lights.

battery_status.warning: never clears. I would suggest that we have battery_status be the actual status. and the state machine can treat it is sticky or not. For logging purposes, we could also consider adding a fault field to battery_status that is set on warning !=0 and never cleared until reset.

Also the nomenclature is open to misinterpretation:

The code makes sense:
```
void
Battery::determineWarning()
{
	// Smallest values must come first
	if (_remaining < _param_emergency_thr.get()) {
		_warning = battery_status_s::BATTERY_WARNING_EMERGENCY;

	} else if (_remaining < _param_crit_thr.get()) {
		_warning = battery_status_s::BATTERY_WARNING_CRITICAL;

	} else if (_remaining < _param_low_thr.get()) {
		_warning = battery_status_s::BATTERY_WARNING_LOW;
	}
}
```
But the display order in QGC is not the ranking order as in the code. 

[image]

airmind, 2017-07-01:
@davids5 thanks for the detailed tests @iZhangHui , that explains the issue.

closed 2018-02-21 by PX4BuildBot