PX4/PX4-Autopilot #6953: Copter starts descending in Follow Me mode

aysh-shariff, 2017-04-01:
I have added a small function in follow me mode for the copter to orbit a target when the target has stopped. However, the copter flies to the target and on detecting that it isn't moving starts descending. 
I am not sure if it was trying to land as I quickly changed to Position Control everytime this happened.
http://logs.px4.io/plot_app?log=a71d3f38-e9f1-412d-812b-9444b91aa5c5

Here is the log where the target is driving in circles and the copter is following it. http://logs.px4.io/plot_app?log=7986d696-3c83-4908-8d3e-0f6f18a71f24

When I looked closely at my SITL logs, I found the same behaviour.

These are the setpoints I am using for the orbiting.
 struct position_setpoint_triplet_s *pos_sp_triplet_circle = _navigator->get_position_setpoint_triplet();
       pos_sp_triplet_circle->previous.valid = false;
     pos_sp_triplet_circle->current.valid = true;
     pos_sp_triplet_circle->next.valid = false;
     pos_sp_triplet_circle->current.alt_valid = true;
     pos_sp_triplet_circle->current.alt= _param_min_alt.get();
     pos_sp_triplet_circle->current.type = position_setpoint_s::SETPOINT_TYPE_FOLLOW_TARGET;
     pos_sp_triplet_circle->current.lat =_new_wp.lat;
     pos_sp_triplet_circle->current.lon =_new_wp.lon;
     pos_sp_triplet_circle->current.yaw_valid= true;
     pos_sp_triplet_circle->current.yaw =heading;
     pos_sp_triplet_circle->current.velocity_valid=true;
     pos_sp_triplet_circle->current.vx = 15.0f;
     pos_sp_triplet_circle->current.vy = 10.0f;

I'm suspecting I have misunderstood some of them and using them incorrectly. I would really appreciate if someone helped me figure out the issue as I haven't got any response on the gitter channel.

Apologies if this is not the right place for this question.

mhkabir, 2017-04-02:
Custom code snippets like this are impossible to support. Let us know if there are any issues in vanilla Follow-Me mode.

aysh-shariff, 2017-04-04:
@mhkabir I'm pretty sure it is something to do with altitude setpoint. 
Is it not the right way to set the altitude in the position setpoint for Auto modes?

closed 2017-04-07 by aysh-shariff