PX4/PX4-Autopilot #6957: PX4FLOW LPE not working: Z is reversed

tomatac, 2017-04-02:
I am trying to get a stable position estimation using only PX4FLOW but it looks like the Z axis is reversed. 
"M1:OPTICAL_FLOW_RAD.distance" is very stable around 1 m (based on sonar)
"M1:LOCAL_POSITION_NED.z" variation is reversed relative to "M1:OPTICAL_FLOW_RAD.distance". See the graph below.
[image]

M1:OPTICAL_FLOW_RAD.quality is around 250
LPE_FUSION is 162
I also tried LPE_FUSION = 34 (no baro) and "M1:LOCAL_POSITION_NED.z" stays close to zero.
Firmware version 1.6.0 rc

mhkabir, 2017-04-02:
The graph is fine. Local Z in in NED, so it is reversed. You must have baro enabled, because there are some things hardcoded in the estimator.

tomatac, 2017-04-02:
Thanks! I got it, because the z is represented in NED (north-east-down) coordinates system the z value is reversed. Is this NED coordinate system "frame NED" or "earth NED"? 
Then the main issue is that the M1:LOCAL_POSITION_NED.x and M1:LOCAL_POSITION_NED.y are not consistent. They stay around zero when I walk and the quadcopter few meters forward/reverse and left/right. I do not know how to troubleshoot this issue. 
The goal is to get a stable position estimation to be able to go in offboard control mode and to be able to control the quadcopter based on a vision system that runs on a companion computer.

jgoppert, 2017-04-02:
The frame is north, east, down with the projection following the vehicle. The z origin is sea level.

The reason that x and y don't move while you are walking is that land_detector is enabled. You can disable it in (LPE_FUSION) within qgroundcontrol and test it by walking, but then set it back on before you fly. While landed (the logic is throttle below ~40%, or disarmed, and not rotating), it prevents drift when the optical flow is not getting good data and will allow takeoff. There is a hand off when you takeoff from the land detector to the optical flow. If the optical flow doesn't get good data after takeoff, it will attempt to land if in auto, or fallback to altitude control mode if in position control.

tomatac, 2017-04-02:
Thank you! I will try that.
So when flying higher the LOCAL_POSITION_NED.z should always be negative.
Is LPE_FUSION = 162 the recommended setting for flying LPE with PX4FLOW only?
Should "flow gyro compensation" be OFF for LPE_FUSION?

tomatac, 2017-04-02:
What parameters should I monitor to make sure that the hand off from takeoff to the optical flow is going well? Is that happend at a fixed altitude?

tomatac, 2017-04-02:
@jgoppert I tried setting the land detector OFF in LPE_FUSION but it did not make a difference. See the graphs below:
1. LOCAL_POSITION_NED.x and M1:LOCAL_POSITION_NED.y walking the quadcopter 2 meters south-north back to the origin and then east-west and back to the origin.
[image]

2. For reference this graph is recorded directly connected to the flow camera and walking the quadcopter front-back and left-right.
[image]

In fact the LPE_FUSION = 162 means that the "fuse land detector" is off
[image]
Is there another means to turn on / off the land detector?
Do you have any troubleshooting suggestions based on the graphs shown above?

mhkabir, 2017-04-02:
I would suggest enabling land detector and gyro compensation and giving it a test-fly in position control. It should work fine.

jgoppert, 2017-04-02:
I always use flow gyro compensation. Make sure the camera is in focus as well.

tomatac, 2017-04-02:
Thanks! I will turn the flow gyro compensation on. It makes sense. I was just following other recommendations on github.
Is it a good test to align the vehicle facing north and move it forward/backward and observe positive/negative LOCAL_POSITION_NED.x changes, respectively and do the same moving it right/left for positive/negative LOCAL_POSITION_NED.y changes? Is this a good pre-flight test?
I will also try to give it a test flight after that in position control.

mhkabir, 2017-04-02:
Yes.

jgoppert, 2017-04-02:
Yes, you should be able to do that as long as loan detector is off. And
that is what I typically do when running new code or in a new environment
to make sure flow is working.

tomatac, 2017-04-02:
Thanks! Is there a way to confirm that the land detector is off other than turning the bit off in LPE_FUSION?

tomatac, 2017-04-02:
@jgoppert are there other parameters that I should be watching during the pre-flight test?

jgoppert, 2017-04-02:
Make sure you local z looks reasonable. Also, to test px4flow by itself you can just look at local_position topic on px4flow (when plugged into px4flow usb). This estimates position just using sonar on px4flow and the camera. It doesn't have a compass so this is only good for testing (doesn't know if you rotate around z), but I used it all the time to make sure flow is working well independent of the estimator.

tomatac, 2017-04-02:
@jgoppert, thank you! That makes sense. I tried that, but the question is, when you test the px4flow by itself are looking at x and y or only at z? Is there a x and y parameter that represents the distance moved from the origin? My OPTICAL.FLOW.flow_x and OPTICAL.FLOW.flow_y do not quite make sense (see the graph above).

jgoppert, 2017-04-02:
@tomatac OPTICAL_FLOW.flow_x is confusing for a human to look at. Try looking at LOCAL_POSITION.x, y, z. You can see this by enabling the parameter send_lpos on the px4flow.

tuofeichen, 2017-04-03:
@jgoppert  A quick question here. Is there a way to debug px4flow with a tera ranger (or separate range finder than the native sonar? ).

jgoppert, 2017-04-03:
@tuofeichen I typically use a lidar with flow and LPE. I still can debug the flow using its onboard sonar. If that works, the other sensor should give you similar results or better.

tuofeichen, 2017-04-03:
@jgoppert yeah I'm just wondering if there's a quick way to directly look at the LOCAL_POSITION topic of the flow even with a different rangefinder. (I desoldered the sonar). the flow probably need to read the i2c bus  in that case, not super critical, no worries!

jgoppert, 2017-04-03:
@tuofeichen Yeah, it is not possible out of the box if you have desoldered it.

jgoppert, 2017-04-03:
@tomatac Are you good closing this now? If you have any other questions we can move it to http://discuss.px4.io/.

tomatac, 2017-04-03:
I got more data I just recorded per your instructions. I will post in few minutes. The PX4FLOW data looks good but I think that the information is not processed correctly by the PX4 firmware.

jgoppert, 2017-04-03:
@tomatac Please use the new logger, SYS_LOGGER=1, and upload a .ulg file to http://review.px4.io/  The new log format contains a lot more info that is helpful for me to debug your issues.

tomatac, 2017-04-03:
I connected to the PX4FLOW and set the SYS_SEND_LPOS to 1. That allows the PX4FLOW to temporarily publish LOCAL_POSITION_NED.x y and z values. Great debugging tool! When I walked the quadcopter back and forth couple of meters and the result looks good:
[image]
Then I disconnected the micro USB cable from the PX4FLOW and connected to the Pixhawk.
In the same conditions, walking the quadcopter back and forth couple of meters on north-south direction, the results do not make sense:
[image]

The graph below is recorded  with the quadcopter just sitting on the ground, and it does not make sense either:
[image]

I think this is a bug in the PX4 firmware because the values we get from the PX4FLOW are good.
The issue should be renamed to just: "PX4FLOW LPE not working" (the z is not the problem)

I also changed the LPE_FUSION to 226 (turned the "flow gyro compensation" on) as per your recommendation. 

Thank you for your help! Do you have any other troubleshooting suggestions (try a different firmware version, ...)?

tomatac, 2017-04-03:
I will record a log and post it. Thanks for your help!

tomatac, 2017-04-04:
@jgoppert see below the link to the log file. In this test I walked the quadcopter south-north and east-west 2 meters in each direction. That is not reflected by the X Y coordinates from the log. 
http://review.px4.io/plot_app?log=f6f06538-2452-4e91-adfb-a6e507d4298f
I set the SYS_LOGGER to "new logger" in the QGroundControl but I get ".px4log" log files instead of ".ulg" files.
Is the new logger available in the px4fmu-v2_lpe.px4 firmware?
Do you have any other troubleshooting suggestions? Thanks!

tomatac, 2017-04-05:
If the graphs (http://review.px4.io/plot_app?log=f6f06538-2452-4e91-adfb-a6e507d4298f) do not gives us an indication on what might be wrong I would like to try a previous firmware that works with PX4FLOW/LPE? I tried 1.5.5 (nuttx-px4fmu-v2-default.px4) and it did not work. What is good stable version for PX4FLOW/LPE? My plan is, if I get the previous version working, then come back and see what is wrong in 1.6.

jgoppert, 2017-04-05:
I flew master a week ago, so I know that it is probably still  working.

tomatac, 2017-04-05:
Ok. Thanks! Do you have any troubleshooting suggestion? How do we know if the LPE is running based on PX4FLOW and it is not stuck in takeoff mode?

jgoppert, 2017-04-05:
If you can't get it to work with land_detector disabled and walking around, it is best not to try flying it. This is what I do to troubleshoot before I fly. If it is drifting a lot, check your flow quality, it should be around 250. If you can't get your quality that high, you might need lights or try the px4flow firmware branch exposure_for_indoor.

tomatac, 2017-04-06:
See below the link to a real flight: south and then east - west (2-3 meters)
http://review.px4.io/plot_app?log=a38644bb-865c-4b67-bdb8-e1bebd7aaca9
The position control was rejected.

jgoppert, 2017-04-06:
I had a look at your log.

Flow quality is good:
[image]

The sonar from the flow isn't publishing the distance to the px4. It must be the primary distance driver, see code here: https://github.com/PX4/Firmware/blob/master/src/drivers/px4flow/px4flow.cpp#L240

Also, the sonar readings themselves look very poor:
[image]

Also, it looks like you have a lot of vibrations, you might need to balance your props or something. Vibrations can also degrade sonar performance.

[image]

tomatac, 2017-04-06:
Thank you! Is the poor sonar readings the cause for sonar not publishing the distance to the px4? I followed the C++ code but I could not determine the possible cause.
Anyway I am going to purchase a lidar sensor. The quality on the sonar does not look good. Does LIDAR-Lite v3 integrate well with the PX4?
How did you get the flow quality graphs form the log file? I tried FlightPlot and it gave me an error loading the log file. I was in process of trying PX4Tools. Thank you again for your help!

jgoppert, 2017-04-06:
Yeah I use px4tools or pyulog in jupyter notebook. Here is an example just using pyulog.

[Untitled.pdf](https://github.com/PX4/Firmware/files/901998/Untitled.pdf)

tomatac, 2017-04-08:
Thanks! I am new to jupiter notebooks but after I read about them and I was planning to start using them in a Docker container (https://hub.docker.com/r/jupyter/). Do you have one that you recommend? Do you know if there is a Docker image that contains the pyulog and px4tool already?
I got the lidar lite and if everything goes ok we should fly it tomorrow.

tomatac, 2017-04-10:
@jgoppert I got the Lidar Lite 3 installed. The altitude control looks a lot better and the x y position graph looks better. Still the position control is rejected. Here it is one of the flight logs:
http://logs.px4.io/plot_app?log=95cb6bde-d118-4317-aa7b-ca82151c903e
I also got the pyulog on Jupiter notebook working and I can reproduce the graphs that you shared in the pdf file. It would be nice if I can create a notebooks specific for troubleshooting the LPE. I already made a Docker file that is really easy to use and I would like to share the results once I get this working. I am still learning how to extract some of the log parameters (any examples or tips will help here).
The main question remains : Why is the POSITION CONTROL REJECTED? Thanks again for your help!

tomatac, 2017-04-11:
I tried few more flights but I have the same issue: POSITION CONTROL REJECTED.
See the log below and the file attached:
http://logs.px4.io/plot_app?log=3f14be80-3596-4043-8aa6-dacb749d2b08
[LPE_Flight_04-11.pdf](https://github.com/PX4/Firmware/files/912470/LPE_Flight_04-11.pdf)
Is there a way to graph the input from the Lidar sensor?
How do I make sure that the LPE considers only the Lidar input and not the sonar sensor?
I still have LPE_FUSION:fuse_land_detector OFF. Can this be the reason the position control is rejected?
Do you have any other troubleshooting suggestions? Thanks!

tomatac, 2017-04-12:
To help debug this issue, can somebody share a log file of a successful flight where position control was achieved (PX4FLOW + Lidar-Lire, NO GPS). I am looking to compare the parameters and the flight graphs to find out why we are now having issues. I can develop a jupyter notebook to compare the key parameters between logs side by side to make troubleshooting easier.

jgoppert, 2017-04-13:
Sorry I haven't been more responsive, have been on a business trip. I'll
take a look soon and share one of my good logs.

tomatac, 2017-04-13:
@jgoppert Thank you! I learned a lot from this discussion. At issue #7002 @mhkabir mentioned that PWM for Lidar Lite is currently broken". I am not sure what version is he referring to. Do you use the Lidar Lite 2 or 3? I understood that is not a big difference between them.  It would be great to establish a version that works and then see what commit broke the functionality.

nancy03610, 2017-04-13:
@jgoppert Sorry, I have a quick question here. I also use LPE+PX4Flow+sonar, and got drift problem maybe because of the same poor sonar reading. Is it possible to make it work with sonar? Would you please also share a log of PX4Flow+Sonar? Thank you very much!

tomatac, 2017-04-13:
@nancy03610 and @jgoppert it would be nice to establish a baseline that tells us in what conditions the sonar gives acceptable results. From my testing it really depends on the ground surface. Indoors, with propellers not running the tile flooring get very good results but the carpet gives very poor results. It looks like vibrations and air flow also affects the results. @jgoppert pointed out how poor the my sonar signal was in the graph above and I decided to buy the Lidar Lite (Garbage in, garbage out, I cannot expect magic from the software).

jgoppert, 2017-04-13:
Please see this old PR https://github.com/PX4/Firmware/pull/4806. This is
the last time that I tested with sonar and at that point it was working
well.

Video:
https://www.youtube.com/watch?v=t7s8027uKZ4

Log:
http://logs.uaventure.com/view/beRXQ3uES9ctPkv2H5ymDQ

tomatac, 2017-04-13:
@jgoppert that was impressive! The sonar input was not very good on grass but you still achieved a nice stable flight. The raw accel was low.

tomatac, 2017-04-13:
@jgoppert do you use a PWM lidar now? Is it the Lidar-Lite? Now that I have the Lidar installed I prefer to use it. Could you please post a similar log for a Lidar based flight along with a pointer to the firmware version that worked? I would to start there, confirm that I can reproduce the results and then we can try to see at what point the present master got broken.

jgoppert, 2017-04-13:
I'm now using an sf10a on serial from lightware. I haven't flown without vision recently, so give me some time to get a flight in.

tomatac, 2017-04-13:
@jgoppert Do you know if the PWM on the Lidar-Lite ever worked well in a specific commit? Also, when you mentioned vision, is that a companion computer that runs vision? Thank you, I really appreciate your help!

jgoppert, 2017-04-21:
The last time I tested that config personally was about a year ago but I'm pretty confident that it is fine. @ecmnet any input? Yes I'm having a companion computer send the vision updates.

tomatac, 2017-04-21:
@jgoppert tanks. It looks like the pwm was broken on the master. I will watch for an update, try older firmware and report back. Now I am traveling so I will test in about a week.

jgoppert, 2017-04-28:
Let's close this and move to http://discuss.px4.io/

closed 2017-04-28 by jgoppert