PX4/PX4-Autopilot #6961: Camera Trigger

mhkabir, 2017-04-03:
There are several outstanding issues with the camera trigger which might be causing very weird and hard-to-track problems due to direct hardware access.

1. 
> Scenario on Pixhawk hardware:
> 
> Select Trigger mode "Time based always on"
> Select Triggrer interface "MAVLINK ..."
> After powercycle the autopilot does not boot.
> Only workaround I found is to flash it with Ardupilot and then back to PX4.
> 
> Reproducable on master@3961c46

2. https://github.com/PX4/Firmware/issues/6839

> For some reason the trigger is not activated.
> 
> In the QGC section for camera, it was by default set as PWM emitted on port AUX 5 and 6.
> Considering I was need just 1 port (the NEX camera is activated by IR sensor, which work with PWM) to trigger the camera, I disabled the AUX5 and let the AUX6 on. For some reason the AUX6 has never emitted a signal during the whole flight, and the log confirm, nothing came out on port AUX6.

3. https://github.com/PX4/Firmware/issues/6839

> Any chance to set a PWM output for triggering a camera on an MAIN port instead of AUX to make it work?

4. https://github.com/PX4/Firmware/issues/6703

> Configured PX4 on Pixhawk to do Time based trigger always on with:
> TRIG_ACT_TIME: 500ms
> TRIG_INTERVAL: 1000ms
> TRIG_PINS: 123456
> Expected to have equal PWM outputs on all 6 AUX ports.
> This is what I get:
> With RC transmitter ON:
> AUX1: constant 1500 regardless of what is configured in Radio setup.
> AUX2: constant 1500 regardless of what is configured in Radio setup.
> AUX3: constant 1500 regardless of what is configured in Radio setup.
> AUX4: constant 1500 regardless of what is configured in Radio setup.
> AUX5: No signal
> AUX6: No signal
> With RC transmitter OFF:
> AUX1: 1500 for 0.5s, 1700 for 0.5s (expected behavior)
> AUX2: constant 1500
> AUX3: 1500 for 0.5s, 1700 for 0.5s (expected behavior)
> AUX4: constant 1500
> AUX5: No signal
> AUX6: No signal
> 
> Turns out that I can get camera trigger working only while RC is turned OFF and only on two pins which does not make sense.

5. https://github.com/PX4/Firmware/issues/6092

> I've been testing the camera trigger and haven't got it working on Pins 5&6 of the FMU (TRIG_PINS = 56). I've also tried the setting TRIG_PINS = 34 which works fine.
> I'm wondering if there was a bug introduced somewhere because the default pins are in fact 5,6.

6. https://github.com/PX4/Firmware/issues/4941

> The current approach of accessing PWM pins directly is suboptimal as it can conflict with the FMU driver's management of GPIO and PWM outputs. It would be good to coordinate on how to properly layer this.

7. https://github.com/PX4/Firmware/issues/5437

8. https://github.com/PX4/Firmware/issues/5530

The driver needs to ACK the TRIGGER_CONTROL command so that mavros knows the call succeeded.

mhkabir, 2017-04-03:
Open Pull Requests : 

1. https://github.com/PX4/Firmware/pull/6194

tops4u, 2017-04-03:
#6260 related to this too?

mhkabir, 2017-04-03:
@jdzerve Can you give us more information to reproduce the issue? I just checked, and I can't get it to fail boot.

mhkabir, 2017-04-04:
Everyone having issues with the camera trigger please check this PR. It should all be fixed : https://github.com/PX4/Firmware/pull/6973

jdzerve, 2017-04-05:
@mhkabir Sorry, will be able to get to this no sooner than end of next week.

mhkabir, 2017-04-13:
Fixed! Everything is in master now.

closed 2017-04-13 by mhkabir