PX4/PX4-Autopilot #6962: Snapdragon Flight ESC support

mhkabir, 2017-04-03:
1. Qualcomm UART ESC support : 

PX4 does not generally support these boards due to limited availability and initially high price. The only drivers for the Qualcomm ESCs are the legacy (binary) drivers. There may have open versions in the future but not currently, so the `eagle_legacy_default` target needs to be used to use those ESCs. All the drivers were initially binary thus the "legacy" name. You can compare `cmake/configs` for the `qurt_sdflight_default.cmake`(full open-source build) and `qurt_sdflight_legacy.cmake`(mixed build, with binary drivers) to see the differences. You can add any modules you like but will need the platforms `/qurt/fc_addon/uart_esc` module for the Qualcomm ESC.

2. PWM ESC support : 

PX4 Master already supports the use of PWM ESCs

ChristophTobler, 2017-04-03:
@mhkabir thanks! I'll do the documentation as soon as possible

ChristophTobler, 2017-04-04:
@mhkabir here's the PR https://github.com/PX4/Devguide/pull/108

alexandrelheinen, 2017-05-02:
I am having a similar issue:

I am trying to implement a PX4-based autopilot for a fixed-wing UAV. It turns out that the easiest alternative is to use the 4 PWMs output on port J13 which uses `snapdragon_pwm_out` module (https://github.com/PX4/Firmware/tree/master/src/drivers/snapdragon_pwm_out).

As we can observe in the source code, it tries to parse the mixer file into a `MultirotorMixer` object. In practice, I am having the following error from the mini-dm
```
[08500/00]  01:50.030  HAP:34:Trying to initialize mixer from config file /dev/fs/myfixedwing.main.mix  0166  snapdragon_pwm_out.cpp
[08500/03]  01:50.032  HAP:34:Unable to parse from mixer config file  0181  snapdragon_pwm_out.cpp
[08500/03]  01:50.032  HAP:34:Mixer initialization failed.  0337  snapdragon_pwm_out.cpp
```
even if I use the example mixer file given in the DevGuide (https://dev.px4.io/airframes-adding-a-new-frame.html).

Does it have any link with the class `MultirotorMixer`? Could I initialize the fixed-wing mixer without using a `.mix` file, setting up the `MultirotorMixer` directly?

marcelinomalmeidan, 2017-06-14:
Hi,

Was the documentation made for using pwm ESCs? If so, where do I find it?

Thanks!

mhkabir, 2017-06-14:
https://dev.px4.io/en/flight_controller/snapdragon_flight.html

marcelinomalmeidan, 2017-06-14:
Thanks! I had seen that and I am aware of the wiring.
I just don't know how to setup the mainapp.config to use PWM ESCs instead of UART. Would you know that?

Thanks!

closed 2017-10-04 by LorenzMeier