PX4/PX4-Autopilot #6967: JMavSim not launching with fresh copy from master

crashmatt, 2017-04-04:
Fresh copy of repo:
https://github.com/PX4/Firmware.git

Do
make posix_sitl_default jmavsim

Results in 
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..

Diff with old working copy shows that JMavSim has moved or has gone.  Many other libraries appear to be missing or not built.

bkueng, 2017-04-04:
```
git clone --recursive https://github.com/PX4/Firmware.git
cd Firmware
make posix_sitl_default jmavsim
```
works fine for me. Did you clone the submodules?

crashmatt, 2017-04-04:
@bkueng I normally use git gui and recursive clone option.
I tried your cmd line version but it does the same

Now I found the error:
/home/matt/src/Firmware/Tools/jmavsim_run.sh: line 44: ant: command not found

Ant is not available on my machine might answer why it can't build jmavsim.  The older repo clone already has jmavsim built so it doesn't have this problem.
My linux machine has gone through a few updates recently.  Ant may have disappeared.  I will investigate and probably close.

crashmatt, 2017-04-04:
@bkueng That was it.  Ant was missing. I don't know what ant belongs to or if it has been included in the toolchain dependencies.  Good enough for now :-)

closed 2017-04-04 by crashmatt