PX4/PX4-Autopilot #6982: Failsafe mode on while in OFFBOARD mode

L4ncelot, 2017-04-05:
Hi,

I've recently encountered some problem while using **OFFBOARD** mode. I'm using **px4** firmware with **MAVROS**.

I've been developing some algorithms for autonomous flying. I'm using basically code from [this](https://dev.px4.io/ros-mavros-offboard.html) documentation. Some time ago there was no problem. But recently I've seen my copter loosing connection or something like that.

Here's example of my log during flight:
```
[ INFO] [1491397206.779180203, 162.918000000]: current mode: OFFBOARD
[ WARN] [1491397207.327585010, 163.442000000]: PR: request param #597 timeout, retries left 2, and 36 params still missing
[ INFO] [1491397207.790462424, 163.870000000]: FCU: no offboard
[ERROR] [1491397207.792956594, 163.884000000]: FCU: failsafe mode on
[ INFO] [1491397207.838462636, 163.610000000]: current mode: OFFBOARD
[ERROR] [1491397207.874603512, 163.958000000]: FCU: failsafe mode off
[ INFO] [1491397208.059659872, 164.132000000]: current mode: OFFBOARD
[ WARN] [1491397208.434182282, 164.474000000]: PR: request param #598 timeout, retries left 2, and 35 params still missing
[ INFO] [1491397208.630911259, 164.652000000]: current mode: OFFBOARD
[ INFO] [1491397209.184027414, 165.156000000]: current mode: OFFBOARD
[ WARN] [1491397209.580750522, 165.542000000]: PR: request param #599 timeout, retries left 2, and 34 params still missing
[ INFO] [1491397209.727710950, 165.678000000]: current mode: OFFBOARD
[ INFO] [1491397210.286440749, 166.200000000]: current mode: OFFBOARD
[ WARN] [1491397210.675576552, 166.550000000]: PR: request param #601 timeout, retries left 2, and 33 params still missing
[ INFO] [1491397210.838900798, 166.702000000]: current mode: OFFBOARD
[ INFO] [1491397211.409235549, 167.222000000]: current mode: OFFBOARD
[ INFO] [1491397211.967874921, 167.730000000]: current mode: OFFBOARD
[ WARN] [1491397212.474896209, 168.194000000]: PR: request param #602 timeout, retries left 2, and 32 params still missing
[ INFO] [1491397212.529351061, 168.246000000]: current mode: OFFBOARD
[ INFO] [1491397213.089556780, 168.772000000]: current mode: OFFBOARD
[ INFO] [1491397213.663550895, 169.282000000]: current mode: OFFBOARD
[ INFO] [1491397213.712743604, 168.854000000]: FCU: no offboard
[ERROR] [1491397213.719005538, 168.854000000]: FCU: failsafe mode on
[ WARN] [1491397213.970398260, 169.298000000]: PR: request param #603 timeout, retries left 2, and 31 params still missing
[ERROR] [1491397214.050234544, 169.530000000]: FCU: failsafe mode off
[ INFO] [1491397214.275384109, 169.808000000]: current mode: OFFBOARD
[ INFO] [1491397214.849830352, 170.334000000]: current mode: OFFBOARD
[ WARN] [1491397214.885008098, 170.366000000]: PR: request param #605 timeout, retries left 2, and 30 params still missing
[ INFO] [1491397215.436646707, 170.854000000]: current mode: OFFBOARD
[ WARN] [1491397215.987466251, 171.370000000]: PR: request param #606 timeout, retries left 2, and 29 params still missing
[ INFO] [1491397215.987573803, 171.372000000]: current mode: OFFBOARD
[ INFO] [1491397216.590532094, 171.906000000]: current mode: OFFBOARD
[ INFO] [1491397217.134760945, 172.408000000]: current mode: OFFBOARD
```

The message from **px4** firmware tells me that there's no **OFFBOARD** but at the same time when I check state from **MAVROS**, it says there is **OFFBOARD**.

So I have no idea what's going on and if it's my fault or not. Has anybody any idea?

Thank you for any advice.

mhkabir, 2017-04-05:
Your wiring is bad, most likely. Check connections. The parameter pull also keeps failing.

L4ncelot, 2017-04-05:
How should I test the connection? (srry, I'm network noob) I have px4 on navio which is connected through UTP cabel to my PC where simulation in Gazebo runs.

mhkabir, 2017-04-05:
Are you doing HIL ?

L4ncelot, 2017-04-05:
Exactly. I'm running Gazebo simulation on my PC and px4 firmware and some controllers on navio2.

dagar, 2017-04-12:
@L4ncelot any update?

L4ncelot, 2017-04-12:
Actually not, I've been working on something in SITL mode right now. I will have a look at this issue probably next week or so.

TSC21, 2017-08-07:
@L4ncelot updates?

L4ncelot, 2017-08-08:
@TSC21 Well, we're now testing everything with real copter and it seems like some parameters are still missing. I'll post here log from **mavros** when I fix the GPS module.

TSC21, 2017-08-08:
I still think there is some problem with your link.

L4ncelot, 2017-08-14:
So here's the log:

```
[0m[ INFO] [946685723.833591532]: FCU URL: /dev/ttyTHS3:57600[0m
[0m[ INFO] [946685723.834025280]: serial0: device: /dev/ttyTHS3 @ 57600 bps[0m
[0m[ INFO] [946685723.835192109]: GCS bridge disabled[0m
[0m[ INFO] [946685723.926340542]: Plugin 3dr_radio loaded and initialized[0m
[0m[ INFO] [946685723.936071166]: Plugin actuator_control loaded and initialized[0m
[0m[ INFO] [946685723.939189985]: Plugin altitude loaded and initialized[0m
[0m[ INFO] [946685723.952202678]: Plugin command loaded and initialized[0m
[0m[ INFO] [946685723.969688185]: Plugin ftp loaded and initialized[0m
[0m[ INFO] [946685723.989922763]: Plugin global_position loaded and initialized[0m
[0m[ INFO] [946685723.991661422]: Plugin hil_controls loaded and initialized[0m
[0m[ INFO] [946685724.010133424]: Plugin imu_pub loaded and initialized[0m
[0m[ INFO] [946685724.026097020]: Plugin local_position loaded and initialized[0m
[0m[ INFO] [946685724.027811263]: Plugin manual_control loaded and initialized[0m
[0m[ INFO] [946685724.034268734]: Plugin param loaded and initialized[0m
[0m[ INFO] [946685724.044337524]: Plugin rc_io loaded and initialized[0m
[0m[ INFO] [946685724.044415607]: Plugin safety_area blacklisted[0m
[0m[ INFO] [946685724.054054231]: Plugin setpoint_accel loaded and initialized[0m
[0m[ INFO] [946685724.085599509]: Plugin setpoint_attitude loaded and initialized[0m
[0m[ INFO] [946685724.101591772]: Plugin setpoint_position loaded and initialized[0m
[0m[ INFO] [946685724.127222160]: Plugin setpoint_raw loaded and initialized[0m
[0m[ INFO] [946685724.134859376]: Plugin setpoint_velocity loaded and initialized[0m
[0m[ INFO] [946685724.150327891]: Plugin sys_status loaded and initialized[0m
[0m[ INFO] [946685724.158914187]: Plugin sys_time loaded and initialized[0m
[0m[ INFO] [946685724.162196839]: Plugin vfr_hud loaded and initialized[0m
[0m[ INFO] [946685724.172529627]: Plugin waypoint loaded and initialized[0m
[0m[ INFO] [946685724.172619794]: Autostarting mavlink via USB on PX4[0m
[0m[ INFO] [946685724.172693710]: Built-in SIMD instructions: None[0m
[0m[ INFO] [946685724.172752876]: Built-in MAVLink package version: 2016.5.20[0m
[0m[ INFO] [946685724.172793126]: Built-in MAVLink dialect: ardupilotmega[0m
[0m[ INFO] [946685724.172830876]: MAVROS started. MY ID 1.240, TARGET ID 1.1[0m
[0m[ INFO] [946685724.185180655]: IMU: High resolution IMU detected![0m
[0m[ INFO] [946685724.185353988]: IMU: Attitude quaternion IMU detected![0m
[0m[ INFO] [946685724.185528654]: CON: Got HEARTBEAT, connected. FCU: PX4[0m
[0m[ INFO] [946685724.185716236]: RC_CHANNELS message detected![0m
[0m[ INFO] [946685724.186027401]: IMU: High resolution IMU detected![0m
[0m[ INFO] [946685724.186106234]: IMU: Attitude quaternion IMU detected![0m
[33m[ WARN] [946685724.307810700]: TM: Clock skew detected (-946684792.540672779 s). Hard syncing clocks.[0m
[0m[ INFO] [946685725.310208965]: VER: 1.1: Capabilities 0x00000000000024ef[0m
[0m[ INFO] [946685725.312124456]: VER: 1.1: Flight software:     01060500 (f160743df40b9dcc)[0m
[0m[ INFO] [946685725.313268368]: VER: 1.1: Middleware software: 01060500 (f160743df40b9dcc)[0m
[0m[ INFO] [946685725.314279280]: VER: 1.1: OS software:         000000c0 (0000000000000000)[0m
[0m[ INFO] [946685725.316209438]: VER: 1.1: Board hardware:      00000011[0m
[0m[ INFO] [946685725.316660936]: VER: 1.1: VID/PID: 26ac:0011[0m
[0m[ INFO] [946685725.325293065]: VER: 1.1: UID: 3335511239383934[0m
[33m[ WARN] [946685725.333981277]: CMD: Unexpected command 520, result 0[0m
[33m[ WARN] [946685734.996035849]: TM: Clock skew detected (-0.258869270 s). Hard syncing clocks.[0m
[33m[ WARN] [946685740.190198790]: WP: timeout, retries left 2[0m
[0m[ INFO] [946685740.221787651]: WP: mission received[0m
[0m[ INFO] [946685740.936590179]: WP: seems GCS requesting mission[0m
[0m[ INFO] [946685740.939179084]: WP: sheduling pull after GCS is done[0m
[33m[ WARN] [946685742.111687185]: PR: request param #9 timeout, retries left 2, and 174 params still missing[0m
[33m[ WARN] [946685743.271513092]: PR: request param #12 timeout, retries left 2, and 173 params still missing[0m
[33m[ WARN] [946685744.331501021]: PR: request param #13 timeout, retries left 2, and 172 params still missing[0m
[33m[ WARN] [946685745.195508893]: TM: Clock skew detected (0.013241412 s). Hard syncing clocks.[0m
[33m[ WARN] [946685745.390775619]: PR: request param #22 timeout, retries left 2, and 171 params still missing[0m
[0m[ INFO] [946685745.966209592]: WP: mission received[0m
[33m[ WARN] [946685746.465360151]: PR: request param #26 timeout, retries left 2, and 170 params still missing[0m
[33m[ WARN] [946685747.524078585]: PR: request param #30 timeout, retries left 2, and 169 params still missing[0m
[33m[ WARN] [946685748.578493288]: PR: request param #33 timeout, retries left 2, and 168 params still missing[0m
[33m[ WARN] [946685749.637561887]: PR: request param #37 timeout, retries left 2, and 167 params still missing[0m
[33m[ WARN] [946685750.703079208]: PR: request param #50 timeout, retries left 2, and 166 params still missing[0m
[33m[ WARN] [946685751.773283009]: PR: request param #53 timeout, retries left 2, and 165 params still missing[0m
[33m[ WARN] [946685752.830702199]: PR: request param #56 timeout, retries left 2, and 164 params still missing[0m
[... cut ...]
[33m[ WARN] [946685825.273250002]: PR: request param #342 timeout, retries left 2, and 96 params still missing[0m
[33m[ WARN] [946685826.323242308]: PR: request param #346 timeout, retries left 2, and 95 params still missing[0m
[33m[ WARN] [946685826.698736409]: TM: Clock skew detected (-0.061138400 s). Hard syncing clocks.[0m
```

The communication works fine, but the I don't get why the request params are present. The wiring should be fine in my opinion. I can control the copter using MAVROS, so it's not a big deal, it's just weird.

TSC21, 2017-08-14:
`�[0m[ INFO] [946685723.833591532]: FCU URL: /dev/ttyTHS3:57600�[0m`

Why don't you increase the link baud rate to 921600? `param set SYS_COMPANION 921600` on nsh or through QGC.

L4ncelot, 2017-08-14:
@TSC21 Ok, that solved the problem, thank you.

closed 2017-08-14 by L4ncelot