PX4/PX4-Autopilot #6984: FTP read file timeouts

lovettchris, 2017-04-05:
I cannot reliably read a file from px4 log directory using mavlink ftp get command.  I get a few packets, then nothing, even with a retry.  Same from QGC.  So I was playing around and found that adding a 100 microsecond sleep in the _workRead method of mavlink_ftp.cpp makes it work perfectly... weird huh...

See [this change](https://github.com/lovettchris/Firmware/commit/dd6ef20d3a4f912dc20cfc9c22e4adafd70314eb).

I'm using Pixhawk (px4fmu_v2_default).

bkueng, 2017-04-06:
This could be a concurrency issue: `MavlinkFTP::handle_message` and `MavlinkFTP::send` are both called from different threads and access class members. The mavlink mission class has the same issue and maybe others too. It needs a proper evaluation and maybe a re-design, because at the moment it is really not clear that they're accessed concurrently.
Making each class thread-safe is probably not solving it either, because both send mavlink messages, which in turn modifies a shared state.

LorenzMeier, 2017-04-09:
@bkueng Can you look into this? The mission upload fails that result are annoying.

dagar, 2017-04-09:
FYI - existing mavlink ftp test that's currently broken. https://github.com/PX4/Firmware/tree/master/src/modules/mavlink/mavlink_tests

iZhangHui, 2017-04-10:
@lovettchris You can try to increase stack size [here](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L2456) to 
pthread_attr_setstacksize(&receiveloop_attr, PX4_STACK_ADJUSTED(**2400**));

LorenzMeier, 2017-04-30:
@bkueng Any update on this?

bkueng, 2017-05-02:
No, I did not look into that yet.

LorenzMeier, 2017-08-20:
Fixed by @bkueng on master.

closed 2017-08-20 by LorenzMeier