PX4/PX4-Autopilot #6994: PWM lidar ll40ls not working anymore

ChristophTobler, 2017-04-07:
@kd0aij @julianoes ll40ls pwm does not work anymore after this https://github.com/PX4/Firmware/commit/aa9fbbedd54a7a0e346ea03f439dcdf4f90c6113

Any ideas why?

mhkabir, 2017-04-07:
I think I know why. It is potentially similar to why camera trigger got broken. @davids5

LorenzMeier, 2017-04-07:
@mhkabir Could you please add this to your list?

davids5, 2017-04-07:
@ChristophTobler - did you test after https://github.com/PX4/Firmware/commit/8e217b028718fb6f0196e8a42ccff026a82c3215 came in?

@mhkabir here is more [STM32 code in the wrong place](https://github.com/PX4/Firmware/blob/master/src/drivers/pwm_input/pwm_input.cpp) perhaps you can remove all the timer code and refactor the PWMIn into the io_timer_driver - the stubs and resource allocation is there - just not the guts for PWMIn. Another possibility is the io_timer_driver Capture API. Things like "px4_arch_gpioxxxx" can live outside of /drivers/stm32 ut direct HW manipulations of Timer IP block can not. 

Also see https://github.com/PX4/Firmware/pull/6993 and please test the PX4IO is running on V3 HW if after any changes you make add RAM usage to the common shared io timer code.

julianoes, 2017-04-08:
@davids5 this setup is on the Pixracer without PX4IO by the way.

mhkabir, 2017-04-08:
@julianoes io_timer is just a nice API which David wrote to allow shared
access to the timer hardware on the STM32. e.g camera trigger running on
AUX (FMU) pins uses it.

It is not specific to the px4io, or something like that :)

davids5, 2017-04-08:
@julianoes Good to know, for testing. The comment about testing on PX4IO is because the io_timer code is a shared stm32 driver. Changes to that code effect the highly RAM constrained PX4IO and can trigger the out of memory "police lights" which most people do not notice, and then the future bug report is RC calibration is not working.

ChristophTobler, 2017-04-10:
@davids5 It doesn't work with latest master

vinhk, 2017-04-13:
The lidar to pwm not fixed yet?

tomatac, 2017-04-14:
Is there a known version/commit where the PWM input works?

ChristophTobler, 2017-04-15:
Yes it doesn't work after commit https://github.com/PX4/Firmware/commit/aa9fbbedd54a7a0e346ea03f439dcdf4f90c6113 as can be seen in the first comment

mhkabir, 2017-04-15:
@ChristophTobler can you try on master again? If it still doesn't work, try changing the "pwm4" to "none" here : https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L468-L469

ChristophTobler, 2017-04-15:
@mhkabir I'm currently on holidays and therefore cannot test it. But maybe @julianoes can give it a quick try on tuesday? Otherwise, I'll do it of course once I'm back.

ChristophTobler, 2017-04-25:
@mhkabir I have tested it with latest master and it works. Thanks!

closed 2017-04-25 by ChristophTobler

mhkabir, 2017-04-25:
:D

tomatac, 2017-04-30:
What is this command suppose to show "nsh> pwm_input test"?

julianoes, 2017-04-30:
You should see samples of the measured pulse width e.g. 1500 us.

tomatac, 2017-04-30:
Should I see the pulses only if the Lidar is powered or I should see pulses regardless of that?

mhkabir, 2017-04-30:
Only if it's powered.

tomatac, 2017-04-30:
Ok. Thanks! I will connect power to the servo rail.