PX4/PX4-Autopilot #7002: How can I tune Px4 with Px4Flow data to give better hovering?

vinhk, 2017-04-07:
I am able to see Px4Flow data into the Pixhawk but there is still hovering issue as it tends to drift. I am hoping there is a documentation similar to one in Arducopter: http://ardupilot.org/copter/docs/common-px4flow-overview.html

ChristophTobler, 2017-04-10:
You have to tune the position controller. `MPC_Z_VEL_P/I/D` etc.

vinhk, 2017-04-11:
@ChristophTobler is there a way to do testing like so 
[image]

Where users get to walk their quad around without propeller and do row for 1 session and pitch for another session. Those file will be use to log as above picture is shown. This would make tuning much easier.

tomatac, 2017-04-11:
What firmware are you using?
Do you use LPE? If so what is your LPE_FUSION setting?
I am geting POSITION CONTROL REJECTED (Firmware 1.6 - LPE version, PX4FLOW + Lidar Lite 3).

vinhk, 2017-04-11:
@tomatac I am using LPE as well but currently unable to see data from pwm lidar lite v3. For LPE_FUSION, I have it at 242 i think. I have been through your issue forum about rejecting mode for position control. I am not sure why. Mine is not rejecting but it drifts

mhkabir, 2017-04-12:
@vinhk Use the i2c interface of Lidar Lite v3. It now works well. You might need to modify the startup script to test.

vinhk, 2017-04-12:
@mhkabir which startup script part should I modified?

tomatac, 2017-04-13:
Does the raw data from the Lidar get recorded in the flight log? I using PWM.

mhkabir, 2017-04-13:
PWM for Lidar Lite is currently broken.

tomatac, 2017-04-13:
Is it broken in 1.6 or the master? Is there a version where it works?

closed 2017-06-20 by mhkabir