PX4/PX4-Autopilot #7004: EKF2 doesn't work with px4flow

nancy03610, 2017-04-08:
I want to test px4flow. I upload px4fmu-v2_default target and set the params EKF2_AID_MASK and EKF2_HGT_MODE as mentioned in #6864, and I also enabled SENS_EN_MB12XX sensor because I want to use the sonar sensor in the px4flow module.

 I didn't do a real flight test but just hold the flight walked around. It turned out that the light was always blue. I suppose that green light shows it work fine, so I am afraid that the ekf2 didn't work. I do get the flow data in logs and I think the flow quality is ok. Did I miss anything? What should I do to fix it?

I also try the same test for LPE. I chose to use LPE in sys_mc_est_group, and because there is no LPE_FUSION in px4fmu-v2_default, I change this param in the source code.  Everything else was same with ekf2 test, and I got a very same result.

HockyYoung, 2017-04-10:
Maybe You shoud see this https://github.com/PX4/Firmware/issues/6864

nancy03610, 2017-04-10:
Yes, I have read that. I followed their instruction, and tried px4fmu-v2_lpe, but still no luck.

nancy03610, 2017-04-10:
@HockyYoung I read your comment in #6864, found that I am basically doing the same thing with you. I got a problem when changed flight modes. It was ok with altitude, but when I tried to use position mode or hold mode there was a warning 'reject position control' or 'reject auto hold'. Did you meet the same problem?

HockyYoung, 2017-04-10:
I didn't meet the problem just you mentioned.Did you check your PX4Flow date?you can use QGroundControl to analyze the px4flow date by connecting the pixhawk and shake your px4flow and look carefully whether the qulity is changing.Let me know if your px4flow date is well.

mhkabir, 2017-04-10:
@ChristophTobler can you check this please.

nancy03610, 2017-04-10:
Yes, I can see the px4flow output in QGC, it looks fine. But also I notice that in my last time flight log the optical flow quality was low. I think this may be the reason why I failed in altitude mode(as I mentioned in #7017) but cannot explain why I cannot change to position mode.

ChristophTobler, 2017-04-10:
@nancy03610 can you please provide a log here (if it's a .ulg file) http://logs.px4.io/upload or here http://logs.uaventure.com/ (if it's a .px4log file)

nancy03610, 2017-04-10:
Here is my log: http://logs.uaventure.com/view/cCt8EfG4f8XhhcPSbbtcK5

ChristophTobler, 2017-04-10:
@nancy03610 you are using the LPE in this log not the EKF2...

nancy03610, 2017-04-10:
Yes, it is the log for LPE test. I got problem both in LPE and EKF2 when I held the flight walking around. Then I did real flight test with LPE first, and met with the flight mode change problem, which is the log I just uploaded. I didn't do real flight test for ekf2 yet.  Sorry for the misunderstanding.

And here is the log of my hold-flight-walking-around test with EKF2 : http://logs.uaventure.com/view/A5WFhjbjZaiCXfYugu9oFS

ChristophTobler, 2017-04-10:
@nancy03610 So the problem was that the light does not turn green? My understanding is that it just turns green if you have/use GPS.

nancy03610, 2017-04-10:
Oh, I see where I go wrong. I thought green light shows LPE or EKF2 works. Thank you so much. 

By the way, do you know why I cannot change flight mode when using LPE?

ChristophTobler, 2017-04-10:
Good. We'll keep track of the LPE problem in https://github.com/PX4/Firmware/issues/7010

closed 2017-04-10 by ChristophTobler

xuefengleng2016, 2018-02-10:
@nancy03610 @ChristophTobler i also get the same problem and i had set the parameter like you recommended. this is the analysis of the test indoor https://logs.px4.io/plot_app?log=103d64ad-be0b-4f8b-a8be-ea5249328b9d it seems the px4flow not works

ChristophTobler, 2018-02-12:
@xuefengleng2016 Your PX4Flow reports low quality. This can mean
* You have not enough texture on the ground
* It's too dark
* The lens is not focused