PX4/PX4-Autopilot #7005: how to disable GPS in Gazebo simulator?

vinhk, 2017-04-08:
Is there a way to disable gazebo GPS functionality?

mhkabir, 2017-04-08:
`gpssim stop`

closed 2017-04-08 by mhkabir

Coderx7, 2021-08-10:
@mhkabir Hi, please excuse my ignorance but where should we enter this command? when I enter it in terminal it says
```
Command 'gpssim' not found, did you mean:

  command 'gpsim' from deb gpsim
  command 'lpssim' from deb mcrl2

Try: sudo apt install <deb name>

```

rrajnidhi, 2022-03-09:
Hi Did u found the answer for stoping GPS simulaion on GAZEBO ?

Coderx7, 2022-03-09:
@rrajnidhi Hi, I searched alot after this and it seems the only way to have control over the gps simulation, is to use HITL mode, which basically means, to add your FCU, and then using QGC e.g disable its GPS, then Gazebo will use the sensors available on the FCU and simulate everything else. 
I could not find any SITL solution where we can simply disable/enable GPS in gazebo. that would be great. I'm not sure if its possible or HITL is the only way.

rrajnidhi, 2022-03-09:
@Coderx7 thanks a lot for the idea, will try that way too.

Jaeyoung-Lim, 2022-03-09:
@rrajnidhi @Coderx7 You can simply remove the gps from the model: https://github.com/PX4/PX4-SITL_gazebo/blob/25138e803ee8525ee5fe4e6d511506e88e3f819c/models/iris/iris.sdf.jinja#L421-L429

Coderx7, 2022-03-10:
@Jaeyoung-Lim Thanks a lot , I'll give it a try and see if that solves it ( I recall messing around but I'm not sure if I did this already or not, nonetheless, thanks a lot greatly appreciate it)