PX4/PX4-Autopilot #7010: Why I couldn't change flight mode to 'position control' or  'auto hold' when using LPE?

nancy03610, 2017-04-10:
I am working on testing LPE with px4flow.
I uploaded px4fmu-v2_lpe target and followed the instruction in #6864 to set the params in QGC. I set SYS_MC_EST_GROUP to LPE and LPE_FUSION to 162.
I did a flight test, I took off with Stabilized mode and turned to Altitude mode. The flight can keep the same hight but slowly flew away. Is it normal to act like this?
Then I wanted to try position mode and hold mode, but when I set them in QGC I got the warnings 'reject position control' and 'reject auto hold'. Why did this happen? Did I use the right flight mode for the px4flow test?

mhkabir, 2017-04-10:
You do not have the LPE_FUSION flag set up correctly. Optical flow is not enabled.

mhkabir, 2017-04-10:
Send logs.

nancy03610, 2017-04-10:
Sorry, I am just a beginner. How can I send my log?

nancy03610, 2017-04-10:
@mhkabir Here is my log: http://logs.uaventure.com/view/cCt8EfG4f8XhhcPSbbtcK5

nancy03610, 2017-04-15:
I changed LPE_FUSION to optical flow, pub agl as lops down, land detector, gyro compensation and fuse baro. Now it can change to position control mode, although I still don't know why. Anyway, thank you for the response.

closed 2017-04-15 by nancy03610

saptadeepdeb, 2017-05-18:
I was having the same problem. Used 114 LPE_FUSION, and wasnt able to change the flight mode. Position hold and Altitude hold both of them were getting rejected.

I have a laser rangefinder as well, so I didnt wanted to use the Baro of the PX4Flow (reason for keeping the 8th bit of LPE_FUSION as 0). 

Also want to know what does 'pub agl lpos down' mean?