PX4/PX4-Autopilot #7018: Upload/flash fails on master

mhkabir, 2017-04-11:
There is an issue on master where upload to STM32 family boards (tested with FMU v4, FMU v4Pro, FMU v2) will fail. This issue is not present on the current release version. Something seems to have changed in the uploader. Tested on both macOS and Linux, on 2 different computers, with different cables each time.

[image]

LorenzMeier, 2017-04-12:
Same here:
```
Erase  : [====================] 100.0%
Program: [================    ] 80.3%
ERROR: timeout waiting for data (1 bytes)
```

julianoes, 2017-04-12:
Oh, here we go on v3:
```
Erase  : [====================] 100.0%
Program: [==================  ] 93.7%
ERROR: timeout waiting for data (1 bytes)
[100%] Built target upload
```

julianoes, 2017-04-12:
Am I right that this only happens when trying to flash v3 onto Pixhawk? I can't reproduce it with v2 on Pixhawk or v4 on Pixracer.

julianoes, 2017-04-12:
Even raising the serial timeout to 3s did not resolve the issue.

davids5, 2017-04-12:
Has anyone pulled the https://github.com/PX4/Firmware/commits/stable/Tools/px_uploader.py to see if the issue follows the tool?

julianoes, 2017-04-12:
@davids5 makes sense, will do.

julianoes, 2017-04-12:
@mhkabir actually can you please check, cause I can barely reproduce the error unless flashing v3 on a v2 with silicon errate.

davids5, 2017-04-12:
@mhkabir 

Please post the output from the ```ver all``` and the upload for each HW that is failing. 

Also does the  https://github.com/PX4/Firmware/commits/stable/Tools/px_uploader.py not have any issues on the same set of HW?

davids5, 2017-04-22:
@mhkabir - any progress?

mhkabir, 2017-04-23:
@davids5 Sorry about that - I missed your message. Will test tomorrow and get back to you.

LorenzMeier, 2017-04-23:
I see this only with `make upload px4fmu-v3_default` on old HW (which has the silicon errata). @mhkabir does this happen on new HW as well? Can you share the output of `ver all`?

mhkabir, 2017-04-23:
Yes this happens on new hardware as well. All of the boards which I reproduced this on had no silicon errata. Will share `ver all` tomorrow.

mhkabir, 2017-04-24:
Strangely enough, I can't reproduce this anymore.

This is `ver all` from the board I reproduced the failure on previously :

```
nsh> ver all
HW arch: PX4FMU_V4PRO
FW git-hash: 9050979b476b49d7eae2fe9a4f4b373bb2738064
FW version: 1.6.0 0 (17170432)
OS: NuttX
OS version: Release 1.8.0 (17301759)
Build datetime: Apr 22 2017 15:28:09
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
MFGUID: 32343031343451160023001f
MCU: STM32F???, rev. A
UID: 23001F:34345116:32343031 

```
@bresch Also got the same issue on his Dropix (which I don't have access to at the moment), and I checked previously - it had no silicon errata.

LorenzMeier, 2017-04-24:
There were changes / fixes to the script. Closing for now.

julianoes, 2017-04-24:
@LorenzMeier If you're referring to https://github.com/PX4/Firmware/pull/7090, this has nothing to do with the timeout that we get on upload.

mhkabir, 2017-04-24:
Reproduced again on the v4Pro board (`ver all` attached in my last post): 

```
Mohammeds-MacBook-Pro:Firmware kabir$ make px4fmu-v4pro_default upload
[4/5] uploading /Users/kabir/Developme...irmware/nuttx/px4fmu-v4pro_default.px4
Loaded firmware for d,0, size: 1388592 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Found board d,0 bootloader rev 5 on /dev/tty.usbmodem1
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: ÿÿÿÿ
idtype: =FF
vid: ffffffff
pid: ffffffff
coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=

sn: 0023001f3434511632343031
chip: 10006434
family: STM32F???
revision: A
flash 2080768

Erase  : [====================] 100.0% (timeout: 8 seconds) 
Program: [==================  ] 92.9%
ERROR: timeout waiting for data (1 bytes)

```

davids5, 2017-04-24:
> There is an issue on master where upload to STM32 family boards (tested with FMU v4, FMU v4Pro, FMU v2) will fail. This issue is not present on the current release version. Something seems to have [...]

1. Is the above statement still true?
2. Is this a consistent failure?  If not what it the total number of tries vs failures?
3. Is it repeatable with a power cycle on each and every upload.
4. Do you have a debug connector with FTDI connected during the  uploads?
5. On the V4Pro do you have the jumper on the BYPASS-REG installed and what rev is the HW?

Below is the output from my Centos 7 Linux box:

```
HW arch: PX4FMU_V4PRO
FW git-hash: 44a507fcfe18ba3bb1c53c79d081de2801a060c8
FW version: 1.6.0 0 (17170432)
OS: NuttX
OS version: Release 1.8.0 (17301759)
Build datetime: Apr 24 2017 07:34:08
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MFGUID: 32343031343451170040002d
MCU: STM32F???, rev. A
UID: 40002D:34345117:32343031
nsh>
```

```
07:35:19 (master) ~/PX4/Firmware# make px4fmu-v4pro_default upload
[0/1] uploading /root/PX4/Firmware/build_px4fmu-v4pro_default/src/firmware/nuttx/px4fmu-v4pro_default.px4
==========================================================================================================
WARNING: You should uninstall ModemManager as it conflicts with any non-modem serial device (like Pixhawk)
==========================================================================================================
Loaded firmware for d,0, size: 1389824 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Found board d,0 bootloader rev 5 on /dev/serial/by-id/usb-3D_Robotics_PX4_BL_FMU_v4.x_PRO_0-if00
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: ÿÿÿÿ
idtype: =FF
vid: ffffffff
pid: ffffffff
coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=

sn: 0040002d3434511732343031
chip: 10006434
family: STM32F???
revision: A
flash 2080768

Erase  : [====================] 100.0% (timeout: 6 seconds) 
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.
```

mhkabir, 2017-04-24:
> Is the above statement still true?

Yes.

> Is this a consistent failure? If not what it the total number of tries vs failures?

No. It fails approximately once every five tries.

> Is it repeatable with a power cycle on each and every upload.

No, see above statement.

> Do you have a debug connector with FTDI connected during the uploads?

Yes. Not sure what to look out for though.

> On the V4Pro do you have the jumper on the BYPASS-REG installed and what rev is the HW?

Not sure about this one. @eyeam3 should be able to fill you in here.

davids5, 2017-04-24:
>>Do you have a debug connector with FTDI connected during the uploads?

> Yes. Not sure what to look out for though.

Back feeding of voltage on the U[S]ART IO pins so that a USB unplugged conditions is not a clean power cycle.

If you can get it to fail 1/5 times - try to get a n without the FTDI connected.

davids5, 2017-04-24:
@mhkabir - if you do 10 uploads each on V2 HW with both master and "the current released" version. 
You are getting 0 failures with "the current released" uploader?

mhkabir, 2017-04-25:
This is from V2 hardware : 

```
nsh> ver all
HW arch: PX4FMU_V2
FW git-hash: 1a11bc98354659721d6521dc3a2299623600b280
FW version: 1.6.0 0 (17170432)
OS: NuttX
OS version: Release 1.8.0 (17301759)
Build datetime: Apr 24 2017 11:49:35
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MFGUID: 36353534343551190032003f
MCU: STM32F42x, rev. 3
UID: 32003F:34355119:36353534
```

```
[7/7] uploading /home/mathieu/src/Firmware/build_px4fmu-v3_default/src/firmware/nuttx/px4fmu-v3_default.px4
Loaded firmware for 9,0, size: 1458164 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Attempting reboot on /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00 with baudrate=57600...
If the board does not respond, unplug and re-plug the USB connector.
Found board 9,0 bootloader rev 4 on /dev/serial/by-id/usb-3D_Robotics_PX4_BL_FMU_v2.x_0-if00
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: ÿÿÿÿ
idtype: =FF
vid: ffffffff
pid: ffffffff
coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=

sn: 0032003f3435511936353534

Erase  : [====================] 100.0% (timeout: 7 seconds) 
Program: [===============     ] 79.6%
ERROR: timeout waiting for data (1 bytes)
```

mhkabir, 2017-04-25:
More information : I copied the `px_uploader.py` from the v1.5.5 release over to master, and I can still reproduce the fail. Beginning to wonder if it's the image size which triggers it.

mhkabir, 2017-04-25:
OK I can reproduce this on v1.5.5 as well now (FMUv2, `px4fmu-v3_default` target. `ver all` in above post. No FTDI connected)

> This issue is not present on the current release version

Now an invalid statement.

davids5, 2017-04-25:
Ok @mhkabir great feedback - thank you!

>More information : I copied the px_uploader.py from the v1.5.5 release over to master, and I can still reproduce the fail. Beginning to wonder if it's the image size which triggers it.

Yes that is evident in the erase"```Erase  : [====================] 100.0% (timeout: 7 seconds) ```

1) Timeout should be based on the flash size reported back. But it is not simple..... it will need to be based off the 128K sector size. See below.

>>>OK I can reproduce this on v1.5.5 as well now (FMUv2, px4fmu-v3_default target. ver all in above post. No FTDI connected)

>>This issue is not present on the current release version

>Now an invalid statement

So then that leaves both sides of the wire suspect. @mhkabir do you have time to add some instrumentation to master's '''px_uploader.py'''  

Print or track the address that the timeout happens on. 

Reason being F4's sector sizes are not homogeneous. 

[image]

So let see if it maps to a change in the sector size during programing.

dagar, 2017-04-26:
@mhkabir it sounds like we need to instrument the uploader script and reproduce.

mhkabir, 2017-04-29:
I will have a look on Monday.

LorenzMeier, 2017-05-03:
@mhkabir What is your current status on this?

LorenzMeier, 2017-05-03:
Annoying but not critical enough to gate the release.

davids5, 2017-06-24:
@mhkabir https://github.com/PX4/Firmware/issues/7018#issuecomment-298184507 Did you ever instrument the downloader to see the cause?

mhkabir, 2017-06-24:
@davids5 Unfortunately I didn't as this was low prio, and I'm not proficient in Python.

davids5, 2017-06-24:
Ahh - I know how you feel:) @julianoes can you lend a hand here?

julianoes, 2017-06-26:
From what I found when I tried to debug it last, it just seems to time out waiting for an ack after having done a usual write of 252 bytes.

LorenzMeier, 2017-08-20:
Related to USB hubs for me. Closing as stale.

ryanjAA, 2017-10-17:
This is occurring again. 

When uploading from a fmuv2 firmware to a v3 on a pixhawk 1 that has the 2mb flash it fails. Every time.

If I build the v3 firmware and upload at the same time from cli it works.

Once a v3 firmware is on there you can freely upload to other v3 firmware without error from qgc. 

Take it back down to v2 is no problem but back up to v3 and it fails again.

Error is Timeout waiting for bytes at address

The only way to go to a v3 firmware from v2 not from cli is to try uploading the same v3 firmware twice and the second time it takes it.

Tried on MRO Pix1 and Chinese pix1, both with 2mb flash. Different cables and direct connection on Mac.

ryanjAA, 2017-10-17:
This is on Master and daily of qgc.

LorenzMeier, 2017-12-31:
Closing as stale.

closed 2017-12-31 by LorenzMeier

afwilkin, 2020-08-12:
Noting here that I was experiencing this exact issue tonight, and resolved it by uninstalling ModemManager. There is a note of it when manually flashing the device
`WARNING: You should uninstall ModemManager as it conflicts with any non-modem serial device (like Pixhawk)`

However, this does not appear in QGroundControl.

radihoca, 2022-05-19:
I got the deletion error too. Any progress on solving the problem?

julianoes, 2022-05-19:
@radihoca please open a new issue and provide all the detail of what you tried and what results you got.