PX4/PX4-Autopilot #7020: Odd CPU usage on master

bresch, 2017-04-11:
Hi everyone,
I compared the preflights reports between the v1.5.5 release and Master.
The CPU usage on Master looks odd: the load is totally on the scheduler and 0.00% on the tasks.

Preflight report on master:
```PERFORMANCE COUNTERS PRE-FLIGHT

sd write: 0 events, 0us elapsed, 0us avg, min 0us max 0us 0.000us rms
navigator: 1173 events, 136350us elapsed, 116us avg, min 26us max 3688us 339.153us rms
fw l1 control: 2838 events, 117157us elapsed, 41us avg, min 28us max 299us 23.215us rms
fwa_dt: 16468 events, 1691243us elapsed, 102us avg, min 45us max 524us 28.851us rms
mc_att_control: 17905 events, 2364683us elapsed, 132us avg, min 20us max 435us 35.466us rms
ctl_lat: 3051 events, 149868us elapsed, 49us avg, min 27us max 2597us 155.577us rms
sd drop: 0 events
mavlink_txe: 0 events
mavlink_el: 0 events, 0us elapsed, 0us avg, min 0us max 0us 0.000us rms
mavlink_txe: 211 events
mavlink_el: 6064 events, 959010us elapsed, 158us avg, min 54us max 3482us 215.893us rms
mavlink_txe: 0 events
mavlink_el: 6802 events, 2499487us elapsed, 367us avg, min 76us max 905480us 10982.588us rms
io latency: 17585 events, 725380027344us elapsed, 41249930us avg, min 6048314us max 76450888us 20323750.000us rms
io write: 0 events, 0us elapsed, 0us avg, min 0us max 0us 0.000us rms
io update: 17673 events, 12344246us elapsed, 698us avg, min 254us max 19610us 828.044us rms
io_txns: 36244 events, 11404518us elapsed, 314us avg, min 128us max 3547us 227.466us rms
stack_check: 205 events, 2847us elapsed, 13us avg, min 1us max 139us 19.226us rms
sensors: 18093 events, 3329729us elapsed, 184us avg, min 83us max 838us 70.555us rms
aspd_com_err: 0 events
aspd_read: 6096 events, 4600474us elapsed, 754us avg, min 494us max 3848us 573.412us rms
lsm303d_acc_dupe: 13165 events
lsm303d_bad_val: 0 events
lsm303d_bad_reg: 0 events
lsm303d_mag_read: 7250 events, 165410us elapsed, 22us avg, min 22us max 24us 0.399us rms
lsm303d_acc_read: 69050 events, 2033211us elapsed, 29us avg, min 20us max 35us 4.745us rms
l3gd20_dupe: 47059 events
l3gd20_bad_reg: 0 events
l3gd20_err: 0 events
l3gd20_read: 101408 events, 2805885us elapsed, 27us avg, min 21us max 36us 5.854us rms
ctrl_latency: 17027 events, 6665711us elapsed, 391us avg, min 241us max 1075us 78.397us rms
mpu6k_duplicates: 18176 events
mpu6k_reset: 0 events
mpu6k_good_trans: 72468 events
mpu6k_bad_reg: 0 events
mpu6k_bad_trans: 0 events
mpu6k_read: 90648 events, 4690386us elapsed, 51us avg, min 32us max 75us 12.246us rms
mpu6k_gyro_read: 0 events
mpu6k_acc_read: 20 events
hmc5883_conf_err: 0 events
hmc5883_rng_err: 0 events
hmc5883_com_err: 0 events
hmc5883_read: 3123 events, 3423581us elapsed, 1096us avg, min 866us max 11841us 506.228us rms
adc_samples: 72480 events, 204037us elapsed, 2us avg, min 2us max 3us 0.388us rms
ms5611_com_err: 0 events
ms5611_measure: 6238 events, 49476us elapsed, 7us avg, min 6us max 182us 8.977us rms
ms5611_read: 6237 events, 508176us elapsed, 81us avg, min 10us max 2467us 118.834us rms
dma_alloc: 7 events

LOAD PRE-FLIGHT

Processes: 23 total, 1 running, 0 sleeping
**CPU usage: 0.00% tasks, 67.11% sched, 32.89% idle**
DMA Memory: 5120 total, 2560 used 2560 peak
Uptime: 77.002s total, 37.508s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                   37507 32.892   644/  748   0 (  0)  READY 
   1 hpwork                       3038  3.966   992/ 1780 192 (192)  w:sig 
   2 lpwork                         64  0.000   656/ 1780  50 ( 50)  w:sig 
   3 init                        12437  0.000  1720/ 2484 100 (100)  w:sem 
 311 sdlog2_writer                  43  0.000  1176/ 2044  95 ( 95)  READY 
 102 gps                           191  0.165   936/ 1372 220 (220)  w:sig 
[... 77 lines cut ...]

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                   43340 30.217     0/    0   0 (  0)  READY 
   1 hpwork                       3027  3.672   820/ 1592 192 (192)  w:sig 
   2 lpwork                        260  0.166   364/ 1592  50 ( 50)  READY 
   3 init                         1529  0.000  1160/ 2496 100 (100)  w:sem 
 287 sdlog2_writer                  38  6.343  1124/ 2040  95 ( 95)  w:sem 
 102 gps                           207  0.166   924/ 1192 220 (220)  w:sig 
 104 dataman                       234  0.000   748/ 1192  90 ( 90)  w:sem 
 141 sensors                      4181  4.507  1060/ 1696 250 (250)  w:sem 
 143 commander                    2662  2.003  2804/ 3192 140 (140)  READY 
 149 px4io                        2747  3.338   908/ 1392 240 (240)  w:sem 
 157 mavlink_if0                  1227  1.335  1812/ 2392 100 (100)  READY 
 158 mavlink_rcv_if0                42  0.000  1548/ 2096 175 (175)  w:sem 
 172 mavlink_if1                   915  1.001  1716/ 2392 100 (100)  READY 
 174 mavlink_rcv_if1                 7  0.000   956/ 2096 175 (175)  w:sem 
 214 sdlog2                        117  6.510  2508/ 3392 177 (177)  RUN   
 216 commander_low_prio              2  0.000   572/ 2992  50 ( 50)  w:sem 
 250 ekf2                        15099 22.036  4972/ 5792 250 (250)  w:sem 
 252 vtol_att_control             2756  3.505   716/ 1192 245 (245)  w:sem 
 260 mc_att_control               2750  3.672  1268/ 1496 250 (250)  w:sem 
 265 mc_pos_control               2563  3.338  1020/ 1896 250 (250)  w:sem 
 267 fw_att_control               3090  3.672  1036/ 1392 250 (250)  w:sem 
 274 fw_pos_ctrl_l1                188  0.333   892/ 1696 250 (250)  w:sem 
 279 navigator                     726  1.836  1020/ 1496 105 (105)  READY ```

bresch, 2017-04-11:
Also some feelings during the flights (same parameters for the controllers, the only thing I had to change is the gyro calibration because QGC asked it):
- Attitude control is worse on master. You can hear the motors vibrating and the attitude is less stable.
- During position control, the controller works fine but looks more aggressive when braking (not sure if it's a feature or related to the bad stability).

You can really see the difference in the log files below:
- Release v1.5.5: http://logs.uaventure.com/view/MBM8a4gLYuLHKRAt9gcV9R
- Master: http://logs.uaventure.com/view/ZptHsQJVWFRvbXF4Bhf8ZT

Especially on the yaw rates (I use a VTOL so the roll rates are damped by the wings):
V1.5.5:
[image]

Master:
[image]

LorenzMeier, 2017-04-12:
Looks like a reporting bug. The tasks actually show correct load values.

mhkabir, 2017-04-18:
Fixed in https://github.com/PX4/Firmware/pull/7074

closed 2017-04-18 by mhkabir