PX4/PX4-Autopilot #7021: ATT.Pitch became more and more big, vehicle position draft

liang-tang, 2017-04-11:
ATT.Pitch  got big and big, from 0.025 to 0.1 almost in five minutes, the vehicle horizontal position draft more and more quickly , so any issure with mc_pos_control or mc_att_control? 
(software version: 1.5.5
hardware pixhawk)
[image]

liang-tang, 2017-04-11:
Log
http://logs.uaventure.com/view/nwpou7XbSC97DMy226bfYU

bresch, 2017-04-11:
Did you check if your compass is correctly pointing towards the magnetic North?

liang-tang, 2017-04-12:
i haven't check it ... how can i  comfirm this issure caused by compass direction.?

bresch, 2017-04-12:
Let me explain a bit: the attitude estimation and control look fine. The pitch reference is moving with the X position error. My guess is that your position controller is correcting in the wrong direction. (However, I'm surprised that it does not diverge faster and why it is only visible on one axis).
So, to verify this, just go outside and compare the heading displayed by QGC with the compass of your smartphone or with a known geographic reference.

And if you are sure that you can swith back to manual and pilot it manually, you can try to move the pitch stick of your remote to see if the quad position diverge (be careful to switch back quickly to manual). If you are not sure to be able to do this, don't try.

liang-tang, 2017-04-13:
thanks a lot, i will check it before takeoff.

bresch, 2017-04-19:
@liang-tang  Any news?

liang-tang, 2017-04-19:
everything works well.

closed 2017-04-19 by mhkabir