PX4/PX4-Autopilot #7028: Question: Take off and landing modes.

tubeme, 2017-04-12:
Hello people,

Because I'm updating the UM and ready to release the modes page, can somebody clarify what is the behavior of Take off and Landing MODES for:

Take off FW:
Once activated does it spin the motor by itself? 
Difference in take off from airstrip or hand launch?
Once activated or thrown it gains altitude? What altitude it gains? 
After altitude gain does it get in Loiter waiting for switch to another mode?
How it is activated? There was a PR I remember where tthe take of is activated DJI style with just the throttle

Landing FW:
If too high in altitude does it make a spiral up to some lower altitude?
Once it reached this lower altitude in spiral does it straighten until ground touch?
Once on the ground does it disarm automatically? (in the past we've had a lot of broken props because the plane did not stop its engine)
If RTL activated and the plane is loitering for some time what is the strategy for landing after the grace period? Does it straighten the plane before touch down? (before it just landed in spiral, thus touching down the internal wing thus braking sometimes the plane)
What is the command to activate it? Is t done with the throttle lever only or with a switch?

Take off MC:
Once activated it spins and takes off at what altitude it weights for your commands?
How is it activated? Is it DJI style with the throttle or with a switch?

Landing MC:
How is it activated?
Does it disarm the moment it feels it is in the air? or should I disarm it with a command?

I suppose the VTOL is a combination between the two, although there might be a scenario for the VTOL emergency landing in FW mode? Is there such a scenario?

Is there a way to setup Take off and Landing with the old style switch configuration? I might not have 6 pos switch to use the new style.

Thanks

dagar, 2017-04-12:
This might be a good excuse to expand the PX4 user guide. FW planned auto landings have improved substantially recently in QGC. 

@hamishwillee FYI

Should we pick away at these items collaboratively?

MaEtUgR, 2017-04-12:
@tubeme I can only speak for MC.
You can take off manually or automatically.
**Manually**
Go to position mode, arm, as soon as you raise your manual throttle stick above `LNDMC_POS_UPTHR` (default 65%) it enables all controllers and goes to hover throttle `MPC_THR_HOVER` (should be set reasonably to not have a jump). For landing just press the stick down until it is completely landed (goes through multiple steps) and disarm or best set `COM_DISARM_LAND` > 0 and it will diasrm automatically.

**Automatically**
Best to be in hold mode first, arm, switch to takeoff mode and it will fly up pretty fast to `MIS_TAKEOFF_ALT` and wait for further input. Switching out by moving the sticks is currently possible (but will be disabled by default as it looks now here: https://github.com/PX4/Firmware/pull/7025). To Land again automatically you either switch to RTL or Land mode. Land mode just descends in place with `MPC_LAND_SPEED` (pretty slowly as we don't know whats under there) until it hits the ground and you can auto disarm again as above. In RTL it flies first up to `RTL_RETURN_ALT`, in a straight line back to the home position, descends first fast and then with the slow land speed because we know how high the ground should be.

The automatical modes are easy to access through the QGC buttons on the bottom of the main flight screen.

hamishwillee, 2017-04-13:
@dagar 

> This might be a good excuse to expand the PX4 user guide. FW planned auto landings have improved substantially recently in QGC.
> Should we pick away at these items collaboratively?

Yes. Can we discuss Tuesday (I'm heading off for Easter). Have created https://github.com/PX4/px4_user_guide/issues/4 to track this.

FYI the user guide is now in the PX4 github organisation and published on gitbook here: https://docs.px4.io/en/ . Updates should be done as fork/PR. If you want to create docs early, go ahead!

tubeme, 2017-04-13:
Thanks @MaEtUgR 

@dagar Anything about the FW takeoff and landing?

hamishwillee, 2017-04-19:
@dagar OK, so I'm starting this from "first principles" - improving/expanding out the flight modes docs (at the moment this looks like the screenshot below and will live as a peer of [this Follow me mode doc](https://docs.px4.io/en/flying/flight_mode_follow_me.html)). Once we have those we can link from various places to explain the behaviour. 

I have some questions below the screenshot below. I appreciate you may not be able to answer them, but can you please link in the people you think might be able to?

[image]

## Questions
### Multicopter

What speed does it return at?

### Fixed Wing

The current description is "The aircraft returns to the home position and hovers in a circle above the home position."

- Is this correct?
- What height does it return at? Is this still RTL_RETURN_ALT?
- Does it respect the other parameters? So for example, will it descend to RTL_DESCEND_ALT and try and lad if  `RTL_LAND_DELAY` is not -1 ?
- What is the circle radius for hover?
- What is the speed of return?

### VTOL

This has no description. I'd expect it to fly up to safe distance like a copter and land like a copter. I'd expect it to decide whether to fly the return in FW or copter based on the distance. But who knows, and if so, what parameters would it use?

- Does it return as copter or VTOL and how?
- What parameters does it obey.

sanderux, 2017-04-19:
In the case of VTOL:

- The vehicle returns to home in the mode it is currently in (FW or MC)
- The vehicle transitions back to MC
- Dependent on the LAND_DELAY param it will either loiter there or land

hamishwillee, 2017-04-19:
Thanks @sanderux - very useful. W.r.t. to the statement "Dependent on the LAND_DELAY param it will either loiter there or land" is it more accurate to say that the vehicle will perform and ordinary MC RTL landing based on the parameters? The difference being that if this is the case then we can say that the vehicle should also respect the RTL_DESCEND_ALT value.

Also, do you happen to know what the return altitude is (and for FW). Does it respect RTL_RETURN_ALT?

hamishwillee, 2017-04-19:
Also @sanderux do you happen to know the LAND mode VTOL behaviour?

sanderux, 2017-04-22:
@hamishwillee The LAND part in RTL for a VTOL is in essence no different from MC indeed, it respects the same parameters. As for the RTL alt, it will fly at that altitude towards home.

LAND mode looks at the param VT_NAV_FORCE_VT, when any type of landing is commanded in fixedwing and the vehicle has the parameter set it will always switch back to MC first and then land. If this param is disabled it will behave consistent with it's current mode, so land as FW when in FW and land as MC when in MC. The reason for this is that most VTOL vehicles dont have landing gears so this param is enabled by default.

tubeme, 2017-04-25:
Thanks to all! 

Somebody for the FW mode?? BTW the airplane modes are the once that need more clearance on the latest implementations...

hamishwillee, 2017-04-25:
OK, so proposed version for RTL is [here](https://github.com/hamishwillee/px4_user_guide/blob/47c40c42162aed4c24d93cdddb69184b02091ef0/en/flight_modes/rtl.md). @sanderux thanks for all your help, this has very modified version of your VTOL details. 

@dagar , can you please comment on the FW behaviour (as requested by @tubeme). I have *guessed* in that linked doc :-) Specific questions:
I do have some more confirmations/questionss:
- Does FW ignore the RTL_MIN_DIST parameter (I have assumed so and removed it).
- Can FW LAND in RTL? Ie. if I set RTL_LAND_DELAY >0 will it try land after the time expires? If so, will this be just like LAND mode for FW?
- What parameters affect the flight speed and circle? Is the circle behaviour on arrival exactly the same as HOLD mode (and does it transition into HOLD mode or remain in RTL mode).

hamishwillee, 2017-04-26:
@MaEtUgR , can you please review the MC sections of these [four flight mode docs](https://github.com/hamishwillee/px4_user_guide/tree/hgw_landing_takeoff_mc/en/flight_modes)?
- Did I capture the right parameters
- Is the description correct?
- Is the description complete? For example, in RTL if the vehicle goes to the mode at an altitude above the RTL_RETURN_ALT, will it fly back at its current alt or will it first descend to the return alt?

MaEtUgR, 2017-04-26:
@hamishwillee Thanks a lot for documenting!
hopefully someone else corrects me if I tell something wrong: 
- HOLD as far as I know is usually the fallback waiting mode if you fly autonomously without RC. When I fly multicopter with RC and want to stay in place I just center my sticks in positon mode... Also I don't know of any ascending when switching to Hold, it should really just stay where it is.
- RTL It stays on the current altitude if it's higher than RETURN_ALT in the beginning. The delay before the final descend `RTL_LAND_DELAY` is indended to wait for the landing gear to go down which is triggered automatically.
- TAKEOFF I think you ccould switch to the mode and then arm but the QGC UI does not allow that. The vehicle does not switch to Hold after taking off but rather stays in takeoff mode, I don't know why maybe we can change that in the future because it does basically the same. (JFI I'm working on a takeoff with which you can configure how fast it should ramp up in the beginning.)

hamishwillee, 2017-04-27:
HI @MaEtUgR 

Thanks very much. I've [updated the docs](https://github.com/hamishwillee/px4_user_guide/tree/hgw_landing_takeoff_mc/en/flight_modes) in line with this.

Re your comments
* HOLD
  * This now focuses on "pausing a mission" rather than "emergency pause". I left that emergency stuff in though, because it is probably still useful for FW.
  * `Also I don't know of any ascending when switching to Hold, it should really just stay where it is.` - AFAIK it will using MIS_LTRMIN_ALT ([based on this code](https://github.com/PX4/Firmware/blob/master/src/modules/navigator/loiter.cpp#L61)  - I assume this is the loiter mode definition?)
* RTL
  * Updated docs re initial altitude > RETURN_ALT. 
  * Added a note about the RTL_LAND_DELAY intent. This now is very much focused on landing for MC, with the hover "option" being covered mostly in the "adjusting the parameters" section.
* TAKEOFF 
  * I've said you can't arm on basis that QGC is official GCS that we promote and I don't want to confuse people.
  * I removed "Hold" from the description to reduce confusion.

Can you please sanity check the sections again now?

Stifael, 2017-04-27:
@hamishwillee 
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_params.c#L60-L72

MaEtUgR, 2017-04-27:
@hamishwillee Nice!
It seems that you were right about Hold, I never tested/used this before. We have to think about what happens if the pilot wants to fly into a valey...

hamishwillee, 2017-05-01:
Thanks @MaEtUgR . I guess that means MC and VTOL are now pretty much done. Just need to sort out FW and these four can go live :-)

dagar, 2017-05-01:
I've lost track of where we are with this. Which FW parts do you think we need?
In general can we try to emphasize the commonality between MC and FW? These are core PX4 features, but FW and MC sometimes have to handle each step a bit differently.

For example RTL is the same, but a FW and MC loiter and land differently.
In HOLD vehicles always respect MIS_LTRMIN_ALT.

Even takeoff and land can first be discussed generally, as both FW and MC are given the same setpoint type and location.

hamishwillee, 2017-05-02:
@dagar 

> I've lost track of where we are with this. Which FW parts do you think we need?

I'm waiting on you to respond to clarification emails regarding takeoff and landing. 

> In general can we try to emphasize the commonality between MC and FW? These are core PX4 features, but FW and MC sometimes have to handle each step a bit differently.

Did you [read the proposed text](https://github.com/hamishwillee/px4_user_guide/tree/hgw_landing_takeoff_mc/en/flight_modes)? - that is exactly what I've tried to do. 
- Each topic starts with a sentence that outlines the common core behaviour. 
- Then we have separate sections for the details of each vehicle. These do share some common parameters but they share enough differences that we still need the sections. 

> Even takeoff and land can first be discussed generally, as both FW and MC are given the same setpoint type and location.

Yes. But the common stuff is just a sentence. The detail is a bit more. This is what we still need to clarify. 

I don't think this would be all that hard, but we may need to hold as a "chat" or meeting.

dagar, 2017-05-11:
Did you cover everything here? Reopen if not.

closed 2017-05-11 by dagar

hamishwillee, 2017-05-11:
@dagar I didn't have a chance to work on this yes, so I don't know. I don't have rights to reopen issues, but that is OK - have added a note in the associated PR so I'll chase up with you next week (if needed).

hamishwillee, 2017-05-24:
FYI all, the docs updates are now "live" as the [mode docs under here](https://docs.px4.io/en/flight_modes/), the doc [Landing (Fixed Wing)](https://docs.px4.io/en/flying/fixed_wing_landing.html) and these [brief notes on takeoff and landing](https://docs.px4.io/en/flying/basic_flying.html#takeoff-and-landing) at the end of the "Flying 101" doc.
They may not be perfect, so feel free to raise docs bugs if you see any errors or suggestions for improvement,